espressif/esp-idf #5237: RMT errors with v4.2 branch - affected by debug level and printf - possible regression?

DavidAntliff, 2020-05-04:
I have a small demonstration project and an associated library for the DS18B20 one-wire-bus temperature sensor, that I have maintained for the ESP32 for the last few years. The library uses the RMT peripheral to create read slots on the One Wire Bus, and read device activity from the bus. It has worked with all versions of the ESP-IDF since v2.1, including v3.3 and v4.1-beta1.

A few days ago someone tried to use it with the unreleased v4.2 branch of the ESP-IDF and reported that it did not work:

https://github.com/DavidAntliff/esp32-ds18b20-example/issues/19

I was able to reproduce their issue with my own hardware.

It's not clear to me what the exact cause of the issue is, however the log reports errors from the RMT component:

```
Find devices:
Found 0 devices
D (2734) rmt: RMT[0] ERR
D (2734) rmt: status: 0x0c000000
D (2734) rmt: RMT[0] ERR
D (2734) rmt: status: 0x0c000000
D (2734) rmt: RMT[0] ERR
D (2734) rmt: status: 0x00000000
```

The expected result is:

```
Find devices:
foo
  0 : da01162e3d78ee28
Found 1 device
```

I.e. no RMT errors at all.

The very unusual thing is that if `LOG_DEFAULT_LEVEL` is set to Debug, and I add a single line of code in `esp32-ds18b20-example/main/app_main.c` that is just `printf("\n");`, the issue appears to go away.

I.e. change:

```
    // Find all connected devices
    printf("Find devices:\n");
    OneWireBus_ROMCode device_rom_codes[MAX_DEVICES] = {0};
    int num_devices = 0;
    OneWireBus_SearchState search_state = {0};
    bool found = false;
    owb_search_first(owb, &search_state, &found);
    while (found)
    {
```

To this:

```
    // Find all connected devices
    printf("Find devices:\n");
    OneWireBus_ROMCode device_rom_codes[MAX_DEVICES] = {0};
    int num_devices = 0;
    OneWireBus_SearchState search_state = {0};
    bool found = false;
    owb_search_first(owb, &search_state, &found);
    printf("\n");           // THIS LINE ADDED
    while (found)
    {
```

Oddly the boolean result of `owb_search_first(..., found)` (that indicates a device on the bus was detected) seems to depend on the existence of the `printf` immediately *after* it! The function `owb_search_first` uses RMT to generate signals on the bus data line, and it does not detect any devices if the `printf` _immediately following it_ is not present.

To avoid the compiler reordering the code, this also "fixes" the problem:

Replace:

```
    owb_search_first(owb, &search_state, &found);
```

With:

```
    printf("%d\n", owb_search_first(owb, &search_state, &found));
```

However if the Component Config -> Log Output -> Default Log Verbosity (`LOG_DEFAULT_LEVEL`) is set to Warning, then this `printf` change does not fix the issue.

I've also tried adding simple delay instead of `printf` with `ets_delay_us()` and various delays, but it doesn't seem to affect the issue.

This has been reproduced on a DOIT ESP32 board, and multiple ESP-WROOM-32D modules, and multiple DS18B20 devices.

Please advise what extra information I need to provide to help with this issue. What has changed with RMT between 4.1-beta1 and this commit on the 4.2 branch?

## Environment

- Development Kit:      [DOIT ESP32]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v4.2-dev-1303-gcd4fa46f5
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

The following traces may help. They show bus operation when there are *no* slave devices connected, to illustrate the difference with v4.2 branch.

The data line is pulled high by an external resistor, and the GPIO is configured as both an RMT tx and rx, with open-drain enabled (pad_driver = 1).

The master initiates a bus communication by pulling the data line low for 480us, then high for 60us. Then it releases the data line, which is then pulled high by the external resistor.

If a device is present on the bus, it responds by pulling the data line low for a time. If there is no device present, then the bus will stay high:

[image]

However in the situation where it's not working - i.e. debugging is set to Warning, or the `printf` is included (as above), the bus stays low, suggesting that the GPIO driver has not released it.

[image]

With a device attached, the data bus just stays low after the master pulls it low, exactly the same as the image just above.

iCarletto, 2020-05-07:
I David, I was about to open an issue on your repo regarding this problem.
I spent the last two days trying to debug this problem but I could not find any explanation. In my case I do not need to add a printf, simply if `LOG_DEFAULT_LEVEL` is set to Info, no device is detected, however when I set it to Debug it works fine. The weird thing is that in the rmt code there is not even a log line that might actually affect the operation.

DavidAntliff, 2020-05-07:
@iCarletto yes, it's a very strange problem. I'm hoping someone from Espressif sees this soon and can help me debug it. Please use v4.1-beta1 or v3.3 in the meantime.

koobest, 2020-05-12:
Hi, I don't know much about this library, but I made some changes, but I made some modifications to the initialization function `static owb_status _init(owb_rmt_driver_info *info, gpio_num_t gpio_num,
                        rmt_channel_t tx_channel, rmt_channel_t rx_channel)`. I have test on the master branch, it works.  here is my log:
```
Find devices:
  0 : c90416933c6cff28
Found 1 device
Single device c90416933c6cff28 present
Single device optimisations enabled

Temperature readings (degrees C): sample 1
  0: 31.2    0 errors

Temperature readings (degrees C): sample 2
  0: 31.2    0 errors

```

the changes:

```
static owb_status _init(owb_rmt_driver_info *info, gpio_num_t gpio_num,
                        rmt_channel_t tx_channel, rmt_channel_t rx_channel)
{
    owb_status status = OWB_STATUS_HW_ERROR;

    // Ensure the RMT peripheral is not already running
    // Note: if using RMT elsewhere, don't call this here, call it at the start of your program instead.
    //periph_module_disable(PERIPH_RMT_MODULE);
    //periph_module_enable(PERIPH_RMT_MODULE);

    info->bus.driver = &rmt_function_table;
    info->tx_channel = tx_channel;
    info->rx_channel = rx_channel;
    info->gpio = gpio_num;

#ifdef OW_DEBUG
    ESP_LOGI(TAG, "RMT TX channel: %d", info->tx_channel);
    ESP_LOGI(TAG, "RMT RX channel: %d", info->rx_channel);
#endif

    rmt_config_t rmt_tx = RMT_DEFAULT_CONFIG_TX(gpio_num, tx_channel);
    rmt_tx.tx_config.idle_level = 1;
    rmt_tx.tx_config.idle_output_en = true;
    if (rmt_config(&rmt_tx) == ESP_OK)
    {
        if (rmt_driver_install(rmt_tx.channel, 0, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED) == ESP_OK)
        {
            rmt_config_t rmt_rx = RMT_DEFAULT_CONFIG_RX(gpio_num, rx_channel);
            rmt_rx.rx_config.idle_threshold = OW_DURATION_RX_IDLE;
            if (rmt_config(&rmt_rx) == ESP_OK)
            {
                if (rmt_driver_install(rmt_rx.channel, 512, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED) == ESP_OK)
                {
                    rmt_get_ringbuf_handle(info->rx_channel, &info->rb);
                    status = OWB_STATUS_OK;
                }
                else
                {
                    ESP_LOGE(TAG, "failed to install rx driver");
                }
            }
            else
            {
                status = OWB_STATUS_HW_ERROR;
                ESP_LOGE(TAG, "failed to configure rx, uninstalling rmt driver on tx channel");
                rmt_driver_uninstall(rmt_tx.channel);
            }
        }
        else
        {
            ESP_LOGE(TAG, "failed to install tx driver");
        }
    }
    else
    {
        ESP_LOGE(TAG, "failed to configure tx");
    }
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio_num], PIN_FUNC_GPIO);
    gpio_set_direction(gpio_num, GPIO_MODE_INPUT_OUTPUT_OD);
    gpio_matrix_out(gpio_num, RMT_SIG_OUT0_IDX + tx_channel, 0, 0);
    gpio_matrix_in(gpio_num, RMT_SIG_IN0_IDX + rx_channel, 0);
    return status;
}
```

please help test it, thanks !!

iCarletto, 2020-05-12:
@koobest thanks for the help, Well done!
I've just tested your `_init` function with the master branch over a Maxim DS2401P and a DS28E07, both standard and overdrive speed. I used GPIO 4, parasitic power and a 2.2k pullup resistor.

Here is the result:
```
Find devices:
search
  0 : ee00001b44276401
search
  1 : 8b0000221fb8ec2d
Found 2 devices
```

Your function is of course cleaner because of the use of default TX/RX configs, but especially in the last part where the GPIO is configured. However, I have to admit that I did not have time to analyze the differences, they both work but I don't know which one is more correct.

In the end, the reason why @DavidAntliff 's code fails is one, not having cleared the flags. It is sufficient to add this line to the original code and it works:
`rmt_tx.flags = 0;`

Only God knows why the code works by just enabling the log `Debug` level, maybe some shared registries.. Anyway, the important is that now it works

DavidAntliff, 2020-05-12:
@koobest & @iCarletto thank you for looking into this in more detail. The uninitialised structs in `_init` are definitely unwise and I'll sort them out properly. The changes proposed for the GPIO are probably not critical - the end result appears to be the same, from what I can tell. However @koobest's approach is a little cleaner so I'll test it and look at integrating it.

DavidAntliff, 2020-05-14:
I've incorporated some of these changes here: https://github.com/DavidAntliff/esp32-ds18b20-example/pull/20 - once the travis tests pass I'll merge it.

Note that I'm keeping the changes very simple because the (useful) macros like `RMT_DEFAULT_CONFIG_TX` aren't available in IDF v3.3.

I've tested it on a single DS18B20 device with IDF v4.2-dev () and v3.3 and it is working for me in both powered and parasitic-powered mode.

So this looks like a classic case of uninitialised struct fields causing undefined behaviour. Thank you to everyone who helped track it down.

closed 2020-05-14 by DavidAntliff