espressif/esp-idf #5238: [ESP32-S2] How do I use the I2S to control a 8-bit parallel LCD?

csobrinho, 2020-05-04:
## Environment

- Development Kit:      [ESP32-S2-SAOLA]
- Kit version (for WroverKit/PicoKit/DevKitC): [??]
- Module or chip used:  [ESP32-S2]
- IDF version (run ``git describe --tags`` to find it):
    // v4.2-dev-1126-gd85d3d969
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32s2-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [macOS]
- Using an IDE?: [Yes, CLion]
- Power Supply:         [USB]

## Problem Description

I'm trying to use the I2S to control a RM68090 8-bit parallel LCD with the ESP32-S2 based on the existing code that I already have for the ESP32. My code is uses most of the [esp-iot-solution lcd_common](https://github.com/espressif/esp-iot-solution/tree/master/components/i2s_devices/lcd_common) example [with some custom additions/fixes](https://github.com/csobrinho/hydroponics/blob/master/main/driver/i2s_lcd8.c).

I see the RS toggling but no WR toggling. If I try to use the FIFO the same happens.
This is my config:

```
.i2s_lcd_conf = {
        .data_width = 8,
        .data_io_num = {
                GPIO_NUM_2, GPIO_NUM_1, GPIO_NUM_8, GPIO_NUM_7,
                GPIO_NUM_6, GPIO_NUM_5, GPIO_NUM_4, GPIO_NUM_3,
        },
        .ws_io_num = GPIO_NUM_38,
        .rs_io_num = GPIO_NUM_45,
},
.i2s_port = I2S_NUM_0,
```

I've tried to search the [docs](esp32-s2_technical_reference_manual_en.pdf) but not a lot of info about LCD mode. From what I can see on `Table 102: Peripherals and FIFO Address`, the FIFO location changed:

```
#define I2S0_FIFO_ADD 0x6000f000
```

to
```
#define I2S0_FIFO_ADD 0x6000f004
```

I tested with and without this change and I still can't use the DMA and/or the FIFO and that's why I'm requesting some support. I remember reading on previous tickets [[esp-iot-solution/issues/71]](https://github.com/espressif/esp-iot-solution/issues/71) and [[esp-iot-solution/issues/19]](https://github.com/espressif/esp-iot-solution/issues/19#issuecomment-445348857) that 8 bit was rather cumbersome on the ESP32 but was improved for the ESP32-S2 (remember reading something about this on the forum).

Thank you!

koobest, 2020-05-06:
[esp32-s2-parallel-master.tar.gz](https://github.com/espressif/esp-idf/files/4585095/esp32-s2-parallel-master.tar.gz)

Hi, There is no I2S LCD supports for ESP32-S2 chip in IOT-solution currently, but the example in the attachment is how to configure the I2S LCD mode for the ESP32-S2 chip.

koobest, 2020-05-12:
Hi, @csobrinho 
Have you tried the example code?

csobrinho, 2020-05-14:
Yes, my code is based on the example code for the ESP32 but doesn't work
with the ESP32-S2. I even checked the output with a logic analyzer and I
can't see any change on the pins except the RS pin.

koobest, 2020-05-14:
Well, you may have misunderstood what I meant. I mean the compression package I sent you (esp32-s2-parallel-master.tar.gz) . Have you tried it?

csobrinho, 2020-05-19:
Will try it today and post results tomorrow. Thanks!

csobrinho, 2020-05-20:
Hi @koobest, I tried this adapted version but only saw the `WS` toggling, not the `D0-D7` lines.

I adapted the current example for 8 bits instead of 24 but also tried the 24 bits version with same result.

Please find attached the modified version (some headers fixes and different data pins).
[esp32-s2-parallel8-master.tar.gz](https://github.com/espressif/esp-idf/files/4654546/esp32-s2-parallel8-master.tar.gz)

Here's the logic analyzer: [image]

And the log output:
```c
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_system.h"
#include "i2s_parallel.h"

/*   I2S i2s_parallel interface configure part  */

#define  CLK  GPIO_NUM_38

#define  D0  GPIO_NUM_2
#define  D1  GPIO_NUM_1
#define  D2  GPIO_NUM_8
#define  D3  GPIO_NUM_7
#define  D4  GPIO_NUM_6
#define  D5  GPIO_NUM_5
#define  D6  GPIO_NUM_4
#define  D7  GPIO_NUM_3

#define  D8   GPIO_NUM_9
#define  D9   GPIO_NUM_10
#define  D10  GPIO_NUM_11
#define  D11  GPIO_NUM_12
#define  D12  GPIO_NUM_13
#define  D13  GPIO_NUM_14
#define  D14  GPIO_NUM_15
#define  D15  GPIO_NUM_16

#define  D16   GPIO_NUM_17
#define  D17   GPIO_NUM_18
#define  D18  GPIO_NUM_19
#define  D19  GPIO_NUM_20
#define  D20  GPIO_NUM_21
#define  D21  GPIO_NUM_26
#define  D22  GPIO_NUM_27
#define  D23  GPIO_NUM_28

i2s_parallel_pin_config_t i2s_parallel_bus = {
    .bit_width = 8,
    .pin_clk = CLK,
    .data = {D0, D1, D2, D3, D4, D5, D6, D7}, // D8, D9, D10, D11, D12, D13, D14, D15, D16, D17, D18, D19, D20, D21, D22, D23},
};

typedef struct {
    uint8_t bits_7_0;
    // uint8_t bits_15_8;
    // uint8_t bits_23_16;
} parallel_data_t;

[... 97 lines cut ...]
0x40080020: _stext at ??:?

I (191) boot: Loaded app from partition at offset 0x10000
I (191) boot: Disabling RNG early entropy source...
I (191) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (199) cpu_start: Pro cpu up.
I (203) cpu_start: Application information:
I (208) cpu_start: Project name:     parallel
I (213) cpu_start: App version:      1
I (217) cpu_start: Compile time:     May 19 2020 20:57:49
I (223) cpu_start: ELF file SHA256:  96cc017c371dbeb0...
I (229) cpu_start: ESP-IDF:          v4.2-dev-1575-ga3520970f
I (235) cpu_start: Single core mode
I (240) heap_init: Initializing. RAM available for dynamic allocation:
I (247) heap_init: At 3FFC0F48 len 0003B0B8 (236 KiB): DRAM
I (253) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (259) cpu_start: Pro cpu start user code
I (319) spi_flash: detected chip: generic
I (319) spi_flash: flash io: dio
W (319) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (330) cpu_start: Starting scheduler on PRO CPU.
Hello world!
--------I2S version  19052500, bit_width: 8

```

koobest, 2020-05-22:
Hi, @csobrinho

Sorry, the GPIO configuration we gave is incorrect.

8bit mode: I2S0O_DATA_OUT16_IDX ~I2S0O_DATA_OUT23_IDX
16bit mode: I2S0O_DATA_OUT8_IDX ~I2S0O_DATA_OUT23_IDX
24bit mode: I2S0O_DATA_OUT0_IDX ~I2S0O_DATA_OUT23_IDX

please use this one:
```
static void i2s_parallel_set_pin(void)
{
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[i2s_parallel_bus.pin_clk], PIN_FUNC_GPIO);
    gpio_set_direction(i2s_parallel_bus.pin_clk, GPIO_MODE_OUTPUT);
    gpio_set_pull_mode(i2s_parallel_bus.pin_clk,GPIO_PULLUP_ONLY);
    gpio_matrix_out(i2s_parallel_bus.pin_clk, I2S0O_WS_OUT_IDX, true, 0);

    for(int i = 0; i < i2s_parallel_bus.bit_width; i++) {
        PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[i2s_parallel_bus.data[i]], PIN_FUNC_GPIO);
        gpio_set_direction(i2s_parallel_bus.data[i], GPIO_MODE_OUTPUT);
        gpio_set_pull_mode(i2s_parallel_bus.data[i],GPIO_PULLUP_ONLY);
        gpio_matrix_out(i2s_parallel_bus.data[i], I2S0O_DATA_OUT16_IDX + i, false, 0);
    }
} 
```

koobest, 2020-05-27:
@csobrinho 
Have you solved this issue?
thanks !!

csobrinho, 2020-05-29:
Yes, I believe so. It seems to be working but somehow the framerate doesn't seem that good. I'll capture some logic analyzer timings and post here.

One of the things I had to do was swap the low/high byte to make sure the order was good. Is there anything on the hardware that could do this?

My main code is here:
https://github.com/csobrinho/hydroponics/blob/master/components/i2s_parallel/i2s_parallel.c

I also added a new parameter ([buffer policy](https://github.com/csobrinho/hydroponics/blob/master/components/i2s_parallel/i2s_parallel.c#L148-L150)) to the 
```
static inline void i2s_parallel_dma_write(const uint8_t *buf, size_t length, buffer_policy_t policy)
```

So that you can reuse the same buffer. This is very useful when clearing a screen with a specific color. With this change, you can set a small buffer with the color and just setup the necessary `lldesc_t` structures until you reach the desirable `length`.

koobest, 2020-05-29:
The hardware don't support this feature.

csobrinho, 2020-06-03:
Sorry, haven't been able to capture the logic analyser timings. 

On a different note, I saw in the ESP32-S2 datasheet that there is some _magical support for serial 8080 LCD interface using the SPI 2_.

Would this allow me to add a shift register to the SPI port and have WR (and optionally R/D and CS) to toggle and latch the output? That and FIFO (for small register writes) and DMA (for big fills) support would be a killer option to save some precious GPIOs.

**Thanks for your help dealing with this, it has completely unblocked and helped me a lot!**

koobest, 2020-06-03:
In SPI 8080 LCD mode, the data bus is 8 bits wide. Therefore, we cannot control the WR signal through the shift register. But in the CMD phase and DATA phase, we can control the CD signal by setting or clearing the registers `SPI2.misc.cd_cmd_set` and `SPI2.misc.cd_data_set`

csobrinho, 2020-06-03:
Bummer. I was planning to convert the SPI serial data into parallel using a shift register but I need a good way to signal the 8 bit shift is done, latch the data and toggle WR. If I can't do this with the hardware I might need to add:
a) some sort of decade counter attached to the SPI clock, when 8 is reached, reset and toggle wr.
b) some sort of delay line attached to the SPI clk to toggle WR after some ns have passed and then auto reset itself.
c) some other alternative like an SPI LCD but I already have four displays here ready to use.

Any suggestions? Thanks!

koobest, 2020-06-03:
Sorry,  I don't understand why you need convert the SPI serial data into parallel. 
. what is your LCD interface, SPI(serial) or 8080(paral 8bit)? 
can you provide your LCD datasheet?

csobrinho, 2020-06-03:
The LCD is parallel 8080 (RM68090) but I wanted to save some precious GPIOs so was thinking of using the SPI connected to a shift register to convert to parallel and connect the LCD D0-7 to the shift register. The problem is identifying when the 8th clock is shifted and trigger the WR.

csobrinho, 2020-06-03:
If I use the SPI at 40 MHz, I could (at most) shift about 32fps (40M / 8 / (240*320*2))

Alvin1Zhang, 2020-06-10:
Thanks for reporting and updates. Feel free to reopen if the issue still happens. Thanks.

closed 2020-06-10 by Alvin1Zhang

awoyzc, 2021-05-25:
I tested parallel 8bit 16bit and 24bit, 8bit&16bit worked, but 24bit does not work, who has tested parallel 24bit?