espressif/esp-idf #5240: Issues with OTA in esp32

sara-patel, 2020-05-04:
hello, I want to update the esp32 firmware via OTA and initially I tried with the example given ; ESP_IDF\examples\system\ota\simple_ota_example so for that I will be updtaing the frmware with hello-world.bin, but when I am doing the firmware I am getting the following error message:
 (4677) simple_ota_example: Starting OTA example
I (6527) esp_https_ota: Starting OTA...
I (6527) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
I (9667) example_connect: Got IPv6 event!
D (13637) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128
I (13637) esp_https_ota: Connection closed
I (13637) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x05574 ( 21876) map
E (13657) esp_image: invalid segment length 0x3fffff00
E (13657) simple_ota_example: Firmware upgrade failed

Kindly help me to solve the above errors. I tried with few solutions available on the ESP forum but still I am getting the same error,.

shubhamkulkarni97, 2020-05-06:
Hi @sara-patel ,

Can you share version of IDF? (by running  ``git describe --tags``)

Can you share more information about the server used to perform OTA update, is it chunked or non-chunked?

sara-patel, 2020-05-06:
Hello Shubham,
my git version is as : 
Python 3.7.5
Using Git in C:\Program Files\Git\cmd\
git version 2.21.0.windows.1
 And m very new to all this so do not know how to find out whether my server used is chunked or non-chunked.Can you please help with it?

shubhamkulkarni97, 2020-05-06:
> And m very new to all this so do not know how to find out whether my server used is chunked or non-chunked.Can you please help with it?

To get started, you can try [esp-jumpstart](https://docs.espressif.com/projects/esp-jumpstart/en/latest/introduction.html). Specifically for OTA, you can try [this project](https://github.com/espressif/esp-jumpstart/tree/master/6_ota) in esp-jumpstart

sara-patel, 2020-05-06:
ok, thanks, shubham for your quick response. Will try this and will get back.

sara-patel, 2020-05-07:
I am not able to run the above example.  It gives the following error:
CMake Error in main/CMakeLists.txt:
  Cannot find source file:

    D:/white_lion_ir/ESP_IDF/esp-jumpstart/6_ota/main/cloud_cfg/server.cert

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error in main/CMakeLists.txt:
  Cannot find source file:

    D:/white_lion_ir/ESP_IDF/esp-jumpstart/6_ota/main/cloud_cfg/device.cert

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error in main/CMakeLists.txt:
  Cannot find source file:

    D:/white_lion_ir/ESP_IDF/esp-jumpstart/6_ota/main/cloud_cfg/device.key

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error in main/CMakeLists.txt:
  Cannot find source file:

    D:/white_lion_ir/ESP_IDF/esp-jumpstart/6_ota/main/cloud_cfg/deviceid.txt

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error in main/CMakeLists.txt:
  Cannot find source file:

    D:/white_lion_ir/ESP_IDF/esp-jumpstart/6_ota/main/cloud_cfg/endpoint.txt

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

-- Generating done
-- Build files have been written to: D:/white_lion_ir/ESP_IDF/esp-jumpstart/6_ota/build
cmake failed with exit code 1

shubhamkulkarni97, 2020-05-07:
You need to create credentials for your device as mentioned in the [documentation](https://docs.espressif.com/projects/esp-jumpstart/en/latest/remotecontrol.html).

Once the credentials are created, you need to add the files under `esp-jumpstart/6_ota/main/cloud_cfg`.

sara-patel, 2020-05-08:
thanks, shubham. I tried but still facing issues with some errors of missing include files.I  would be really grateful if you suggest ant solution for the errors which I used to get when running ESP_IDF\examples\system\ota\simple_ota_example

shubhamkulkarni97, 2020-05-14:
Which server did you use to perform OTA? Can you share firmware image you used for OTA?

sara-patel, 2020-05-14:
server was my local http server and firmware was hello-world.bin that i used for OTA

shubhamkulkarni97, 2020-05-18:
The OTA example is tested and functional.
Please follow the steps mentioned in the [documentation](https://github.com/espressif/esp-idf/tree/master/examples/system/ota) to successfully run the example.

sara-patel, 2020-05-19:
ok. Thanks.

shubhamkulkarni97, 2020-05-26:
@sara-patel Is your issue resolved?

sara-patel, 2020-05-27:
hello, no the problem is still there but due to some other work urgency I was not able to work on the problem .Will get back once I start reworking on it

shubhamkulkarni97, 2020-06-05:
@sara-patel,

We have found that there are issue in `OpenSSL` server when used with Windows command prompt. Server drops the connection before file transfer is completed.

To perform OTA, you can consider any of the following options to host firmware image:

* Host the file on GitHub and add the following [certificate](https://github.com/espressif/esp-jumpstart/blob/master/6_ota/main/cloud_cfg/github_server.cert) in OTA Example.

* Run a non secure HTTP server in the directory having firmware image. Can consider running `python -m SimpleHTTPServer`. Use `http://<host-ip>:8000/<firmware-name>` as firmware upgrade URL.

* Use `secure_server.py` script attached below to run python HTTPS server. Run `python secure_server.py` and use `https://<host-ip>:8000/<firmware-name>` as firmware upgrade URL. It should work with default certificate provided in ESP-IDF.
**NOTE:** `secure_server.py` script should be placed in same directory having firmware images.
[secure_server.py.zip](https://github.com/espressif/esp-idf/files/4734316/secure_server.py.zip)

Please check if suggestions work for you.

sara-patel, 2020-06-05:
Thank you so much Shubham for your response and will get back to you once I try out OTA the way suggested by you.

shubhamkulkarni97, 2020-07-10:
@sara-patel Any updates on the issue?

mahavirj, 2020-07-14:
@sara-patel Please feel free to re-open if suggestions provided earlier do not work or still facing any issues.

closed 2020-07-14 by mahavirj

sara-patel, 2020-07-14:
still no updates from my end. And will surely update if any doubt or concern regarding the issue