espressif/esp-idf #5241: Compatability existing partitions V3.3 vs V4.0

vanBassum, 2020-05-04:
Hello,

We do have multiple devices in the field that are running on V3.3. This software uses a few partitions to store settings ect. Now I want to upgrade to V4.0 but when I update the devices they enter in a bootloop because of a hard fault.

The update works as following:
We write the new firmware to the OTA partition that is not in use. Then we set that ota partition as the running partition and reset the device.

Are the partitions compatible between V3.3 and V4.0?
This does work perfectly when updating firmware that has been build upon the same V3.3 framework.

`I (1813) spi_flash: detected chip: gd
I (1816) spi_flash: flash io: dio
I (1820) esp_core_dump_flash: Init core dump to flash
E (1835) esp_core_dump_flash: No core dump partition found!
E (1844) esp_core_dump_common: No core dump partition found!
I (1845) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1851) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
assertion "iterator != NULL" failed: file "C:/esp-idf/components/spi_flash/partition.c", line 228, function: esp_partition_get
abort() was called at PC 0x4018652b on core 0
0x4018652b: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: 6e0d99906fdfe0d33d597f2a4f9fb5039251806fe309841c6229c5eaec123591

Backtrace: 0x4009376d:0x3ffc0ed0 0x40093b81:0x3ffc0ef0 0x4018652b:0x3ffc0f10 0x40110f39:0x3ffc0f40 0x400dc5c8:0x3ffc0f60 0x400d2b50:0x3ffc1270 0x40097285:0x3ffc1290
0x4009376d: invoke_abort at C:/esp-idf/components/esp32/panic.c:155

0x40093b81: abort at C:/esp-idf/components/esp32/panic.c:172

0x4018652b: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x40110f39: esp_partition_get at C:/esp-idf/components/spi_flash/partition.c:228 (discriminator 1)

0x400dc5c8: app_main at C:\Users\bas\eclipse-workspace\KC240_Gateway\build/../main/main.cpp:234

0x400d2b50: main_task at C:/esp-idf/components/esp32/cpu_start.c:554

0x40097285: vPortTaskWrapper at C:/esp-idf/components/freertos/port.c:143

E (1897) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x6522df69 instead of 0x0
Rebooting...`

projectgus, 2020-05-07:
Hi @vanBassum ,

> Are the partitions compatible between V3.3 and V4.0?

Yes, this upgrade should work.

In the log snippet, it looks like something is going wrong reading the partition table from the app - the call to `esp_partition_get` is crashing because it can't find the specified partition, and the core dump partition is also not found ` esp_core_dump_flash: No core dump partition found!`

One possibility is that something in the OTA update process is overwriting the partition table. Does the bootloader log the partition table on boot? (It does this by default but if the log level is changed then it might not.) If yes, does it log the same output when booting the working V3.3 firmware compared to the not-working V4.0 firmware?

The other possibility is that the V4.0 app is compiled with a different [partition table offset configuration setting](https://docs.espressif.com/projects/esp-idf/en/v4.0/api-reference/kconfig.html#config-partition-table-offset) compared to the bootloader and the working apps. This means the bootloader can boot the app, but then the app will fail to find any partitions when it runs.

If the V4.0 app's sdkconfig was originally a working V3.3 sdkconfig then it should keep the same partition table offset, but if the sdkconfig was created indepedently then maybe these values have diverged?

vanBassum, 2020-05-07:
Hello,

The problem was indeed the offset of the partition table. These where different between the 2 projects. I changed this and now it is working fine. There where some other settings wrong like the maximum filename length. It's probably a good idea to check these files to prevent future problems.

Thank you for taking the time to reply to my question. 

Regards,
Bas Visscher

closed 2020-05-07 by vanBassum