espressif/esp-idf #5242: idf.py does not recognize set-target command

YaniManster, 2020-05-04:
Hi!
I'm new to ESP development, and started following the [getting started guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html)
I installed everything using the ESP-IDF Tools Installer (esp-idf-tools-setup-2.3.exe) and made only two modifications during installation: set C:\ESP as target folder and C:\esp-idf as repository folder.
After the installation completed, i created a directory D:\_Projects\ESP and copied the hello_world example there. Started the ESP-IDF Command Prompt from the start menu and changed to the hello_world directory, where tried to issue the command: idf.py set-target esp32 which failed. (Attached the screenshot.) By the way, if i skip this one step, the project builds successfully, but i wouldn't dare flash it in fear of bricking the kit.
I have ESP32-DevKitC v4 with ESP32-SOLO-1.
Could someone please help, how to set the target?
[image]

igrr, 2020-05-04:
Hi @YaniManster, you are using ESP-IDF v4.0, but the documentation you have linked to is for the "latest" (i.e. master branch) version. Please use this link for v4.0 documentation: https://docs.espressif.com/projects/esp-idf/en/release-v4.0/get-started/index.html.

`idf.py set-target` command doesn't work in IDF v4.0 because it was introduced only in v4.1. If you are building a project for ESP32 chip, there is no need to use the set-target command — the project will be built for the ESP32 by default. `set-target` is only necessary in v4.1 and later to switch to other chips (like ESP32-S2).

YaniManster, 2020-05-05:
Hi @igrr, Thank you for the quick reply. I installed the ESP-IDF following links in that document, which installed v4.0 for me. I think the maintainer should update the links or the installer as other total newbies also may follow this same path as i did.

Please forgive me for hijacking the opportunity, and allow me two additional questions:
1) Do you think i should install v4.1, or is it ok to use 4.0 for now? I plan on developing applications using WiFi and BLE meshes.
2) Did you happen to mess with Eddystone? If yes, then could you please point me to some documentation about how to implement an Eddystone scanner with ESP32-SOLO-1 without Arduino? I can't seem to find anything using native ESP-IDF project but wouldn't want to deal with Arduino.

Anyways, thanks again

igrr, 2020-05-05:
@YaniManster if you have used the IDF Tools Installer on Windows, there was a step where you can choose the version to install. The documentation for the master branch and for release/v4.0 both point to the same installer version, and you can choose in the installer which version you need.

More information about versions, including some considerations why to choose one version or another, can be found in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html. `v4.0` is already released, while `release/v4.1` branch is in beta phase at the moment. If you are only starting your project development, it is recommended to use the more recent version, as that will save you effort of upgrading later on. Both Wi-Fi and BLE mesh features are present in IDF v4.0 and v4.1.

GazTurtle, 2020-06-27:
Hi YaniManster, thanks for posting this, I made exactly the same mistake and ended up here too.

closed 2020-07-24 by igrr

jjaronetwork, 2021-01-03:
Hilarious