espressif/esp-idf #5245: Using lwIP DHCPv6 on esp_netifs

xkevin190, 2020-05-05:
lwIP supports DCHPv6, so in theory this could've enabled, I need it generally to set a prefix for downstream hosts, so IPv6 addresses can/could be configured.

A way to enable DHCPv6 and configure it (setting the downstream prefix for connected hosts, etc.).

**See also**
- [lwIP: DCHPv6](http://www.nongnu.org/lwip/2_1_x/group__dhcp6.html)

luisan00, 2020-05-05:
This could be a great feature :+1:

david-cermak, 2020-05-06:
Thanks for posting this request!

I think the DHCPv6 in lwIP is fairly limited, supports only stateless configuration, which is more/less functional even without the DHCPv6. What would be the actual advantage of using DHCPv6 in your use case?

> I need it generally to set a prefix

This should be doable with the default [IPv6 autoconfig](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-ipv6-autoconfig), while running some kind of RA. For example with [radvd](https://en.wikipedia.org/wiki/Radvd) settings
```
    interface en0
    {
        AdvSendAdvert on;
        ...
        prefix ${your_prefix}::/64
        {
              ....
        };
    };
```
should work to assign arbitrary prefix for esp hosts.

david-cermak, 2021-08-30:
@luisan00 @xkevin190 Any update from your side? As said above, users should be able to run IPv6 auto-configuration without any limitation and even with attaching DNS info enabling [`CONFIG_LWIP_IPV6_RDNSS_MAX_DNS_SERVERS`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-ipv6-rdnss-max-dns-servers).

the lwip's DHCPv6 might be helpful for some corner cases too, in IPv6 only networks, typically for setting DNS servers in the stateless configuration. I am planning on making the `LWIP_IPV6_DHCP6` a new config option and explaining this specific usecase in the [lwip docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html) to close this issue.

closed 2021-09-07 by espressif-bot