espressif/esp-idf #5247: Got ESP_ERR_NVS_NOT_ENOUGH_SPACE even when NVS shows a lot of free_entries

AxelLin, 2020-05-06:
## Environment

- Module or chip used: ESP32-WROOM-32
- IDF version: v4.2-dev-1395-gce2a99dc2353
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: NO
- Power Supply: USB

## Problem Description

free_entries/used_entries/total_entries shows 597/537/1134.
But when I test saving a long string (a certificate with 3.2KB), I got ESP_ERR_NVS_NOT_ENOUGH_SPACE.
This looks like a bug because I only use 47% of the space.

In my application, I have a lot of small strings and a few long strings (like certificate/key).
I got ESP_ERR_NVS_NOT_ENOUGH_SPACE when writing a certificate string.
The device has been tested for quite long time so it is indeed read/write for many times.
But it looks like the free_entries/used_entries numbers do not help.
Any API to figure out what's going on?

### Expected Behavior

Writing a certificate with < 4000 length should work.
And if it does not work, had better provide information about what's going on.

In additional, how to know how many read/write cycle is allow on my NVS partition.

AxelLin, 2020-05-06:
I'm wondering if this is a fragment problem, so even nvs has free space it still
cannot save a long string.
Does nvs_erase_all() help to avoid fragment problem?
(I mean if I use nvs_erase_all() to clear all settings and set all settings back)

0xjakob, 2020-05-07:
@AxelLin Hi. Yes, indeed, it can be a an inherent problem with fragmentation in the nvs pages. Strings are written into one page only (Hence the theoretical max size of 4000 bytes: page_size[4096] - header[32] - entry_states[32] - initial_string_entry[32]). 

Upon writing, the storage tries to request a page, if the item doesn't fit into the current page: https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/src/nvs_storage.cpp#L356.
If there is only one free page left, the entries of the page with the least used entries are copied to the current free page, freeing another page: https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/src/nvs_pagemanager.cpp#L179. One spare free spare page is always required. The current free page with the copied items is then used to write the item. If the item still doesn't fit, then `ESP_ERR_NVS_NOT_ENOUGH_SPACE` is returned: https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/src/nvs_storage.cpp#L363.

If we assume that the used entries are evenly distributed over all your nvs pages, then a write string operation with 3.2KB is guaranteed to fail (3.2KB > 47% of page [4KB]). Even if they aren't distributed evenly, the operation is still likely to fail. 

One quick workaround would be to use blobs. They can be divided into multiple chunks over multiple pages and hence should be less prone to fragmentation.

However, it is probably best to save certificate and private key in a separate raw partition and then use the `esp_partition*` api to read/write from it, possibly with transparent encryption. Having the data in a partition also makes it possible to map it into memory, hence saving a few KBs of your DRAM.

BTW: do you use the certificates for signing/authentication? Then you might be interested in the upcoming digital signature feature of the ESP32-S2, if that chip is an option: https://github.com/espressif/esp-idf/blob/master/components/esp32s2/include/esp_ds.h.

closed 2020-05-18 by AxelLin

JamieDriver, 2022-02-23:
Is this not also related to the 'must be one free page' requirement ?  ie. the one used internally for defrag/consolidation purposes ?
eg. I can enter blob data into my 16kb nvs, until I get to ~360/504 entries used, then it starts to error with ESP_ERR_NVS_NOT_ENOUGH_SPACE.

Debugging suggests I have 3 full pages, and one empty (reserved) page that I cannot use - so in my case ~75% full (by `nvs_get_stats()` values) is actually full to capacity, for all practical purposes  ?

igrr, 2022-02-23:
@JamieDriver  That is a good point, we should exclude the space in that auxiliary page when calculating available size.

JamieDriver, 2022-02-23:
Or maybe the `nvs_stats_t` struct needs an extra data member, for 'reserved_entries', or similar ?  

Also, this feature (ie that one page is reserved) didn't seem to be documented on https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/storage/nvs_flash.html.

Thanks for your swift response!

0xjakob, 2022-02-24:
@JamieDriver 
> Is this not also related to the 'must be one free page' requirement ?

Not exactly. The issue here is that strings need consecutive space. If you have a large NVS partition but still the same relative fill rate, then the issue still persists, while an additional free page more or less becomes insignificant.

However, the general suggestions and ideas you and @igrr mentioned are still valid.