espressif/esp-idf #5248: ESP32 Wrover Device Crashing When Using ESP-MQTT

mntolia, 2020-05-06:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32
- IDF version : 4
- Build System:  CMake
- Operating System:     Linux
- Using an IDE?: PlatformIO on Visual Studio Code
- Power Supply: USB

I am using Arduino as an ESP-IDF Component

## Problem Description

The ESP32 device crashes when it runs esp_mqtt_client_init() function

### Code to reproduce this issue

```cpp
/* WiFi scan Example
   This example code is in the Public Domain (or CC0 licensed, at your option.)
   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "sdkconfig.h"
#include "mqtt_client.h"
#include <Arduino.h>
#include <WiFi.h>
#include "esp_log.h"
#include "esp_event.h"
#include "sdkconfig.h"

const char *ssid = "1233";
const char *password = ",,,";

void setup()
{
    Serial.begin(115200);
    //WiFi Login
    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(1000);
        Serial.println("Establishing connection to WiFi..");
    }

    Serial.println("Connected to network");

    // MQTT Setup
    Serial.println("Connecting to MQTT Broker");
    esp_mqtt_client_config_t mqtt_config;
    mqtt_config.uri = "mqtt://broker.aliv.com.sg";
    mqtt_config.username = "Test_user";
    mqtt_config.password = "=3jSbtjRiE8R";
    mqtt_config.client_id = "esp32";
    Serial.println("Initializing Client");
    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_config);
    Serial.println("Registering Callback");
    Serial.println("Connecting");
    esp_mqtt_client_start(client);
}

void loop()
{    
}

```

## Debug Logs

```
␀␀Establishing connection to WiFi..
␛[0;32mI (2166) tcpip_adapter: sta ip: 192.168.1.126, mask: 255.255.255.0, gw: 192.168.1.1␛[0m
Establishing connection to WiFi..
Connected to network
Connecting to MQTT Broker
Initializing Client
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060a30  A0      : 0x80001459  A1      : 0x3ffbf590  
A2      : 0xa5a5a5a5  A3      : 0xa5a5a5a1  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x80092e6e  A9      : 0x3ffbf560  
A10     : 0x3ffbf7c0  A11     : 0x00001800  A12     : 0x3ffae720  A13     : 0x00000000  
A14     : 0x00060e23  A15     : 0x3ffc2180  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xa5a5a5a5  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: b115abeb711f7ee246687759dee00f73e823f322d7bbbadd52c96e866cf19dbc

Backtrace: 0x400014d9:0x3ffbf590 |<-CORRUPTED

Rebooting...
```

## Other items if possible

https://gist.github.com/mntolia/2cf2576b3c9ae3c6bd82b189c82dcd54

david-cermak, 2020-05-07:
Hi @mntolia 

Most of the items in the configuration struct are mutually exclusive, others must be set to zero if the feature is disabled. In other words, the init struct must be all zero's except for just few items one configures and this construct does not assure zero init:
```
esp_mqtt_client_config_t mqtt_config;
```
You can either `memset()` to zero or initialize using the C99 designated initializers (they kindof work from C++ as well if the fields are ordered and child objects are somehow enclosed)

Can you please update this line as suggested below and check if this fixes your issue?
```
esp_mqtt_client_config_t mqtt_config = { 0 };
```

mntolia, 2020-05-07:
Hi @david-cermak ,
Thanks a lot. I got the device connected. But I am facing another issue. When I run the esp_mqtt_client_register_event() function according to this example:
https://github.com/espressif/esp-idf/blob/ce2a99dc23533f40369e4ab0d17ffd40f4b0dd72/examples/protocols/mqtt/ssl/main/app_main.c
I get this error:
`argument of type "int" is incompatible with parameter of type "esp_mqtt_event_id_t"
`
Additionally in the event handler function:
```
static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data) {
    ESP_LOGD(TAG, "Event dispatched from event loop base=%s, event_id=%d", base, event_id);
        mqtt_event_handler_cb(event_data);

}
```
I get this error:
```
invalid conversion from 'void*' to 'esp_mqtt_event_handle_t' {aka 'esp_mqtt_event_t*'} [-fpermissive]
```

If you could point out where I am wrong it would be really appreciated :)

david-cermak, 2020-05-09:
@mntolia  This is an issue with C++ build, has been fixed in https://github.com/espressif/esp-mqtt/commit/8a1e1a5a9ffce89313aaa209375e1def893a4c73, but apparently that hasn't made it yet to arduino. As a workaround you can cast the variables to the expected types or even avoid using an event loop altogether.
It is still possible to configure `mqtt_event_handler_cb()` as a plain callback in client config:
```
    mqtt_config.event_handle = mqtt_event_handler_cb;
```
And remove both definition of the `mqtt_event_handler()` and registration of the handler `esp_mqtt_client_register_event()`

mntolia, 2020-06-12:
Thanks David. I will close this ticket.

closed 2020-06-12 by mntolia