espressif/esp-idf #5249: Crashing when trying to run PPPos

ajinkyaw, 2020-05-07:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]

- Kit version (for WroverKit): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [external 5V|]

## Problem Description
I am trying to run PPPos example on ESP32 and SIm800c module, but while running the it crashes. Below is information i have copied from serial monitor.
Can anyone help me in resolving this issue. 

//Detailed problem description goes here.
I (29) boot: ESP-IDF v4.2-dev-701-g0ae960f2f-dirty 2nd stage bootloader
I (29) boot: compile time 10:58:15
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x12fd8 ( 77784) map
I (143) esp_image: segment 1: paddr=0x00023000 vaddr=0x3ffb0000 size=0x021dc (  8668) load
I (147) esp_image: segment 2: paddr=0x000251e4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /root/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1817

I (150) esp_image: segment 3: paddr=0x000255ec vaddr=0x40080400 size=0x0aa2c ( 43564) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x45040 (282688) map
0x400d0020: _stext at ??:?

I (284) esp_image: segment 5: paddr=0x00075068 vaddr=0x4008ae2c size=0x00140 (   320) load
0x4008ae2c: vEventGroupDelete at /root/esp/esp-idf/components/freertos/event_groups.c:660 (discriminator 1)

I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
I (294) cpu_start: Pro cpu up.
I (298) cpu_start: Application information:
I (302) cpu_start: Project name:     pppos_client
I (308) cpu_start: App version:      1
I (312) cpu_start: Compile time:     May  7 2020 10:58:21
I (318) cpu_start: ELF file SHA256:  202a68bca2050778...
I (324) cpu_start: ESP-IDF:          v4.2-dev-701-g0ae960f2f-dirty
I (331) cpu_start: Starting app cpu, entry point is 0x400811b0
0x400811b0: call_start_cpu1 at /root/esp/esp-idf/components/esp32/cpu_start.c:273

I (323) cpu_start: App cpu up.
I (342) heap_init: Initializing. RAM available for dynamic allocation:
I (349) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (355) heap_init: At 3FFB3B20 len 0002C4E0 (177 KiB): DRAM
I (361) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (367) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (374) heap_init: At 4008AF6C len 00015094 (84 KiB): IRAM
I (380) cpu_start: Pro cpu start user code
I (398) spi_flash: detected chip: generic
I (399) spi_flash: flash io: dio
W (399) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (409) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
ppp phase changed[1]: phase=0
I (424) uart: queue free spaces: 30
W (434) esp-modem: Rx Break
W (484) esp-modem: Rx Break
W (534) esp-modem: Rx Break
E (934) esp-modem: esp_modem_dte_send_cmd(229): process command timeout
E (934) dce_service: esp_modem_dce_sync(48): send command failed
E (934) sim800: sim800_init(460): sync failed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d40e6  PS      : 0x00060630  A0      : 0x800d24b5  A1      : 0x3ffb58c0  
0x400d40e6: app_main at /home/aziz/pppex/esp-idf-master/examples/protocols/pppos_client/main/pppos_client_main.c:260

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffbb23c  A5      : 0x3ffb9bfc  
A6      : 0x00000001  A7      : 0x00060023  A8      : 0x800d40e4  A9      : 0x3ffb5890  
A10     : 0x00000000  A11     : 0x3ffbcc80  A12     : 0x400d50c8  A13     : 0x00000000  
0x400d50c8: sim800_deinit at /home/aziz/pppex/esp-idf-master/examples/protocols/pppos_client/components/modem/src/sim800.c:428

A14     : 0x00000200  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000007c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 202a68bca2050778e5d6ff25743fe3a78bb94aad490ce1c3480a7a251cba2bf7

Backtrace: 0x400d40e3:0x3ffb58c0 0x400d24b2:0x3ffb59e0 0x4008a5fd:0x3ffb5a00
0x400d40e3: app_main at /home/aziz/pppex/esp-idf-master/examples/protocols/pppos_client/main/pppos_client_main.c:254

0x400d24b2: main_task at /root/esp/esp-idf/components/esp32/cpu_start.c:561

0x4008a5fd: vPortTaskWrapper at /root/esp/esp-idf/components/freertos/xtensa/port.c:143

david-cermak, 2020-05-07:
Hi @ajinkyaw 

Please check that your connection to the module is correct and stable. Having the `esp_modem_dce_sync()` failed, as the first serial command to the module, and also seeing the UART breaks
```
W (534) esp-modem: Rx Break
```
Seems like something is wrong with the connection. Please check the pins are connected correctly, not swapped, signals look reasonably stable, etc.

Another issue is the subsequent crash in module deinit if the initial command fails. The example should just `abort()` and not try to deinit in case of a communication failure, as it does on current master. I believe this is just a minor issue with the version of your example, since you are on a `v3.2-dev`

ajinkyaw, 2020-05-07:
@david-cermak  
HI .
After reconnecting the connections

I (29) boot: ESP-IDF v4.2-dev-701-g0ae960f2f-dirty 2nd stage bootloader
I (29) boot: compile time 10:58:15
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x12ff8 ( 77816) map
I (143) esp_image: segment 1: paddr=0x00023020 vaddr=0x3ffb0000 size=0x021dc (  8668) load
I (147) esp_image: segment 2: paddr=0x00025204 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /root/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1817

I (150) esp_image: segment 3: paddr=0x0002560c vaddr=0x40080400 size=0x0aa0c ( 43532) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x450b4 (282804) map
0x400d0020: _stext at ??:?

I (285) esp_image: segment 5: paddr=0x000750dc vaddr=0x4008ae0c size=0x00160 (   352) load
0x4008ae0c: vEventGroupDelete at /root/esp/esp-idf/components/freertos/event_groups.c:651

I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
I (294) cpu_start: Pro cpu up.
I (298) cpu_start: Application information:
I (302) cpu_start: Project name:     pppos_client
I (308) cpu_start: App version:      1
I (312) cpu_start: Compile time:     May  7 2020 10:58:21
I (318) cpu_start: ELF file SHA256:  b5532bcdb0fc3601...
I (324) cpu_start: ESP-IDF:          v4.2-dev-701-g0ae960f2f-dirty
I (331) cpu_start: Starting app cpu, entry point is 0x400811b0
0x400811b0: call_start_cpu1 at /root/esp/esp-idf/components/esp32/cpu_start.c:273

I (324) cpu_start: App cpu up.
I (342) heap_init: Initializing. RAM available for dynamic allocation:
I (349) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (355) heap_init: At 3FFB3B20 len 0002C4E0 (177 KiB): DRAM
I (361) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (367) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (374) heap_init: At 4008AF6C len 00015094 (84 KiB): IRAM
I (380) cpu_start: Pro cpu start user code
I (398) spi_flash: detected chip: generic
I (399) spi_flash: flash io: dio
W (399) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (409) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
ppp phase changed[1]: phase=0
I (424) uart: queue free spaces: 30
E (934) esp-modem: esp_modem_dte_send_cmd(229): process command timeout
E (934) dce_service: esp_modem_dce_sync(48): send command failed
E (934) sim800: sim800_init(460): sync failed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d40da  PS      : 0x00060830  A0      : 0x800d24a9  A1      : 0x3ffb58c0  
0x400d40da: app_main at /home/aziz/pppex/esp-idf-master/examples/protocols/pppos_client/main/pppos_client_main.c:260

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffbb238  A5      : 0x3ffb9c00  
A6      : 0x00000001  A7      : 0x00060023  A8      : 0x800d40d8  A9      : 0x3ffb5890  
A10     : 0x00000000  A11     : 0x3ffbcc7c  A12     : 0x400d5104  A13     : 0x00000000  
0x400d5104: sim800_deinit at /home/aziz/pppex/esp-idf-master/examples/protocols/pppos_client/components/modem/src/sim800.c:428

A14     : 0x00000200  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000007c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: b5532bcdb0fc3601defe47de18095279a168e07f66f1e38e423dde8a9aaaf621

Backtrace: 0x400d40d7:0x3ffb58c0 0x400d24a6:0x3ffb59e0 0x4008a5fd:0x3ffb5a00
0x400d40d7: app_main at /home/aziz/pppex/esp-idf-master/examples/protocols/pppos_client/main/pppos_client_main.c:254

0x400d24a6: main_task at /root/esp/esp-idf/components/esp32/cpu_start.c:561

0x4008a5fd: vPortTaskWrapper at /root/esp/esp-idf/components/freertos/xtensa/port.c:143

the lines 

E (934) esp-modem: esp_modem_dte_send_cmd(229): process command timeout
E (934) dce_service: esp_modem_dce_sync(48): send command failed
E (934) sim800: sim800_init(460): sync failed

are in red only. Can u guide me more.

david-cermak, 2020-05-09:
@ajinkyaw I'm afraid there's not much to suggest here, as it's failing in the first, initial connection to the module, ESP32 sending `AT\r` and expecting an `OK` from the module, which for some reason doesn't come. Could you for example check the lines using a logic analyser or just connect the module itself to a terminal and send the plain `AT` command and check the response?

eflukx, 2020-05-10:
Similar problem here. I have working code with PPPoS working nicely with my BG96 module (code adapted from example code), but when modem cannot be properly synced with (e.g. when not properly reset but still in ppp/data session) the fatal panic/crash occurs. This can be countered by putting the modem in the correct mode when attempting to connect, but when it fails I would expect a graceful fail not a panic an subsequent reboot. At least I would really like to know where this originates, as a `LoadProhibited` can be considered a (serious?) bug at any time.

Ps. The esp_modem (dce/dte stuff) seems to be engineered quite nicely in an extensible and generic way, but it can be a bit hard to follow sometimes (structs pointing to structs, filled with function pointers etc.). It would be very helpful to have a quick write-up (in readme.md) available on its architecture.

ajinkyaw, 2020-05-11:
@david-cermak : After your reply i tried playing directly with module with AT commands and other commands. On Serial / Terminal i am receiving proper replies on the queries i am making.

Also i double checked the connections. I have gone through the issues others have raised too, but still no luck.

david-cermak, 2020-05-20:
Have checked connection with SIM800L on `4.2-dev` (as it seems that is the actual version you used, not the v3.2 mentioned in the report) and it seems ok, could acknowledge these two issues (also mentioned above):
* sync fails if the modem is in a wrong state
* program should `abort()` instead of trying to deinit the modem

I personally find both of these *not that* serious, as the examples are here just to demonstrate the functionality, plus abort() would cause crash (and reboot with default config) anyway. Agree though that this needs to be fixed, too!

@ajinkyaw Are you able to connect at all, seeing the failure constantly, even after you reset the modem and wait till the led blinks? If this is just an intermittent error, we could just address the issues above, but from your report it seems like your connection never actually worked. 

As a side note, please note that the SIM800L is largely obsolete nowadays (G2 only) -- not working in most counties, others will stop supporting G2 soon.

Alvin1Zhang, 2020-08-18:
@ajinkyaw Thanks for reporting and updates. Would you please help share if any updates for the issue? Thanks.

mazorvanunu, 2020-09-09:
hi, facing same subject mmm at first the example worked properly (got local ip and lat&lon gps location) 
using esp32 wrover-b
changed the username\password and apn settings in menuconfig

Alvin1Zhang, 2020-12-22:
Thanks for reporting, feel free to reopen.

closed 2020-12-22 by Alvin1Zhang

Bess2546, 2024-02-17:
ไม่ได้เรื่อง

Bess2546, 2024-02-17:
Gak sus sus