espressif/esp-idf #5250: PPPos client will it support SIM7600 module?

poojavernekar, 2020-05-07:
I am trying to make use of PPPos protocol between ESP32-WROOM-32D and SIM7600CE-T module.
But I am getting run time error, when i flash the program.
Please let me know whether PPPos protocol supports SIM7600 or not.

david-cermak, 2020-05-07:
Hi @poojavernekar 

I've skimmed through the SIM7600 datasheet and AFAICT it doesn't seem to support PPP mode. Looking at the supported AT commands, there's just a socket interface. You can use the socket related AT commands with the `esp-modem` component to some extent, but would need to implement these specific commands. There's no such example in esp-idf demonstrating TCP/IP socket networking with SIM7600 or similar.

poojavernekar, 2020-05-07:
Ok thank you david,

I have worked with AT commands, by making use of uart_asys_tx_rx example program, where i passed AT commands, its working fine and I am getting responses.
Now the task is to share data using MQTT protocol,( ESP32-WROOM-32D and SIM7600 modem) 
But my problem is I dont want to use WIFI connection of ESP32, i want to share or establish internet gateway connection from SIM7600 to ESP32 module, 
So i was trying to make use of PPPos protocol.
Please suggest is there any way to achieve internet gateway connection from SIM7600 to ESP32??

poojavernekar, 2020-05-14:
Hey tried to resolve the issue by establishing gateway connection between SIM7600 and ESP32 using AT commands, below mentioned link
https://m2msupport.net/m2msupport/data-call-at-commands-to-set-up-gprsedgeumtslte-data-call/

closed 2020-08-03 by espressif-bot