espressif/esp-idf #5251: Binaries differ with same code but multiple build runs (CMake, esp as library)

shangl, 2020-05-07:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.3.2-107-g722043f73
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description
Compiled binaries differ between different builds even if nothing changed in the code. This happens at least when esp-idf is used as a library (I have not tried other configurations). 

First of all this is highly worrying. Binaries should be reproducible and should not randomly vary. Secondly, this is quite critical to us as we use an external PKI to sign the binaries for secure boot. 

### Expected Behavior
The hashes between the binaries without any code changes should be same.
### Actual Behavior
The hashes differ
### Steps to reproduce

I can reproduce this with a freshly cloned repository with the example build_system/cmake/idf_as_lib with the following commands:
cd examples/build_system/cmake/idf_as_lib
./build.sh
mv build build.bkp
rm sdkconfig
./build.sh
sha256sum build/idf_as_lib.bin build.bkp/idf_as_lib.bin 

**759e17e59ac3063a3e2ae167f48cf1f1e825bc29d9757490b0ae2df13c2e3f78**  build/idf_as_lib.bin
**f1bf34285a6975f2c2514cfb287dbdbf616624a70ea4c7b11eeddf05e7851117**  build.bkp/idf_as_lib.bin

A recursive diff between the two build folders show that even the sublibraries already differ:
diff -qr build build.bkp/
Files build/bootloader/bootloader.bin and build.bkp/bootloader/bootloader.bin differ
Files build/bootloader/bootloader.elf and build.bkp/bootloader/bootloader.elf differ
Files build/bootloader/CMakeFiles/CMakeOutput.log and build.bkp/bootloader/CMakeFiles/CMakeOutput.log differ
Files build/bootloader/esp-idf/bootloader_support/CMakeFiles/idf_component_bootloader_support.dir/src/bootloader_init.c.obj and build.bkp/bootloader/esp-idf/bootloader_support/CMakeFiles/idf_component_bootloader_support.dir/src/bootloader_init.c.obj differ
Files build/bootloader/esp-idf/bootloader_support/libbootloader_support.a and build.bkp/bootloader/esp-idf/bootloader_support/libbootloader_support.a differ
Files build/bootloader/esp-idf/efuse/libefuse.a and build.bkp/bootloader/esp-idf/efuse/libefuse.a differ
Files build/bootloader/esp-idf/log/liblog.a and build.bkp/bootloader/esp-idf/log/liblog.a differ
Files build/bootloader/esp-idf/main/libmain.a and build.bkp/bootloader/esp-idf/main/libmain.a differ
Files build/bootloader/esp-idf/micro-ecc/libmicro-ecc.a and build.bkp/bootloader/esp-idf/micro-ecc/libmicro-ecc.a differ
Files build/bootloader/esp-idf/soc/libsoc.a and build.bkp/bootloader/esp-idf/soc/libsoc.a differ
Files build/bootloader/esp-idf/spi_flash/libspi_flash.a and build.bkp/bootloader/esp-idf/spi_flash/libspi_flash.a differ
Files build/CMakeFiles/CMakeOutput.log and build.bkp/CMakeFiles/CMakeOutput.log differ
Files build/esp-idf/app_trace/libapp_trace.a and build.bkp/esp-idf/app_trace/libapp_trace.a differ
Files build/esp-idf/app_update/CMakeFiles/idf_component_app_update.dir/esp_app_desc.c.obj and build.bkp/esp-idf/app_update/CMakeFiles/idf_component_app_update.dir/esp_app_desc.c.obj differ
Files build/esp-idf/app_update/libapp_update.a and build.bkp/esp-idf/app_update/libapp_update.a differ
Files build/esp-idf/bootloader_support/libbootloader_support.a and build.bkp/esp-idf/bootloader_support/libbootloader_support.a differ
Files build/esp-idf/cxx/libcxx.a and build.bkp/esp-idf/cxx/libcxx.a differ
Files build/esp-idf/driver/libdriver.a and build.bkp/esp-idf/driver/libdriver.a differ
Files build/esp-idf/efuse/libefuse.a and build.bkp/esp-idf/efuse/libefuse.a differ
Files build/esp-idf/esp32/libesp32.a and build.bkp/esp-idf/esp32/libesp32.a differ
Files build/esp-idf/espcoredump/libespcoredump.a and build.bkp/esp-idf/espcoredump/libespcoredump.a differ
Files build/esp-idf/esp_event/libesp_event.a and build.bkp/esp-idf/esp_event/libesp_event.a differ
Files build/esp-idf/esp_ringbuf/libesp_ringbuf.a and build.bkp/esp-idf/esp_ringbuf/libesp_ringbuf.a differ
Files build/esp-idf/ethernet/libethernet.a and build.bkp/esp-idf/ethernet/libethernet.a differ
Files build/esp-idf/freertos/libfreertos.a and build.bkp/esp-idf/freertos/libfreertos.a differ
Files build/esp-idf/heap/libheap.a and build.bkp/esp-idf/heap/libheap.a differ
Files build/esp-idf/log/liblog.a and build.bkp/esp-idf/log/liblog.a differ
Files build/esp-idf/lwip/liblwip.a and build.bkp/esp-idf/lwip/liblwip.a differ
Files build/esp-idf/mbedtls/libmbedtls.a and build.bkp/esp-idf/mbedtls/libmbedtls.a differ
Files build/esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a and build.bkp/esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a differ
Files build/esp-idf/mbedtls/mbedtls/library/libmbedtls.a and build.bkp/esp-idf/mbedtls/mbedtls/library/libmbedtls.a differ
Files build/esp-idf/mbedtls/mbedtls/library/libmbedx509.a and build.bkp/esp-idf/mbedtls/mbedtls/library/libmbedx509.a differ
Files build/esp-idf/micro-ecc/libmicro-ecc.a and build.bkp/esp-idf/micro-ecc/libmicro-ecc.a differ
Files build/esp-idf/newlib/libnewlib.a and build.bkp/esp-idf/newlib/libnewlib.a differ
Files build/esp-idf/nvs_flash/libnvs_flash.a and build.bkp/esp-idf/nvs_flash/libnvs_flash.a differ
Files build/esp-idf/pthread/libpthread.a and build.bkp/esp-idf/pthread/libpthread.a differ
Files build/esp-idf/smartconfig_ack/libsmartconfig_ack.a and build.bkp/esp-idf/smartconfig_ack/libsmartconfig_ack.a differ
Files build/esp-idf/soc/libsoc.a and build.bkp/esp-idf/soc/libsoc.a differ
Files build/esp-idf/spi_flash/libspi_flash.a and build.bkp/esp-idf/spi_flash/libspi_flash.a differ
Files build/esp-idf/tcpip_adapter/libtcpip_adapter.a and build.bkp/esp-idf/tcpip_adapter/libtcpip_adapter.a differ
Files build/esp-idf/vfs/libvfs.a and build.bkp/esp-idf/vfs/libvfs.a differ
Files build/esp-idf/wpa_supplicant/libwpa_supplicant.a and build.bkp/esp-idf/wpa_supplicant/libwpa_supplicant.a differ
Files build/esp-idf/xtensa-debug-module/libxtensa-debug-module.a and build.bkp/esp-idf/xtensa-debug-module/libxtensa-debug-module.a differ
Files build/idf_as_lib.bin and build.bkp/idf_as_lib.bin differ
Files build/idf_as_lib.elf and build.bkp/idf_as_lib.elf differ

### Code to reproduce this issue

Standard example in examples/build_system/cmake/idf_as_lib without any changes.

projectgus, 2020-05-08:
Hi @shangl,

Thanks for the detailed bug report and letting us know about this. Reproducible builds has been a goal for ESP-IDF for a while and we're actually closer to this than ever in the CMake-based build system, but it's still not immediately achievable (as made clear in your report). However it should be achievable very soon.

There are a few factors involved, I'll try to break them down individually and explain what can be done about them:

---

> Files build/esp-idf/app_update/CMakeFiles/idf_component_app_update.dir/esp_app_desc.c.obj and build.bkp/esp-idf/app_update/CMakeFiles/idf_component_app_update.dir/esp_app_desc.c.obj differ
> Files build/idf_as_lib.bin and build.bkp/idf_as_lib.bin differ
> Files build/idf_as_lib.elf and build.bkp/idf_as_lib.elf differ

By default, ESP-IDF apps have the build time & date compiled into an "app description" structure which can be accessed via an API.

This can be disabled by turning off this config flag:
https://docs.espressif.com/projects/esp-idf/en/v3.3.2/api-reference/kconfig.html#config-app-compile-time-date

---

> Files build/bootloader/bootloader.bin and build.bkp/bootloader/bootloader.bin differ
> Files build/bootloader/bootloader.elf and build.bkp/bootloader/bootloader.elf differ
> Files build/bootloader/esp-idf/bootloader_support/CMakeFiles/idf_component_bootloader_support.dir/src/bootloader_init.c.obj and build.bkp/bootloader/esp-idf/bootloader_support/CMakeFiles

The bootloader logs the time that it was compiled as part of the default startup sequence.

One workaround for this is to change the [bootloader log level](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/api-reference/kconfig.html#config-log-bootloader-level) to Warning, Error or None - meaning no Info level logs are compiled in.

Another workaround is to patch the log line out in the IDF source:

```
diff --git a/components/bootloader_support/src/bootloader_init.c b/components/bootloader_support/src/bootloader_init.c
index 17d0fb161c..2a44a691ac 100644
--- a/components/bootloader_support/src/bootloader_init.c
+++ b/components/bootloader_support/src/bootloader_init.c
@@ -148,7 +148,7 @@ static esp_err_t bootloader_main()
     wdt_reset_check();
     ESP_LOGI(TAG, "ESP-IDF %s 2nd stage bootloader", IDF_VER);
 
-    ESP_LOGI(TAG, "compile time " __TIME__ );
+    //ESP_LOGI(TAG, "compile time " __TIME__ );
     ets_set_appcpu_boot_addr(0); 

 #ifdef CONFIG_BOOTLOADER_WDT_ENABLE
```

We'll add a config option so it's possible to disable this log line properly.

---

> even the sublibraries already differ

Archive files (.a) are a simple archive format which includes a modification timestamp in the header. So in most of these cases all of the `.obj` files in the archive have the same contents, and are included in the same order, but the timestamps are different.

`ar` provides a "deterministic mode" flag which sets all timestamps and other metadata to zero. CMake doesn't have native support for this, but it [looks like we can patch it in](https://github.com/libgit2/libgit2/commit/d630887bb6ab91a55e72fddc65db93ee6abcd984) without too much fuss.

For now, if all of the .obj files are the same then the actual library data should be the same as well and this shouldn't cause any problems for the final binary contents.

---

There is one other barrier to reproducible builds which is not shown in your diffs, I encountered it in about 50% of my tests just now. We recently fixed a bug on master branch (ec2e666a6ef7d9841393f5f9996472cd1df11ab1) where the linker script generated by the ESP-IDF build system contained some non-determinism in clause ordering. Ordering differences in the linker script can lead to objects being linked in a different order, which leads to different binaries. If you see two different `.ld` files in two subsequent builds, this is the likely root cause.

This fix hasn't been backported to the `release/v4.0` or `release/v3.3` branches yet, but we'll look at doing so. Once it's fixed there, with the two modifications shown above then clean builds of the same project using all the same paths, the same source files and configuration file should produce the same .elf and .bin files.

---

PS Just in case you also encounter this while doing further experiments, I ran into an unexpected behaviour in the `idf_as_lib` example starting from IDF v4.0 - the top-level CMakeLists.txt specifies that the sdkconfig file is located in the build directory, but the `build-esp32.sh` script deletes and recreates the build directory each time so the project always builds with default config. This is obviously sub-optimal, will fix the example so the sdkconfig file is located in the main project directory similar to the v3.x behaviour.

shangl, 2020-05-11:
@projectgus : thank you very much for the quick response. Do you know approximately when the fixes will be available?

projectgus, 2020-05-20:
Hi @shangl,

Sorry for the slow reply the second time around!

Regarding the linker script non-determinism, this has been backported to release/v3.3 branch in commit ccbca4570926b11083c14e3c1f24937e530a23f2 so you can use this now. The fix will be in the v3.3.3 stable bugfix release when this is available.

The release/v4.0 branch is currently frozen as v4.0.1 is about to be released, but it will be merged ASAP after the v4.0.1 release and included in the next v4.0.2 bugfix release.

Regarding the other tweaks - there is no single config option yet to get reproducible builds but if you set all the config items steps shown above (or optionally comment the one log line in the bootloader) then you can achieve the same goal. Once a single config option is added to control this then this GitHub issue will be updated.

shangl, 2020-05-28:
Hi @projectgus , thank you very much, the fixes look good.

projectgus, 2020-05-29:
Hi @shangl ,

Glad you're able to make it work. If it's OK with you then I'd prefer to keep this open until we have a documented way to achieve this.

Angus

shangl, 2020-05-29:
@projectgus : sure, no problem.

I discovered another issue and I am not sure if whether it connects to this problem, should be different ticket or is a bug at all: I can also see that when changing the flash size, also the application binary starts varying (even with all the adjustments mentioned above). Is this desired / expected? If yes, what is the reason for a different binary when just the flash size is changed in sdkconfig?

shangl, 2021-08-23:
@projectgus : it looks like this issue was fixed e.g. for version 4.2.1 but it came back in version 4.3. could this be fixed there too?

igrr, 2021-08-23:
@shangl does this issue occur in 4.3 with the same steps to reproduce as described in your original post?

Alvin1Zhang, 2021-09-09:
@shangl Thanks for reporting, would you please help share if any updates? Thanks.

igrr, 2021-12-01:
I think the issue in release/v4.3 branch will be fixed by https://github.com/espressif/esp-idf/commit/404ee09181375b39d91a8b198bba0afc43d1b21b. @shangl if you are still working on this, please try to update to that commit, let us know if the behavior is still not correct.

closed 2022-03-21 by igrr