espressif/esp-idf #5253: ULP Assembler failure

stevewms, 2020-05-08:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

igrr, 2020-05-08:
Hi @stevewms, it is a bit hard to understand from the truncated log, but it looks like the ULP assembly file is being passed to the Xtensa compiler. Could you please attach your component CMakeLists.txt file? At the same time, could you compare it to one of the ULP examples in IDF, and see if you are doing something different from the example?

stevewms, 2020-05-10:
iggr,
Thanks for your reply.  I had tried to organize my app, with about 5 components plus the main program such that the ULP code was a component also.  Using your suggestion, I moved the ULP code under the main directory in a directory called "ulp", following the example in the idf, including the way the main program's CMakeLists.txt file was set up, and now everything works.  Thanks for the pointer to a solution.
But this does bring up the question, is it possible to organize a project such that the ULP code is treated as a component?  If so, how would that work?  Are there any examples anywhere to which you could point me?
I'm ready to close this as a bug.  It's not a bug. But I would like to see any response to the questions I asked about treating a file of ULP code as a component.

igrr, 2020-05-10:
@stevewms Compared to the main application, the ULP application has a much simpler build process, where all the object files are linked directly into the ULP executable. I don't see a straightforward way to add "components" support to the current ULP build system. It is possible to build multiple ULP applications (i.e. one in each component), and choose which one to load at run time. But combining a ULP app from several components is not possible at the moment.

stevewms, 2020-05-10:
iggr,
Thanks for your reply.  It makes sense and creating multiple ULP apps for various components is something I'd not even thought of, not that I have any great need for it at this time.
Again, thanks.

closed 2020-05-10 by stevewms