espressif/esp-idf #5254: ULP wakeup

fpgamaster, 2020-05-08:
I have an ULP code which runs from normal mode and from deep sleep.

To generate an interrupt (wake up) the following code is used:
        .global wake_up
wake_up:
        /* Check if the system can be woken up */
        READ_RTC_FIELD(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_RDY_FOR_WAKEUP)
        and r0, r0, 1
        jump wake_up, eq

        /* Wake up the SoC, end program */
        wake
        /* Stop ULP timer */
        WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0)
        halt
 
BUT if the ULP code is started from normal mode the following code runs forever:
wake_up:
        /* Check if the system can be woken up */
        READ_RTC_FIELD(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_RDY_FOR_WAKEUP)
        and r0, r0, 1
        jump wake_up, eq

So, is there a way to detect whether the ULP code is running from a deep sleep or normal mode and to wake up properly?

Alvin1Zhang, 2020-05-09:
@fpgamaster Thanks for reporting this feature request, we will evaluate.

igrr, 2020-05-12:
Hi @fpgamaster, It seems that there is a (yet undocumented) bit in the same `RTC_CNTL_LOW_POWER_ST_REG` register, `RTC_CNTL_MAIN_STATE_IDLE` (bit 27). If this bit is set, the system is not in the sleep mode (i.e. running normally).
Note that when this bit is not 1, it might also mean that the system is transitioning to or from the sleep mode. If the system is entering the sleep mode, you may wait until RTC_CNTL_RDY_FOR_WAKEUP bit is set — eventually it will get set and you will be able to wake up the system. On the other hand, if the system is exiting the sleep mode, RTC_CNTL_RDY_FOR_WAKEUP will not get set, but RTC_CNTL_MAIN_STATE_IDLE will eventually become set. You might need to take this into account when implementing your wake-up logic.

fpgamaster, 2020-05-13:
Hi @igrr , Thanks! This is exactly I was looking for. It looks like this is the same bit as in s2 RTC_CNTL_MAIN_STATE_IS_IDLE found in header files.

AxelLin, 2022-02-22:
> Hi @fpgamaster, It seems that there is a (yet undocumented) bit in the same `RTC_CNTL_LOW_POWER_ST_REG` register, `RTC_CNTL_MAIN_STATE_IDLE` (bit 27). 

@igrr 
This is asked again recently: https://github.com/espressif/esp-idf/issues/8341
Any chance to update the document?

closed 2022-09-12 by espressif-bot