espressif/esp-idf #5255: Linker error when using mbedtls_hkdf

An00bIS47, 2020-05-09:
## Environment

- Development Kit:      
- Kit version 
- Module or chip used:  ESP32
- IDF version: v4.2-dev-1415-ga2263571b
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:    macOS
- Using an IDE?: No
- Power Supply:      USB

## Problem Description

This minimal example using the example esp-idf-template should at least compile and link successfully. It doesn't make sense but it shows the error I get.

### Expected Behavior

Linking succeeds

### Actual Behavior

This is the error:
```
LD build/app-template.elf
/Users/michael/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/michael/Development/hkdf-test/build/main/libmain.a(main.o):(.literal.app_main+0x4): undefined reference to `mbedtls_hkdf'
/Users/michael/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/michael/Development/hkdf-test/build/main/libmain.a(main.o): in function `app_main':
/Users/michael/Development/hkdf-test/main/main.c:30: undefined reference to `mbedtls_hkdf'
collect2: error: ld returned 1 exit status
make: *** [/Users/michael/Development/hkdf-test/build/app-template.elf] Error 1
```

### Steps to reproduce

make -j8

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "mbedtls/hkdf.h"
#include "mbedtls/md.h"

void app_main(void)
{
    int i = 0;
    while (1) {
        printf("[%d] Hello world!\n", i);

	    mbedtls_md_info_t *sha512_info;

	    sha512_info=mbedtls_md_info_from_type(MBEDTLS_MD_SHA512);

        mbedtls_hkdf(sha512_info, (uint8_t*) 0x00, 0x00, (uint8_t*)0x00, 0x00, (uint8_t*) 0x00, 0x00, (uint8_t*) 0x00, 0x00);    
        i++;
        vTaskDelay(5000 / portTICK_PERIOD_MS);
    }
}

```

## Debug Logs

```
```

## Other items if possible

ESP-Marius, 2020-05-09:
Hi,

`mbedtls_hkdf` is not included in the default build. It can be enabled in [esp_config.h](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/include/mbedtls/esp_config.h) by removing:

```
#ifdef MBEDTLS_HKDF_C
#undef MBEDTLS_HKDF_C
#endif
```

An00bIS47, 2020-05-09:
@ESP-Marius: Wow, that was fast :) This did fix the linker error and solve my problem :) Thank you very much!!

closed 2020-05-09 by An00bIS47