espressif/esp-idf #5258: esp-idf-v4.0 编译报错

feverGeek, 2020-05-10:
刚刚搭建好的环境用idf.py build 编译报错:
[ 31%] Built target __idf_lwip
[ 32%] Building C object esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth_mac_dm9051.c.obj
[ 32%] Building C object esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth_phy_dm9051.c.obj
[ 32%] Building C object esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth.c.obj
In file included from /root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:15:0:
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h: In function 'atomic_flag_test_and_set_explicit':
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:271:31: error: request for member '__val' in something not a structure or union
  __atomic_exchange_n(&(object)->__val, desired, order)
                               ^
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:387:10: note: in expansion of macro 'atomic_exchange_explicit'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
          ^
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:387:2: error: incompatible type for argument 1 of '__atomic_exchange_n'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
  ^
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h: In function 'atomic_flag_clear_explicit':
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:285:28: error: request for member '__val' in something not a structure or union
  __atomic_store_n(&(object)->__val, desired, order)
                            ^
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:394:2: note: in expansion of macro 'atomic_store_explicit'
  atomic_store_explicit(&__object->__flag, 0, __order);
  ^
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:394:2: error: incompatible type for argument 1 of '__atomic_store_n'
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_install':
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:91:47: error: request for member '__val' in something not a structure or union
 #define atomic_init(obj, value)  ((void)((obj)->__val = (value)))
                                               ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:174:5: note: in expansion of macro 'atomic_init'
     atomic_init(&eth_driver->ref_count, 1);
     ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_uninstall':
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:283:27: error: request for member '__val' in something not a structure or union
  __atomic_load_n(&(object)->__val, order)
                           ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:27:15: note: in definition of macro 'ETH_CHECK'
         if (!(a))                                                                 \
               ^
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:365:2: note: in expansion of macro 'atomic_load_explicit'
  atomic_load_explicit(object, memory_order_seq_cst)
  ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:215:15: note: in expansion of macro 'atomic_load'
     ETH_CHECK(atomic_load(&eth_driver->ref_count) == 1,
               ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:215:5: error: incompatible type for argument 1 of '__atomic_load_n'
     ETH_CHECK(atomic_load(&eth_driver->ref_count) == 1,
     ^
In file included from /root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:15:0:
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c: In function 'esp_eth_increase_reference':
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:273:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_add(&(object)->__val, operand, order)
                              ^
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:355:2: note: in expansion of macro 'atomic_fetch_add_explicit'
  atomic_fetch_add_explicit(object, operand, memory_order_seq_cst)
  ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:341:5: note: in expansion of macro 'atomic_fetch_add'
     atomic_fetch_add(&eth_driver->ref_count, 1);
     ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:341:5: error: incompatible type for argument 1 of '__atomic_fetch_add'
In file included from /root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:15:0:
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c: In function 'esp_eth_decrease_reference':
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:279:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_sub(&(object)->__val, operand, order)
                              ^
/root/Code/esp-idf-v4.0/components/newlib/include/stdatomic.h:361:2: note: in expansion of macro 'atomic_fetch_sub_explicit'
  atomic_fetch_sub_explicit(object, operand, memory_order_seq_cst)
  ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:352:5: note: in expansion of macro 'atomic_fetch_sub'
     atomic_fetch_sub(&eth_driver->ref_count, 1);
     ^
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c:352:5: error: incompatible type for argument 1 of '__atomic_fetch_sub'
/root/Code/esp-idf-v4.0/components/esp_eth/src/esp_eth.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'

suda-morris, 2020-05-10:
https://github.com/espressif/esp-idf/issues/4804

https://github.com/espressif/esp-idf/issues/5219

Alvin1Zhang, 2020-05-11:
@feverGeek Thanks for reporting, feel free to reopen if you still have the issue or file another ticket for new issue. Thanks.

closed 2020-05-11 by Alvin1Zhang