espressif/esp-idf #5261: Unified Provisioning vs. BluFi

antoniuschan99, 2020-05-11:
Playing with both Unified Provisioning and BluFi I noticed that with Unified Provisioning using BLE, if the user types an incorrect password, there is no way for the user to re-enter without having to restart the ESP (which then disconnects the user from the bluetooth connection). Is there a way around this? I was hoping wifi_prov_mgr_disable_auto_stop() would solve this but it doesn't look like it does?

Also, it seems like Espressif is focusing on Unified Provisioning, and BluFi may become legacy. Is this true? The code in BluFi is quite verbose, whereas Unified Provisioning is light (which is good!).

Thank you

AshUK, 2020-06-23:
also curious about this. Which is the preferred provisioning mechanism for new designs ?

antoniuschan99, 2020-06-28:
I ended up using blufi because it had better error handling. You can refer to the android app.

shahpiyushv, 2020-06-29:
@antoniuschan99 @AshUK , Unified Provisioning is recommended for newer designs. @antoniuschan99 , can you elaborate on what error handling did you find missing in Unified Provisioning? We can address it if possible. 

Meanwhile, the current provisioning state machine does not allow sending credentials again. We will check how it can be handled, for BLE as well as SoftAP, since the behaviours are supposed to be same for both.

antoniuschan99, 2020-07-01:
If the user enters the password incorrectly then it’s pretty much it. Connection has to be reesablished all over again (this is stated in the docs). Also blufi has custom parameters so my app allows users to set parameters into nvs flash and retrieve data from the esp32.

Will blufi be deprecated in the future? I literally just finished the entire provisioning for it!

shahpiyushv, 2020-07-01:
> If the user enters the password incorrectly then it’s pretty much it. Connection has to be reesablished all over again (this is stated in the docs)

That's right. I will check how it can be addressed. 

> Also blufi has custom parameters so my app allows users to set parameters into nvs flash and retrieve data from the esp32.

This is already supported in unified provisioning. [This example](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr) demonstrates a custom_prov_data_handler. You can have multiple such handlers. Have you checked the new [Android](https://github.com/espressif/esp-idf-provisioning-android) and [iOS](https://github.com/espressif/esp-idf-provisioning-ios) provisioning libraries? The sendData method for [ESPDevice](https://espressif.github.io/esp-idf-provisioning-ios/Classes/ESPDevice.html) would be the app side function to send custom data.

> Will blufi be deprecated in the future? I literally just finished the entire provisioning for it!

Though we recommend Unified Provisioning for new projects, there are no plans of deprecating BluFi.

antoniuschan99, 2020-07-02:
Ok thanks great to know! 🙏

closed 2020-07-02 by antoniuschan99