espressif/esp-idf #5262: Allow DTLS in MBEDTLS

alorbach, 2020-05-11:
Please enable CONFIG_MBEDTLS_SSL_PROTO_DTLS=y for the next release, which includes DTLS Support into MBEDTLS library.

There are some API's using DTLS in Smart Home environment like (Philips Hue Entertainment API). And without DTLS it is not possible to use them. 

I have compiled libraries myself with DTLS support and got it successfully running with Philips Hue Entertainment API.

mahavirj, 2020-05-11:
@alorbach You could achieve that by having `sdkconfig.defaults` file in application directory with entry `CONFIG_MBEDTLS_SSL_PROTO_DTLS=y ` (more information at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#custom-sdkconfig-defaults). Let me know if that helps to fix your problem?

alorbach, 2020-05-11:
@mahavirj thanks I know how to build the libs myself and have the setting enabled. Thats what I did with the current esp32 release (1.0.4). 

My request is to enable DTLS by default as it is a reasonable feature for IOT platform like esp32.

mahavirj, 2020-05-11:
> My request is to enable DTLS by default as it is a reasonable feature for IOT platform like esp32. And it is not needed to rebuild the mbedtls libraries then if you want to use it.

As far as IDF is concerned, currently only `coap` examples rely on this configuration, and hence its  enabled as part of these examples only. This also allows to not have any footprint impact on rest of applications (covers larger set in IDF) which do not have any requirement on DTLS.

alorbach, 2020-05-11:
So it would not hurt any other part of the components if DTLS is enabled by default?

negativekelvin, 2020-05-14:
Since this is regarding prebuilt Arduino libraries it should go here https://github.com/espressif/arduino-esp32

alorbach, 2020-05-15:
Allright I will open an issue there.

mahavirj, 2020-05-15:
Just to provide an additional update:

In ESP-IDF, DTLS is already enabled by default because of following configuration coming from `coap` component .

https://github.com/espressif/esp-idf/blob/5e436be77f68b1ba7d2ebbefb8aeda34f6c71276/components/coap/Kconfig#L13-L17

This has additional ~10K code/text impact on non-DTLS examples (e.g. `https_request`) and small impact on data memory as well. We will fix this, as in to have this enabled for `coap` related examples only. This issue will be closed once fix reaches to github.

alorbach, 2020-05-15:
Well in esp32 release 1.03, it was NOT enabled and I checked out 1.04 about 2 month ago and it was still not enabled. So I had to enable it myself, build thje libraries and replace them in my environment. 

I have describved the details here: 
https://github.com/alorbach/wificlientsecureudp

closed 2020-05-22 by espressif-bot