espressif/esp-idf #5263: ESP_GATT_CONN_LMP_TIMEOUT when connecting with GATTC example

AlpinProgrammer, 2020-05-11:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.0
- Build System: Make
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Hi !
With the ESP as the client, I want to connect to a specific device (server), filtered by name.
I am encountering a problem while connecting to this device: after approximately 40s, I get a disconnect with reason 0x22 (_ESP_GATT_CONN_LMP_TIMEOUT_).

However, it works using nRF connect, nRF connect desktop and using a nRF devkit.
Also, I do not get any error using the blecent nimBLE example (I deactivated the Alert Notification service check).

Besides, I do not know what could cause this kind of timeout and why it behaves differently with bluedroid with respect to nimBLE.

Thanks in advance :)

### Expected Behavior

I can successfully connect to a server using _esp_ble_gattc_open_ function.

### Actual Behavior

After approximately 40s after using _esp_ble_gattc_open_ function, I get a disconnect error with reason 0x22.

### Code to reproduce this issue

Example _examples/bluetooth/bluedroid/ble/gatt_client_ changing const _remote_device_name_ to my target name.

## Debug Logs

```
I (6333) GATTC_DEMO: connect to the remote device
I (6353) GATTC_DEMO: stop scan successfully
W (46443) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0022
W (46453) BT_L2CAP: L2CA_EnableUpdateBleConnParams - unknown BD_ADDR 9d30760fed07
E (46453) BT_GATT: GATTC_Discover Illegal param: disc_type 1 conn_id = 3
E (46463) BT_APPL: discovery on server failed
I (46463) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (46473) GATTC_DEMO: REMOTE BDA:
I (46473) GATTC_DEMO: 9d 30 76 0f ed 07
E (46483) GATTC_DEMO: discover service failed, status 135
I (46483) GATTC_DEMO: open success
I (46493) GATTC_DEMO: ESP_GATTC_DISCONNECT_EVT, reason = 34
```

AlpinProgrammer, 2020-05-18:
Hello,
I have some follow-up information.
The peripheral we wish to connect to is ST Microelectronic based.
Here is a link to a related issue with ST devices: [https://github.com/espressif/esp-idf/issues/1051](https://github.com/espressif/esp-idf/issues/1051)

I will let you know about the ST chip used as soon as I have the information.

If, as it seems, it is due to the peripheral BLE implementation as seen in the linked issue, is there any way to by-pass this issue ?
In that case it seems odd that it works with nimBLE stack and not with bluedroid.

Thanks.

WCCWCC, 2020-05-22:
Hi @AlpinProgrammer ,

     About LL_RESPONSE_TIMEOUT (ESP_GATT_CONN_LMP_TIMEOUT) question:
I can add some debug information, can you help me reproduce it again?

    If you meant it, I will track this issue quickly！

AlpinProgrammer, 2020-05-25:
Hi @WCCWCC , 
Sure ! Thank you very much.
The chip used is a BlueNRG-MS with core v4.1.
It was asserted on another issue ([https://github.com/espressif/esp-idf/issues/1051](https://github.com/espressif/esp-idf/issues/1051)) that this BLE stack had a problem.
As I described, the problem is noticeable using the bluedroid gatt_server example when connecting to a ST device.

AlpinProgrammer, 2020-05-25:
By commenting the following, it seems that it should resolve the issue. However, it does not solve the problem.
https://github.com/espressif/esp-idf/blob/463a9d8b7f9af8205222b80707f9bdbba7c530e1/components/bt/host/bluedroid/stack/btm/btm_ble_gap.c#L4110-L4126

Thanks in advance

WCCWCC, 2020-05-26:
Hi，@AlpinProgrammer 
 * The idf commit id I used: 280f0eff
 * Replace the lib in the attachment to the esp-idf/components/bt/controller/lib  path
 * If the replacement is successful, you will find the print of **BT controller compile version [82a164a]** in the log

Note:  If the compilation is not successful after replacing the library, please switch the commit of idf to the same as mine. If it still fails to compile, please provide me with **the commit id** of the idf you are currently using

[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/4680330/libbtdm_app.zip)

Alvin1Zhang, 2020-08-18:
Thanks for reporting, would you please help share if any updates for the issue? Thanks.

AlpinProgrammer, 2020-08-18:
Hello @Alvin1Zhang .
I wasn't able to solve this issue using the Bluedroid stack and using libbtdm_app.zip batch didn't solve the problem.
I didn't look further into this problem as we quickly decided to switch to nimBLE stack.
It worked perfectly with nimBLE !

Thank you for your help :)

WCCWCC, 2020-08-18:
Hi, @AlpinProgrammer 

The lib provided before is used to debug this problem.

Would you please provide a log of the recurring problem to help us solve this problem.

closed 2021-11-17 by Alvin1Zhang