espressif/esp-idf #5265: Bluedroid Ble encrypted MITM communication

promptps, 2020-05-11:
## Environment

- Development Kit:     ESP32 DOIT (ESP32 WROOM 4MB)
- Kit version :  DevKit: v1
- Module or chip used:   ESP32-WROOM-32
- IDF version :  v4.2-dev-532-g6f7be93
- Build System:  CMake | idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: Yes ( Eclipse IDE [Version: 2019-06 (4.12.0)] with ESP-IDF Eclipse plugin v1.0.1 )
- Power Supply:    USB

## Problem Description
Need to know about the process for communication using Bluedroid stack with ESP_GATT_AUTH_REQ_MITM in Central BLE device.
//Detailed problem description goes here.
I am testing the ESP32 BLE communication with Android device with nRF Connect Application.
I have configured Bluedroid stack as Central/Client BLE and I am Able to perform Bonding and communication with the characteristics having read/write permissions.
Now when the permission is change to read encrypted MITM in nRF Connect App, ESP32 is unable to perform read and write operation though we have set auth request type as "ESP_GATT_AUTH_REQ_MITM".

Please update me, if there is any change required in procedure of encrypted MITM communication to make it possible.
Please check the attached logs as well as source and sdkconfig files for reference.

### Expected Behavior
The device must able to perform read and write operation with characteristics having permissions Read/Write Encrypted MITM.

### Actual Behavior
The device is unable to communicate and gives error on calling read and write api.

### Steps to reproduce

1. Configure the bluedroid gatt secure client example code or use the attached sdk config and test code.
2. Change the read write permission to ESP_GATT_AUTH_REQ_MITM.
3. change the permission in nRF Connect app for the current selected characteristics.Uncheck read and write permission and check read encrytped MITM  and write encrytped MITM in app.
4. Start the Advertiser with name as fixed in Code.
5. Wait for the pairing request (if done for first time).
6. As per the test code, device will perform read and write after 10 seconds.
here as soon as device performs read and write operation, it gets error.

## Debug Logs

```
I (5365) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
I (5365) AppBle: Received event: eAppBleSearchServices!
I (20935) AppBle: Received event: eAppBleWrite!
I (20935) GATTC_DEMO: Writing handle:42, connid:0 
E (20935) BT_GATT: Unknown operation encryption completed
W (34975) BT_APPL: bta_gattc_conn_cback() - cif=1 connected=0 conn_id=1 reason=0x0013
E (34975) GATTC_DEMO: write char failed, error status = 85
E (34975) GATTC_DEMO: Write Failed 
I (34975) GATTC_DEMO: ESP_GATTC_DISCO
```

## Other items if possible

[gatt_client_demo.elf.zip](https://github.com/espressif/esp-idf/files/4609611/gatt_client_demo.elf.zip)
[bludroid_central.zip](https://github.com/espressif/esp-idf/files/4609612/bludroid_central.zip)
[Bluedroidble_errlog11052020.txt](https://github.com/espressif/esp-idf/files/4609627/Bluedroidble_errlog11052020.txt)

Alvin1Zhang, 2020-05-12:
@promptps Thanks for the detailed report, we will look into.

zhp0406, 2023-11-20:
Has your issue been resolved?

closed 2025-03-14 by Alvin1Zhang