espressif/esp-idf #5266: static ip set in wifi- station mode in esp32

sowmimalu, 2020-05-11:
hai ,
      In my application i need to connect with the static IP given.
i can able to connect when used
ESP_ERROR_CHECK(tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_STA, &ip_info);
it can able get ip with dynamic and connect with wifi
then i tried with static,
tcpip_adapter_init();
	 tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA); // Don't run a DHCP client
tcpip_adapter_ip_info_t ip_info;
ip_info.ip.addr = ipaddr_addr("192.168.43.3");
ip_info.gw.addr = ipaddr_addr("192.168.43.1");
ip_info.netmask.addr = ipaddr_addr("255.255.255.0");
tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA, &ip_info);
 printf("IP Address:  %s\n", ip4addr_ntoa(&ip_info.ip));
	 printf("Subnet mask: %s\n", ip4addr_ntoa(&ip_info.netmask));
	 printf("Gateway:     %s\n", ip4addr_ntoa(&ip_info.gw));
	 ESP_ERROR_CHECK(tcpip_adapter_dhcpc_start(TCPIP_ADAPTER_IF_STA));

 but always i am getting invalid static ip ...

could you please tell me on how to fix the static ip and to connect with wifi??

david-cermak, 2020-05-12:
Hi @sowmimalu 

No need to start the DHCP client after setting the address.
``` c
tcpip_adapter_set_ip_info(WIFI_IF_STA, &ip);
```
This should be enough to setup static IP address, after that an IP event should be received (depending on your event registration, e.g. `ip_sta_got_ip_handler()` if you register that function for the `IP_EVENT` base with the `IP_EVENT_STA_GOT_IP` event id)

sowmimalu, 2020-05-14:
tried now its  working...thank you

AxelLin, 2020-08-03:
This should be closed. @Alvin1Zhang

Alvin1Zhang, 2020-08-03:
Thanks for the notes @AxelLin . @sowmimalu Glad that it is working, feel free to reopen if the issue still happens.

closed 2020-08-03 by Alvin1Zhang