espressif/esp-idf #5267: crash in select

aep, 2020-05-11:
## Environment

-
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):  v4.2-dev-1206-g741960d5c
- Build System:       idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Using an IDE?: No
- Power Supply:    USB

## Problem Description

a simple select(3)  will crash when adding more than one fd to the fd read set

```
0x401438c0: find_value at /opt/toolchains/esp32/idf/components/pthread/pthread_local_storage.c:192

A2      : 0x00000008  A3      : 0x00000003  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000002  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc67b0  
A10     : 0x3ffc7290  A11     : 0x00060523  A12     : 0x00060520  A13     : 0x00060523  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace:0x401438bd:0x3ffc67d0 0x400d301d:0x3ffc67f0 0x400f4dc9:0x3ffc6810 0x400f4f4f:0x3ffc6830 0x400d2a73:0x3ffc6850 0x400d6ddc:0x3ffc68c0 0x40143959:0x3ffc68f0 0x400d521f:0x3ffc6910 0x40092951:0x3ffc7200
0x401438bd: find_value at /opt/toolchains/esp32/idf/components/pthread/pthread_local_storage.c:191

0x400d301d: pthread_getspecific at /opt/toolchains/esp32/idf/components/pthread/pthread_local_storage.c:206

0x400f4dc9: sys_thread_sem_get at /opt/toolchains/esp32/idf/components/lwip/port/esp32/freertos/sys_arch.c:495

0x400f4f4f: lwip_get_socket_select_semaphore at /opt/toolchains/esp32/idf/components/lwip/port/esp32/vfs_lwip.c:52

0x400d2a73: esp_vfs_select at /opt/toolchains/esp32/idf/components/vfs/vfs.c:898

0x400d6ddc: io_select_impl_wait at /home/aep/proj/zznursery/modules/net/modules/io/src/select.zz:119

0x40143959: io_wait at /home/aep/proj/zznursery/modules/net/modules/io/src/lib.zz:275

0x400d521f: xarrier_xarrier at /home/aep/proj/corona/esp/carrier/xarrier/src/lib.zz:40

0x40092951: vPortTaskWrapper at /opt/toolchains/esp32/idf/components/freertos/xtensa/port.c:143
```

dobairoland, 2020-05-12:
Hello @aep. Thanks for your report. 

> a simple select(3) will crash when adding more than one fd to the fd read set

Do you mean that all the file descriptors are socket file descriptors?

If yes then please try to disable [CONFIG_VFS_SUPPORT_SELECT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-vfs-support-select) which will bypass the VFS layer and `select()` will directly use `lwip_select()`. Do you experience any change?

dobairoland, 2020-05-12:
Here is an [example of select](https://github.com/espressif/esp-idf/tree/master/examples/system/select). It could help to check if you do all the required initializations.

aep, 2020-05-12:
thanks! i'll try to modify that example until it crashes

dobairoland, 2020-06-04:
@aep Do you have any update regarding this issue?

maikeriva, 2020-06-22:
@dobairoland I experience the same problem with ESP-IDF **stable release v4.0.1** when opening an HTTPS connection with esp-http-client. Here's the traceback:

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401d2cf8  PS      : 0x00060530  A0      : 0x800d3a20  A1      : 0x3ffe3260  
0x401d2cf8: find_value at /home/micheleriva/esp-idf/components/pthread/pthread_local_storage.c:192

A2      : 0xc2004136  A3      : 0x00000003  A4      : 0x00060520  A5      : 0x00000001  
A6      : 0x00060520  A7      : 0x3ffbb884  A8      : 0x8009a8c6  A9      : 0x3ffe3240  
A10     : 0x3ffbb4fc  A11     : 0x00000000  A12     : 0x3ffb63ec  A13     : 0x00000001  
A14     : 0x00060523  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xc2004136  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: acb09f97e4a349dc

Backtrace: 0x401d2cf5:0x3ffe3260 0x400d3a1d:0x3ffe3280 0x401667b3:0x3ffe32a0 0x40167771:0x3ffe32c0 0x40152d95:0x3ffe3300 0x4014033e:0x3ffe3350 0x4014038c:0x3ffe33a0 0x40140b6a:0x3ffe33f0 0x40140e7f:0x3ffe3420 0x4013f07f:0x3ffe3450 0x401d3bfd:0x3ffe3470 0x4013dc3d:0x3ffe3490 0x4013de85:0x3ffe34c0 0x400d8298:0x3ffe34e0 0x400dcedb:0x3ffe3510 0x400dcfb2:0x3ffe3990 0x40099505:0x3ffe39b0
0x401d2cf5: find_value at /home/micheleriva/esp-idf/components/pthread/pthread_local_storage.c:191

0x400d3a1d: pthread_getspecific at /home/micheleriva/esp-idf/components/pthread/pthread_local_storage.c:206

0x401667b3: sys_thread_sem_get at /home/micheleriva/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:455

0x40167771: netconn_gethostbyname_addrtype at /home/micheleriva/esp-idf/components/lwip/lwip/src/api/api_lib.c:1329 (discriminator 2)

0x40152d95: lwip_getaddrinfo at /home/micheleriva/esp-idf/components/lwip/lwip/src/api/netdb.c:342

0x4014033e: resolve_host_name at /home/micheleriva/esp-idf/components/esp-tls/esp_tls.c:62 (discriminator 6)

0x4014038c: esp_tcp_connect at /home/micheleriva/esp-idf/components/esp-tls/esp_tls.c:101

0x40140b6a: esp_tls_low_level_conn at /home/micheleriva/esp-idf/components/esp-tls/esp_tls.c:570

0x40140e7f: esp_tls_conn_new_sync at /home/micheleriva/esp-idf/components/esp-tls/esp_tls.c:697

0x4013f07f: ssl_connect at /home/micheleriva/esp-idf/components/tcp_transport/transport_ssl.c:74

0x401d3bfd: esp_transport_connect at /home/micheleriva/esp-idf/components/tcp_transport/transport.c:165

0x4013dc3d: esp_http_client_connect at /home/micheleriva/esp-idf/components/esp_http_client/esp_http_client.c:1016

0x4013de85: esp_http_client_open at /home/micheleriva/esp-idf/components/esp_http_client/esp_http_client.c:1172
```
One thing to notice related to thread storage pointers is that I set `component config`->`FreeRTOS`->`number of thread local storage pointers` equal to 10 in menuconfig. I don't recall this happing before this change, but I have not tested it yet.

Not being able to perform HTTPS connections is pretty critical for any kind of IoT application, and IDF v4.0.1 is labelled as stable. Is there a quick way to fix this?

dobairoland, 2020-06-22:
Hi @maikeriva. I don't think so that you have the same issue. Select is not even i the backtrace. Please open a new issue for it.

dobairoland, 2020-06-22:
Hi @aep. I'm closing this issue for lack of activity. Please let me know if you have follow-up information as we've discussed earlier.

closed 2020-06-22 by dobairoland