espressif/esp-idf #5268: Unexpectedly high heap allocations during HTTPS OTA

boarchuz, 2020-05-12:
## Environment

- Module or chip used:  bare WROOM-32D
- IDF version: v4.2-dev-1320-g1aebfdf6a
- Using Espressif VS Code extension, compiler esp-2020r1-8.2.0

## Problem Description

While testing HTTPS OTA, I checked its impact on heap memory in my program by dumping the free/minimum memory before and after, by using esp_get_free_heap_size() and esp_get_minimum_free_heap_size(). 
I noticed that the minimum had gone from 115K beforehand, to **<10K** after. Needless to say, this 100K on the heap far exceeds the overhead I expected.

I have reproduced it somewhat with simple_ota_example. The allocations seem to increase with the time taken to erase the partition, so I contrived this partition table with a very large ota_0 to exaggerate the effect:

```
# Name,   Type, SubType, Offset,    Size,   Flags
phy_init, data, phy,     0xC000,    0x1000,
factory,  app,  factory, 0x10000,   0xEC000,
nvs,      data, nvs,     0xFC000,   0x2000,
otadata,  data, ota,     0xFE000,   0x2000,
ota_0,    app,  ota_0,   0x100000,  0x2F0000,
ota_1,    app,  ota_1,   0x3F0000,  0x10000,
```

After inserting some logging, eg:
`ESP_LOGW(TAG, "Starting: %u, min: %u", esp_get_free_heap_size(), esp_get_minimum_free_heap_size());`

Here's the relevant output:

> I (4206) simple_ota_example: Starting OTA example
> W (4206) simple_ota_example: Starting: 231256, min: 222080
> W (4216) HTTP_CLIENT: Before http connect: 229288, min: 222080
> W (5486) HTTP_CLIENT: After http connect: 196928, min: 187696
> I (5516) esp_https_ota: Starting OTA...
> I (5526) esp_https_ota: Writing to partition subtype 16 at offset 0x100000
> W (5526) esp_https_ota: Before esp_ota_begin: 196596, min: 187696
> W (5526) esp_ota_ops: Before esp_partition_erase_range: 191328, Min: 187696
> E (10496) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (10546) task_wdt:  - IDLE0 (CPU 0)
> E (10546) task_wdt: Tasks currently running:
> E (10546) task_wdt: CPU 0: ipc0
> E (10546) task_wdt: CPU 1: IDLE1
> W (12886) esp_ota_ops: After esp_partition_erase_range: 153004, Min: 114116
> W (12886) esp_https_ota: After esp_ota_begin: 159756, min: 114116

The WDT is to be expected, and there's the expected allocations for mbedTLS and the HTTP client (~40K).
However, note that the minimum reached just 114K at some point, indicating a mystery further 70K total was allocated.

For what it's worth, here is similar output from my more complex program where memory was down to just 1624 bytes at some stage during esp_ota_begin:

> W (23344) MY-OTA: Free: 115728, Min: 105352
> W (23344) HTTP_CLIENT: Before http connect: 115728, min: 105352
> W (23854) HTTP_CLIENT: After http connect: 71104, min: 61844
> I (23974) MY-OTA: Starting OTA...
> W (23974) MY-OTA: Free: 62456, Min: 61844
> I (23974) MY-OTA: Writing to partition subtype 16 at offset 0x180000
> W (23974) esp_ota_ops: Before esp_partition_erase_range: 62456, Min: 61844
> W (27244) esp_ota_ops: After esp_partition_erase_range: 12472, Min: 12472
> I (27244) MY-OTA: esp_ota_begin succeeded
> I (27244) MY-OTA: Please Wait. This may take time
> W (27254) MY-OTA: Free: 60172, Min: 1624

My concern is that OTA downloads can take quite some time, and I have other concurrent tasks - in this case, WiFi is in APSTA mode with a httpd server up - so I am not at all confident of stability if OTA is sporadically chewing up 100K+ of memory.

Alvin1Zhang, 2020-05-12:
@boarchuz Thanks for your detailed report, we will look into.

boarchuz, 2020-05-12:
The TLSF branch shows a significant improvement. 
Where I had a minimum of <5K consistently as in the output above (on master), the exact same program on TLSF branch leaves me with 45-55K.

I'm not entirely convinced that the problem doesn't exist on this branch, but its effect is significantly reduced for some reason.

mahavirj, 2020-05-13:
@boarchuz 

> Needless to say, this 100K on the heap far exceeds the overhead I expected.

Above observation is also true with respect to any of the HTTPS OTA examples provided in IDF? For typical TLS connection, it can take up-to 30K-32K memory (with asymmetric content length configuration), hence bit skeptical that this is entirely coming in the context of OTA task.

You also might want to check per task heap utilization in this case, example at https://github.com/espressif/esp-idf/tree/master/examples/system/heap_task_tracking should be helpful for this.

boarchuz, 2020-05-13:
@mahavirj Thanks for that. Using the heap task tracking I can now clearly see the internal "wifi" task accumulating a lot of memory until it maxes out while the OTA task is busy in esp_ota_begin. It is freed approximately when esp_ota_begin returns.

Here's an annotated log output with the interesting stuff. The allocation for TLS/HTTP can be seen in my "D-OTA" task as expected, but note the CAP_8BIT for "wifi" between esp_ota_begin and esp_ota_begin==ESP_OK:
(CAP_32BIT == 0 in every case)

> **Free: 109564, min: 99900**
> 
> **esp_http_client_init**
> 
> (16155) Heap Info:
>  * Task: wifi -> CAP_8BIT: 26544
>  * Task: D-OTA -> CAP_8BIT: 5404
> 
> **Free: 107000, Min: 99900**
> 
> **esp_http_client_open**
> 
> (16175) Heap Info:
>  * Task: wifi -> CAP_8BIT: 26544
>  * Task: D-OTA -> CAP_8BIT: 7420
> 
> (16205) Heap Info:
>  * Task: wifi -> CAP_8BIT: 26544
>  * Task: D-OTA -> CAP_8BIT: 9100
> 
> (16215) Heap Info:
>  * Task: wifi -> CAP_8BIT: 26768
>  * Task: D-OTA -> CAP_8BIT: 45712
> 
> (16225) Heap Info:
>  * Task: wifi -> CAP_8BIT: 28612
>  * Task: D-OTA -> CAP_8BIT: 54172
> 
> (16915) Heap Info:
>  * Task: wifi -> CAP_8BIT: 26544
>  * Task: D-OTA -> CAP_8BIT: 53800
> 
> **Free: 59548, Min: 51044**
> 
> **esp_http_client_fetch_headers**
> 
> (16925) Heap Info:
>  * Task: wifi -> CAP_8BIT: 26544
>  * Task: D-OTA -> CAP_8BIT: 49688
> 
> **Free: 60728, Min: 51044**
> 
> **esp_ota_begin**
> 
> (16965) Heap Info:
>  * Task: wifi -> CAP_8BIT: 26544
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> (18485) Heap Info:
>  * Task: wifi -> CAP_8BIT: 55296
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> (18645) Heap Info:
>  * Task: wifi -> CAP_8BIT: 65472
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> (18955) Heap Info:
>  * Task: wifi -> CAP_8BIT: 68864
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> (19565) Heap Info:
>  * Task: wifi -> CAP_8BIT: 82432
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> (19915) Heap Info:
>  * Task: wifi -> CAP_8BIT: 82432
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> (19985) Heap Info:
>  * Task: wifi -> CAP_8BIT: 82432
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> (20035) Heap Info:
>  * Task: wifi -> CAP_8BIT: 82432
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> (20105) Heap Info:
>  * Task: wifi -> CAP_8BIT: 82432
>  * Task: D-OTA -> CAP_8BIT: 49632
> 
> **esp_ota_begin == ESP_OK**
> 
> **Free: 4116, Min: 4056**
> 
> (20155) Heap Info:
>  * Task: wifi -> CAP_8BIT: 27656
>  * Task: D-OTA -> CAP_8BIT: 50188
> 
> **Free: 60060, min: 4056**

And another that more clearly shows it climbing:

> **Free: 54712, Min: 51460**
> 
> **esp_ota_begin**
> 
> (32335) Heap Info:
>  * Task: wifi -> CAP_8BIT: 33208
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (33715) Heap Info:
>  * Task: wifi -> CAP_8BIT: 36540
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (34175) Heap Info:
>  * Task: wifi -> CAP_8BIT: 43264
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (34325) Heap Info:
>  * Task: wifi -> CAP_8BIT: 43264
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (34485) Heap Info:
>  * Task: wifi -> CAP_8BIT: 46656
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (34645) Heap Info:
>  * Task: wifi -> CAP_8BIT: 55136
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (34805) Heap Info:
>  * Task: wifi -> CAP_8BIT: 58528
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (35125) Heap Info:
>  * Task: wifi -> CAP_8BIT: 63616
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (35385) Heap Info:
>  * Task: wifi -> CAP_8BIT: 67008
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (35455) Heap Info:
>  * Task: wifi -> CAP_8BIT: 68704
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> (35525) Heap Info:
>  * Task: wifi -> CAP_8BIT: 70400
>  * Task: D-OTA -> CAP_8BIT: 49628
> 
> **esp_ota_begin == ESP_OK**
> 
> **Free: 10104, Min: 6540**
> 
> (35595) Heap Info:
>  * Task: wifi -> CAP_8BIT: 35640
> 
> **Free: 52384, min: 6540**

Static wifi allocation didn't have any significant effect as far as I can tell. While idle CAP_8BIT was much higher as expected (53680 vs 26544), during esp_ota_begin this increased in the same manner as above to ~83k before releasing.

igrr, 2020-05-13:
Could you please paste your Wi-Fi buffer configuration here? (run `grep WIFI_ sdkconfig`)

boarchuz, 2020-05-13:
```
CONFIG_ESP_MAC_ADDR_UNIVERSE_WIFI_STA=y
CONFIG_ESP_MAC_ADDR_UNIVERSE_WIFI_AP=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
# CONFIG_ESP32_WIFI_STATIC_TX_BUFFER is not set
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=1
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=32
# CONFIG_ESP32_WIFI_CSI_ENABLED is not set
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=6
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=6
CONFIG_ESP32_WIFI_NVS_ENABLED=y
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
# CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1 is not set
CONFIG_ESP32_WIFI_SOFTAP_BEACON_MAX_LEN=752
CONFIG_ESP32_WIFI_MGMT_SBUF_NUM=32
# CONFIG_ESP32_WIFI_DEBUG_LOG_ENABLE is not set
CONFIG_ESP32_WIFI_IRAM_OPT=y
CONFIG_ESP32_WIFI_RX_IRAM_OPT=y
# CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE is not set
CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER=20
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16
CONFIG_WIFI_PROV_AUTOSTOP_TIMEOUT=30
```

And if I set "Type of WiFi TX buffers" to Static:
```
CONFIG_ESP_MAC_ADDR_UNIVERSE_WIFI_STA=y
CONFIG_ESP_MAC_ADDR_UNIVERSE_WIFI_AP=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER=y
# CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER is not set
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=0
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM=16
# CONFIG_ESP32_WIFI_CSI_ENABLED is not set
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=6
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=6
CONFIG_ESP32_WIFI_NVS_ENABLED=y
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
# CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1 is not set
CONFIG_ESP32_WIFI_SOFTAP_BEACON_MAX_LEN=752
CONFIG_ESP32_WIFI_MGMT_SBUF_NUM=32
# CONFIG_ESP32_WIFI_DEBUG_LOG_ENABLE is not set
CONFIG_ESP32_WIFI_IRAM_OPT=y
CONFIG_ESP32_WIFI_RX_IRAM_OPT=y
# CONFIG_ESP32_WIFI_ENABLE_WPA3_SAE is not set
CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER=20
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16
CONFIG_WIFI_PROV_AUTOSTOP_TIMEOUT=30
```

mahavirj, 2020-05-13:
@boarchuz I think buffering will be in RX path, can you please try reducing `CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM`?

boarchuz, 2020-05-13:
@mahavirj That's it, there's a clear direct correlation between that value and the max wifi task heap. 
At 8, for example, "wifi" maxes out at ~50k for CAP_8BIT; at 6, it maxes out at ~34.5K (compared to >80 with default value 32).
Am I to understand then that this is normal behaviour, and that it's my responsibility to limit allocations by lowering this value (and changing static options to compensate)?

I get a few of these at lower values, fyi:
> E (16475) wifi:esf_buf: t=8 l=1600 max:6, alloc:6 no eb, TXQ_BLOCK=0
> E (16625) wifi:esf_buf: t=8 l=1600 max:6, alloc:6 no eb, TXQ_BLOCK=0
> E (17055) wifi:esf_buf: t=8 l=1600 max:6, alloc:6 no eb, TXQ_BLOCK=4000
> E (17185) wifi:esf_buf: t=8 l=1600 max:6, alloc:6 no eb, TXQ_BLOCK=4000

Side issue: The description for CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM says that 0 = unlimited, yet setting to 0 results in an error as it is < CONFIG_ESP32_WIFI_RX_BA_WIN:

> #error "WiFi configuration check: WARNING, WIFI_RX_BA_WIN should not be larger than WIFI_DYNAMIC_RX_BUFFER_NUM!"

igrr, 2020-05-13:
> Am I to understand then that this is normal behaviour, and that it's my responsibility to limit allocations by lowering this value (and changing static options to compensate)?

Yes, there is a set of tradeoffs associated with picking the default values (peak performance, peak memory use, etc). In practice these buffer settings need to be set depending on the application, there is a guide here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-buffer-configure.

boarchuz, 2020-05-13:
@igrr  Alright, that makes sense. 
Thanks, everyone.

mahavirj, 2020-05-13:
@boarchuz 

Re-opening issue, couple of things would like to check here:

- Just wondering if `esp_ota_begin` is correctly placed in https ota state machine, we will check on this. If we erase partition before beginning TLS connection, it may reduce memory pressure. Downside being, erase may happen even if upgrade was not desired (since upgrade version is part of TLS stream itself).

- PR https://github.com/espressif/esp-idf/pull/5246 can also be helpful here.

We will keep this issue updated post some investigations on our side.

Thanks.

boarchuz, 2020-05-13:
Thanks for that idea, I've given it a try myself and it is much better.
The up-to-55k additionally allocated by "wifi" (from 26k to >80k) during esp_ota_begin is now freed before my OTA task needs its ~45K. In effect, eliminating this overlap means my minimum is now typically about 35k (up from 5k).

Although I've also noticed that **I was missing the SPI_FLASH_YIELD_DURING_ERASE changes** reintroduced here: https://github.com/espressif/esp-idf/commit/b2417819fdbb5d00103b2a0612bb98d09f9e5533
This provides a significant benefit: "wifi" task only ever needs an additional maximum of 15-20k (for a total of about 45k) during esp_ota_begin, and I can see this value correctly fluctuating in that time rather than increasing continuously.
(It also explains my observations on TLSF branch above; it was simply because my local TLSF branch is up-to-date and includes that commit.)

While I can save even more memory in the OTA init phase by combining your approach with that config option (for a minimum >50k), it seems trying to avoid concurrent allocation might be in vain given that the wifi task heap is growing and shrinking for the duration of the download anyway.

mahavirj, 2020-05-14:
@boarchuz 

> Although I've also noticed that I was missing the SPI_FLASH_YIELD_DURING_ERASE changes reintroduced here: b241781

Probably this explains the problem then. Yield during erase provides opportunity of networking stack task to process buffers from WiFi driver, thus alleviating memory pressure.

> While I can save even more memory in the OTA init phase by combining your approach with that config option (for a minimum >50k)

Can you please clarify which approach? Any specific patch the you created?

boarchuz, 2020-05-15:
Yeah I was following that issue too and I've now realised how poor my config was so I feel silly on two counts!

The approach I tried was to reorder HTTPS OTA from the default of something like this:
`http_client_open (+50k) -> esp_ota_begin (+up to 50k momentarily) -> download...` (maximum at any one time: ~100k)
to:
`esp_ota_begin (+up to 50k momentarily) -> http_client_open (+50k)-> download...` (maximum at any one time: ~50k)

I believe this is what you were suggesting but I may have misunderstood. With my poor config, that was a great improvement.

However now that I've realised the problem is in my config and I've had even a brief chance to optimise, it already looks much better:
`http_client_open (+30k) -> esp_ota_begin (+up to 15k momentarily) -> download...` (maximum at any one time: ~45k)

Considering that WiFi is of course very active throughout the download, it's natural for the heap to be fluctuating for some time, so with a decent configuration there isn't as much to gain by reordering (with my current setup, at least):
`esp_ota_begin (+up to 15k momentarily) -> http_client_open (+30k) -> download (+up to 5-10k momentarily)` (maximum at any one time: ~35-40k)

mahavirj, 2020-05-15:
@boarchuz 

> I believe this is what you were suggesting but I may have misunderstood. With my poor config, that was a great improvement.

Yes. 

> Considering that WiFi is of course very active throughout the download, it's natural for the heap to be fluctuating for some time, so with a decent configuration there isn't as much to gain by reordering (with my current setup,

Agree and confirm observations on my side as well are similar to yours. So we are good with respect to `https_ota` state machine and there is no change in ordering is required here. If there are no followup questions from your side then we can close this issue. (IMO, PR https://github.com/espressif/esp-idf/pull/5246 is still good to have, but that I will follow separately).

Thanks.

closed 2020-05-15 by boarchuz