espressif/esp-idf #5271: parttool.py crashes when retrieving partition info

XaviDCR92, 2020-05-12:
## Environment

- Development Kit:      [N/A]
- Kit version (for WroverKit/PicoKit/DevKitC): [N/A]
- Module or chip used:  [N/A]
- IDF version (run ``git describe --tags`` to find it):
    v4.1-beta1-182-g9778b163b
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [N/A]

## Problem Description

parttool.py crashes when attempting to retrieve partition from valid partition name and table.

### Expected Behavior
Partition information should be returned.
### Actual Behavior
This error occurs when a valid partition name is given, nothing is returned otherwise. Input partition file is ensured to be valid.
### Steps to reproduce
Execute the following instruction on the root of a project folder where partition table file is also stored:
`$ parttool.py --partition-table-file partition_table.csv get_partition_info --partition-name ota_data`
Where `partition_table.csv` is a valid partition table file and `ota_data` can be any valid partition name from the table.
### Code to reproduce this issue
N/A
## Debug Logs

```
Traceback (most recent call last):
  File "/home/x/esp-idf/components/partition_table/parttool.py", line 346, in <module>
    main()
  File "/home/x/esp-idf/components/partition_table/parttool.py", line 339, in main
    op(**common_args)
  File "/home/x/esp-idf/components/partition_table/parttool.py", line 208, in _get_partition_info
    for i in info:
TypeError: 'NoneType' object is not iterable
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

KonstantinKondrashov, 2020-05-13:
Hi @XaviDCR92!
Could you try to add `--info size`?

`parttool.py --partition-table-file partitions.csv get_partition_info --partition-name otadata --info size`

I think you expected that it will work and return offset and size, for this you can use `--info size offset` .
I will fix this by adding the default option for the `info` option [offset, size].

Thanks

XaviDCR92, 2020-05-13:
That solved it. Thank you!

Alvin1Zhang, 2020-05-14:
Reopen until the fix is available on GitHub.

closed 2020-05-22 by espressif-bot