espressif/esp-idf #5273: PEAP test failed with FreeRADIUS on Ubuntun 20.04

AxelLin, 2020-05-12:
## Environment

- Module or chip used: ESP32-WROOM-32
- IDF version: v4.2-dev-1415-ga2263571b5ff
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: NO
- Power Supply: USB

## Problem Description

Below shows the messages from FreeRADIUS:
(97) Received Access-Request Id 0 from 192.168.1.6:1028 to 192.168.1.113:1812 length 210
(97)   User-Name = "bob@mytest.com"
(97)   NAS-IP-Address = 192.168.1.6
(97)   Called-Station-Id = "ba5510cfaad0"
(97)   Calling-Station-Id = "c82b96ae4d1c"
(97)   NAS-Identifier = "ba5510cfaad0"
(97)   NAS-Port = 63
(97)   Framed-MTU = 1400
(97)   State = 0x9309e537920bfc6c3fb380500d115bea
(97)   NAS-Port-Type = Wireless-802.11
(97)   EAP-Message = 0x0202004619800000003c16030100370100003303010000004075441c3e4112187d050f410d7ed684573c78b110e378d8822ad190a500000c003d0035003c002f000500040100
(97)   Message-Authenticator = 0xe04e0909336d5edbeaa3c79396c8444c
(97) session-state: No cached attributes
(97) # Executing section authorize from file /etc/freeradius/3.0/sites-enabled/default
(97)   authorize {
(97)     policy filter_username {
(97)       if (&User-Name) {
(97)       if (&User-Name)  -> TRUE
(97)       if (&User-Name)  {
(97)         if (&User-Name =~ / /) {
(97)         if (&User-Name =~ / /)  -> FALSE
(97)         if (&User-Name =~ /@[^@]*@/ ) {
(97)         if (&User-Name =~ /@[^@]*@/ )  -> FALSE
(97)         if (&User-Name =~ /\.\./ ) {
(97)         if (&User-Name =~ /\.\./ )  -> FALSE
(97)         if ((&User-Name =~ /@/) && (&User-Name !~ /@(.+)\.(.+)$/))  {
(97)         if ((&User-Name =~ /@/) && (&User-Name !~ /@(.+)\.(.+)$/))   -> FALSE
(97)         if (&User-Name =~ /\.$/)  {
(97)         if (&User-Name =~ /\.$/)   -> FALSE
(97)         if (&User-Name =~ /@\./)  {
(97)         if (&User-Name =~ /@\./)   -> FALSE
(97)       } # if (&User-Name)  = notfound
(97)     } # policy filter_username = notfound
(97)     [preprocess] = ok
(97)     [chap] = noop
(97)     [mschap] = noop
(97)     [digest] = noop
(97) suffix: Checking for suffix after "@"
(97) suffix: Looking up realm "mytest.com" for User-Name = "bob@mytest.com"
(97) suffix: No such realm "mytest.com"
(97)     [suffix] = noop
(97) eap: Peer sent EAP Response (code 2) ID 2 length 70
(97) eap: Continuing tunnel setup
(97)     [eap] = ok
(97)   } # authorize = ok
(97) Found Auth-Type = eap
(97) # Executing group from file /etc/freeradius/3.0/sites-enabled/default
(97)   authenticate {
(97) eap: Expiring EAP session with state 0x9309e537920bfc6c
(97) eap: Finished EAP session with state 0x9309e537920bfc6c
(97) eap: Previous EAP request found for state 0x9309e537920bfc6c, released from the list
(97) eap: Peer sent packet with method EAP PEAP (25)
(97) eap: Calling submodule eap_peap to process data
(97) eap_peap: Continuing EAP-TLS
(97) eap_peap: Peer indicated complete TLS record size will be 60 bytes
(97) eap_peap: Got complete TLS record (60 bytes)
(97) eap_peap: [eaptls verify] = length included
(97) eap_peap: (other): before SSL initialization
(97) eap_peap: TLS_accept: before SSL initialization
(97) eap_peap: TLS_accept: before SSL initialization
(97) eap_peap: <<< recv TLS 1.3  [length 0037] 
(97) eap_peap: >>> send TLS 1.0 Alert [length 0002], fatal protocol_version 
**(97) eap_peap: ERROR: TLS Alert write:fatal:protocol version
tls: TLS_accept: Error in error
(97) eap_peap: ERROR: Failed in __FUNCTION__ (SSL_read): error:14209102:SSL routines:tls_early_post_process_client_hello:unsupported protocol
(97) eap_peap: ERROR: System call (I/O) error (-1)
(97) eap_peap: ERROR: TLS receive handshake failed during operation
(97) eap_peap: ERROR: [eaptls process] = fail
(97) eap: ERROR: Failed continuing EAP PEAP (25) session.  EAP sub-module failed**
(97) eap: Sending EAP Failure (code 4) ID 2 length 4
(97) eap: Failed in EAP select
(97)     [eap] = invalid
(97)   } # authenticate = invalid
(97) Failed to authenticate the user
(97) Using Post-Auth-Type Reject
(97) # Executing group from file /etc/freeradius/3.0/sites-enabled/default
(97)   Post-Auth-Type REJECT {
(97) attr_filter.access_reject: EXPAND %{User-Name}
(97) attr_filter.access_reject:    --> bob@mytest.com
(97) attr_filter.access_reject: Matched entry DEFAULT at line 11
(97)     [attr_filter.access_reject] = updated
(97)     [eap] = noop
(97)     policy remove_reply_message_if_eap {
(97)       if (&reply:EAP-Message && &reply:Reply-Message) {
(97)       if (&reply:EAP-Message && &reply:Reply-Message)  -> FALSE
(97)       else {
(97)         [noop] = noop
(97)       } # else = noop
(97)     } # policy remove_reply_message_if_eap = noop
(97)   } # Post-Auth-Type REJECT = updated
(97) Delaying response for 1.000000 seconds
Thread 4 waiting to be assigned a request
Waking up in 0.6 seconds.
(97) Sending delayed response
(97) Sent Access-Reject Id 0 from 192.168.1.113:1812 to 192.168.1.6:1028 length 44
(97)   EAP-Message = 0x04020004
(97)   Message-Authenticator = 0x00000000000000000000000000000000
Waking up in 3.9 seconds.
(97) Cleaning up request packet ID 0 with timestamp +1912
Ready to process requests

sagb2015, 2020-05-12:
Hi @AxelLin , our code does not support TLS 1.3 yet. However, we are planning to port the TLS functionality over mbedtls. So it is may be possible to support it in near future.

AxelLin, 2020-05-12:
> Hi @AxelLin , our code does not support TLS 1.3 yet. However, we are planning to port the TLS functionality over mbedtls. So it is may be possible to support it in near future.

Hi @sagb2015 

I have below settings in my freeradius setting:
                tls_min_version = "1.0"
                tls_max_version = "1.2"

But it seems wpa_supplicant does attempt TLS 1.3 because eap_peap **received** TLS 1.3.
(97) eap_peap: <<< recv TLS 1.3 [length 0037]
(97) eap_peap: >>> send TLS 1.0 Alert [length 0002], fatal protocol_version

AxelLin, 2020-05-15:
@sagb2015 

I also tried various mobile phone and notebook devices, they all work.
Only ESP32 cannot join the AP.
I'm confused now. I'm not sure the problem is on ESP32 side or the FreeRADIUS side.
I thought TLS1.3 will be used only when both ends support TLS1.3, it should not be a problem if
esp32 does not support TLS1.3.

If the problem is on FreeRADIUS side, what's the recommend FreeRADIUS setup to test ESP32 wpa2 enterprise?
Any comments?

sagb2015, 2020-05-16:
@AxelLin Can you try the solution [here](https://github.com/FreeRADIUS/freeradius-server/issues/2385#issuecomment-502643208) . You can set TLS version in freeradius config.

AxelLin, 2020-05-16:
hi @sagb2015 
I already set 
tls_min_version = "1.0"
tls_max_version = "1.2"

But it still does not work.
Below shows my configuration:
$ sudo freeradius -CX
FreeRADIUS Version 3.0.20
Copyright (C) 1999-2019 The FreeRADIUS server project and contributors
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE
You may redistribute copies of FreeRADIUS under the terms of the
GNU General Public License
For more information about these matters, see the file named COPYRIGHT
Starting - reading configuration files ...
including dictionary file /usr/share/freeradius/dictionary
including dictionary file /usr/share/freeradius/dictionary.dhcp
including dictionary file /usr/share/freeradius/dictionary.vqp
including dictionary file /etc/freeradius/3.0/dictionary
including configuration file /etc/freeradius/3.0/radiusd.conf
including configuration file /etc/freeradius/3.0/proxy.conf
including configuration file /etc/freeradius/3.0/clients.conf
including files in directory /etc/freeradius/3.0/mods-enabled/
including configuration file /etc/freeradius/3.0/mods-enabled/dynamic_clients
including configuration file /etc/freeradius/3.0/mods-enabled/pap
including configuration file /etc/freeradius/3.0/mods-enabled/chap
including configuration file /etc/freeradius/3.0/mods-enabled/exec
including configuration file /etc/freeradius/3.0/mods-enabled/mschap
including configuration file /etc/freeradius/3.0/mods-enabled/linelog
including configuration file /etc/freeradius/3.0/mods-enabled/detail
including configuration file /etc/freeradius/3.0/mods-enabled/echo
including configuration file /etc/freeradius/3.0/mods-enabled/logintime
including configuration file /etc/freeradius/3.0/mods-enabled/ntlm_auth
including configuration file /etc/freeradius/3.0/mods-enabled/unix
including configuration file /etc/freeradius/3.0/mods-enabled/cache_eap
including configuration file /etc/freeradius/3.0/mods-enabled/digest
including configuration file /etc/freeradius/3.0/mods-enabled/realm
including configuration file /etc/freeradius/3.0/mods-enabled/preprocess
including configuration file /etc/freeradius/3.0/mods-enabled/attr_filter
including configuration file /etc/freeradius/3.0/mods-enabled/sradutmp
including configuration file /etc/freeradius/3.0/mods-enabled/eap
including configuration file /etc/freeradius/3.0/mods-enabled/expr
including configuration file /etc/freeradius/3.0/mods-enabled/unpack
including configuration file /etc/freeradius/3.0/mods-enabled/replicate
including configuration file /etc/freeradius/3.0/mods-enabled/always
including configuration file /etc/freeradius/3.0/mods-enabled/files
including configuration file /etc/freeradius/3.0/mods-enabled/utf8
including configuration file /etc/freeradius/3.0/mods-enabled/passwd
including configuration file /etc/freeradius/3.0/mods-enabled/radutmp
including configuration file /etc/freeradius/3.0/mods-enabled/expiration
including configuration file /etc/freeradius/3.0/mods-enabled/soh
including configuration file /etc/freeradius/3.0/mods-enabled/detail.log
including files in directory /etc/freeradius/3.0/policy.d/
including configuration file /etc/freeradius/3.0/policy.d/debug
including configuration file /etc/freeradius/3.0/policy.d/control
including configuration file /etc/freeradius/3.0/policy.d/filter
including configuration file /etc/freeradius/3.0/policy.d/dhcp
including configuration file /etc/freeradius/3.0/policy.d/operator-name
including configuration file /etc/freeradius/3.0/policy.d/rfc7542
including configuration file /etc/freeradius/3.0/policy.d/cui
including configuration file /etc/freeradius/3.0/policy.d/eap
including configuration file /etc/freeradius/3.0/policy.d/moonshot-targeted-ids
including configuration file /etc/freeradius/3.0/policy.d/accounting
including configuration file /etc/freeradius/3.0/policy.d/abfab-tr
including configuration file /etc/freeradius/3.0/policy.d/canonicalization
including files in directory /etc/freeradius/3.0/sites-enabled/
including configuration file /etc/freeradius/3.0/sites-enabled/inner-tunnel
including configuration file /etc/freeradius/3.0/sites-enabled/default
main {
[... 642 lines cut ...]
   	lifetime = 0
   	idle_timeout = 30
   }
}
listen {
  	type = "auth"
  	ipv6addr = ::
  	port = 0
   limit {
   	max_connections = 16
   	lifetime = 0
   	idle_timeout = 30
   }
}
listen {
  	type = "acct"
  	ipv6addr = ::
  	port = 0
   limit {
   	max_connections = 16
   	lifetime = 0
   	idle_timeout = 30
   }
}
Configuration appears to be OK

sagb2015, 2020-05-17:
Hi @AxelLin What is the openssl version on your machine?

AxelLin, 2020-05-18:
> Hi @AxelLin What is the openssl version on your machine?
1.1.1f-1ubuntu2

sagb2015, 2020-05-18:
@AxelLin, we tested with FreeRadius 3.0.12 and 3.0.21 versions and results were different (3.0.12 => no issues and 3.0.21=> same issue as you reported). Another change that helped was enabling TLS12. You can enable the same through menuconfig. The support for the same was recently added and we kept it disabled since we observed a few interop issues with TLSv1.2 enabled. We are in process of debugging those issues.

sagb2015, 2020-05-19:
@AxelLin  Here is one more alternative.
Adding the following lines in freeradius config should help. 
disable_tlsv1 = no
disable_tlsv1_1 = no

These options are set to
disable_tlsv1_2 = no
disable_tlsv1_1 = yes
disable_tlsv1 = yes

by default in the freeradius version 3_0_20 onwards.

AxelLin, 2020-05-20:
Hi @sagb2015 

> @AxelLin Here is one more alternative.
> Adding the following lines in freeradius config should help.
> disable_tlsv1 = no
> disable_tlsv1_1 = no

Above changes does not work in my test.

Below is my config:
   tls-config tls-common {
        verify_depth = 0
        ca_path = "/etc/freeradius/3.0/certs"
        pem_file_type = yes
        private_key_file = "/etc/ssl/private/ssl-cert-snakeoil.key"
        certificate_file = "/etc/ssl/certs/ssl-cert-snakeoil.pem"
        ca_file = "/etc/ssl/certs/ca-certificates.crt"
        private_key_password = <<< secret >>>
        dh_file = "/etc/freeradius/3.0/certs/dh"
        fragment_size = 1024
        include_length = yes
        auto_chain = yes
        check_crl = no
        check_all_crl = no
        cipher_list = "DEFAULT"
        cipher_server_preference = no
        ecdh_curve = "prime256v1"
        disable_tlsv1 = no
        disable_tlsv1_1 = no
        tls_max_version = "1.2"
        tls_min_version = "1.0"
    cache {
        enable = no
        lifetime = 24
        max_entries = 255
    }

BTW, in default configuration it already comment out below lines.
        #       disable_tlsv1_2 = no
                #disable_tlsv1_1 = yes
                #disable_tlsv1 = yes
So I think the default setting actually is the same as your setting.

sagb2015, 2020-05-20:
@AxelLin Is TLS1.2 enabled or disabled in your sdkconfig?

AxelLin, 2020-05-20:
Hi @sagb2015 
It does not work no matter the CONFIG_WPA_TLS_V12 is set or not.
I still got : (212) eap_peap: >>> send TLS 1.3  [length 0002]

negativekelvin, 2020-05-20:
> This is fixed by FreeRadius 3.0.17 with tls_max_version = "1.2" **in the eap module configuration**.

sagb2015, 2020-06-09:
HI @AxelLin  Do you still see the issue with tls_max_version = "1.2" in eap module config?

AxelLin, 2020-06-09:
hi @sagb2015 ,
Hi
It does not work even with tls_max_version = "1.2" in eap module config.
Does it really work for you by setting tls_max_version = "1.2" on Ubuntu 20.04.

AxelLin, 2020-06-16:
> @AxelLin, we tested with FreeRadius 3.0.12 and 3.0.21 versions and results were different (3.0.12 => no issues and 3.0.21=> same issue as you reported). Another change that helped was enabling [...]

Hi @sagb2015 
From your above response, I suppose you can test the issue.
However, all your suggestions (enabling TLS12, set disable_tlsv1 = no disable_tlsv1_1 = no, set tls_max_version = "1.2" ) does not work in my test.
So I'm wondering how is the status? Does it really work for you by setting tls_max_version = "1.2"?

Also how is the status of interop issues with TLSv1.2 enabled?

AxelLin, 2020-07-17:
> Hi @AxelLin , our code does not support TLS 1.3 yet. However, we are planning to port the TLS functionality over mbedtls. So it may be possible to support it in near future.

@sagb2015 
Will IDF-v4.3 include TLS1.3 support?

sagb2015, 2020-07-22:
@AxelLin  Apologies for the late reply! We have added port of TLS using mbedtls recently, that should help you with TLS 1.2. I already see that you are trying to test with it since you raised a few issues. 

Regarding TLS 1.3, we do not have any plans yet. It has not matured enough yet ( eg a few issues https://github.com/FreeRADIUS/freeradius-server/issues/2385).

AxelLin, 2020-07-22:
@sagb2015 

I will re-test again once I got fix for CONFIG_WPA_MBEDTLS_CRYPTO=y issue.
I'm wondering why my phone and my laptop does not hit the TLS1.3 issue, only esp-idf test fails.
(I don't think my phone support TLS1.3, I still feel it's something wrong in esp-idf).

Also I'm wondering if you have any configuration on esp-idf works with FreeRADIUS on Ubuntu 20.04?
I asked this because I tried all your suggestions but they don't work anyway, so does it work for you?

sagb2015, 2020-07-22:
Hi @AxelLin Here is the FreeRadius config that we use for testing - [config.txt](https://github.com/espressif/esp-idf/files/4961077/config.txt)

AxelLin, 2020-07-22:
@sagb2015 

In v4.3-dev-472-gcf056a7d0b90, the config WPA_TLS_V12 no longer exist.
So what is the status about TLSv1.2, is TLS v1.2 enabled or disabled now?

AxelLin, 2020-07-23:
@sagb2015 

Tested on v4.3-dev-472-gcf056a7d0b90 with CONFIG_WPA_MBEDTLS_CRYPTO=n.
I tried your config settings but it still fails, attached my log.
My freeradius (3.0.20+dfsg-3build1) is standard package from ubuntu-20.04, what's your distro and freeradius version?

[freeradius_log.txt](https://github.com/espressif/esp-idf/files/4963441/freeradius_log.txt)

sagb2015, 2020-07-24:
@AxelLin  We  tried using same free-radius server version are able to reproduce it now. The fix is  under review and should be available soon on github.

sagb2015, 2020-07-24:
Additionally, this issue should not be seen with CONFIG_WPA_MBEDTLS_CRYPTO=y.

AxelLin, 2020-08-03:
It's fixed in v4.3-dev-771-gc77c4ccf6c43, thanks.

closed 2020-08-03 by AxelLin

do1fmd, 2021-03-13:
I also had this problem after upgrading to Ubuntu 20.04:

> (97) eap_peap: <<< recv TLS 1.3 [length 0037]
> (97) eap_peap: >>> send TLS 1.0 Alert [length 0002], fatal protocol_version
> (97) eap_peap: ERROR: TLS Alert write:fatal:protocol version
> tls: [...]

As a workaround, I changed the cipher_list in /etc/freeradius/3.0/mods-enabled/eap to the following:
`                cipher_list = "DEFAULT@SECLEVEL=1"`
This enables TLS 1.0 in OpenSSL for EAP-Requests.

AxelLin, 2021-03-13:
@do1fmd
It's working for me with Ubuntu 20.04 now.
If you still hit issue, you probably need to report a new issue.

do1fmd, 2021-03-13:
@AxelLin 
It is working for me with the workaround mentioned above.
I just wanted to share my solution.

AxelLin, 2021-03-13:
My point is it should work even without the workaround.
Some FreeRADIUS may disable TLS 1.0, then you will hit the same issue again. Had batter to figure out that is wrong.

kapilkedawat, 2021-03-13:
Hi @do1fmd ,

Please upgrade your idf version to https://github.com/espressif/esp-idf/commit/50b4cf92860d99cdd3815793ac29b5e40cdb2323 or later, My suggestion will be not to use the workaround you mentioned(basically we should be using TLS1.2 unless your authentication server deployment is very old).

Those changes along with CONFIG_WPA_MBEDTLS_CRYPTO=y force ESP32 to use much more secure ciphers during TLS session.

In case you still face any issues, feel free to open up a new issue. Additionally, you can validate server config by connecting other WLAN clients besides ESP32.