espressif/esp-idf #5274: uart txfifo random data after esp_restart

RonvanderPlas, 2020-05-12:
## Environment

- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.2-dev-1303-gcd4fa46
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- IDE: VSCode
- Power Supply: external 5V

## Problem Description

The UART TX FiFo seems to miss align it's pointer after calling `esp_restart()`. This was first noticed when trying to send messages(various lengths) over UART, after a restart some random data was put out but still containing the right length of the expected message. After that the buffer would be miss aligned and only half the message and half the previous message would be send out, while still using the right data length.
I did stripped my code down to bare minimum to test this problem further and reproduce the issue.

### Steps to reproduce

1. Create task that sends data over UART every 2 seconds
2. After 1 minute call esp_restart()
3. On the UART output some garbage data would be noticeable around the restart

## Output log

```cpp
(11:41:35.692) 11h 22h 33h 44h
(11:41:37.792) 11h 22h 33h 44h
(11:41:39.892) 11h 22h 33h 44h
(11:41:41.992) 11h 22h 33h 44h
(11:41:44.092) 11h 22h 33h 44h
(11:41:45.344) 30h D0h E1h 2Fh 7Dh DEh 59h 6Fh // restart with garbage data on first 2 messages
(11:41:47.443) 11h 22h 33h 44h
(11:41:49.543) 11h 22h 33h 44h
(11:41:51.643) 11h 22h 33h 44h
(11:41:53.743) 11h 22h 33h 44h
(11:41:55.844) 11h 22h 33h 44h
(11:41:57.943) 11h 22h 33h 44h
(11:42:00.043) 11h 22h 33h 44h
(11:42:02.143) 11h 22h 33h 44h
(11:42:04.243) 11h 22h 33h 44h
(11:42:06.343) 11h 22h 33h 44h
(11:42:08.444) 11h 22h 33h 44h
(11:42:10.543) 11h 22h 33h 44h
(11:42:12.643) 11h 22h 33h 44h
(11:42:14.743) 11h 22h 33h 44h
(11:42:16.843) 11h 22h 33h 44h
(11:42:18.943) 11h 22h 33h 44h
(11:42:21.043) 11h 22h 33h 44h
(11:42:23.143) 11h 22h 33h 44h
(11:42:25.243) 11h 22h 33h 44h
(11:42:27.345) 11h 22h 33h 44h
(11:42:29.443) 11h 22h 33h 44h
(11:42:31.543) 11h 22h 33h 44h
(11:42:33.643) 11h 22h 33h 44h
(11:42:35.743) 11h 22h 33h 44h
(11:42:37.843) 11h 22h 33h 44h
(11:42:39.943) 11h 22h 33h 44h
(11:42:42.042) 11h 22h 33h 44h
(11:42:44.143) 11h 22h 33h 44h
(11:42:45.394) 11h 22h 33h 44h 11h 22h 33h 44h // restart 
(11:42:47.494) 30h D0h E1h 2Fh // garbage data on third and fourth message
(11:42:49.595) 7Dh DEh 59h 6Fh
(11:42:51.694) 11h 22h 33h 44h
(11:42:53.794) 11h 22h 33h 44h
(11:42:55.894) 11h 22h 33h 44h

```
### Code to reproduce this issue

```cpp
static void restart_task()
{
    const char buffer[4] = {0x11,0x22,0x33,0x44};
    uint32_t timer=0;
    while(1)
    {
        if(esp_log_timestamp() > timer)
        {
            timer = esp_log_timestamp()+2000;
            uart_write_bytes(MCU_UART_NUM,buffer,4);
        }
        if(esp_log_timestamp() > 60000)
        {
            ESP_LOGI("MAIN","Restarting now");
            esp_restart();
        }
        vTaskDelay(100/portTICK_PERIOD_MS);

    }
}
void app_main()
{

    ESP_LOGI("MAIN","Restart test firmware init");
    const char buffer[4] = {0x11,0x22,0x33,0x44};
    const uart_config_t uart_config =
    {
        .baud_rate = 9600,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };

    ESP_ERROR_CHECK(uart_driver_install(MCU_UART_NUM, UART_RX_BUF_SIZE * 2, 0, 0, NULL, 0));
    ESP_ERROR_CHECK(uart_param_config(MCU_UART_NUM, &uart_config));
    ESP_ERROR_CHECK(uart_set_pin(MCU_UART_NUM, MCU_UART_TX_PIN, MCU_UART_RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
    
    
    uart_write_bytes(MCU_UART_NUM,buffer,4);

    xTaskCreate(&restart_task, "Reboot_task", 2 * 1024, NULL, 5, NULL);
}
```

## Workaround / possible fix
The problem seems to be related to the commit on uart.c adding a txfifo reset.
https://github.com/espressif/esp-idf/commit/886745326ebe9bafccde1a570d6cb0995c21dcaa

By removing the `uart_hal_txfifo_rst(&(uart_context[uart_num].hal));` the output doesn't contain garbage data and behaves as expected

koobest, 2020-05-12:
Hi, @RonvanderPlas
This is a know issue, we have create a MR to fix this issue. will be merged into master soon.

thanks !!

AxelLin, 2020-06-03:
> Hi, @RonvanderPlas
> This is a know issue, we have create a MR to fix this issue. will be merged into master soon.
> 
> thanks !!

Seems the fix is still not yet available?

Alvin1Zhang, 2020-06-03:
@AxelLin Thanks for the updates, the fix has been merged, but will take a while to be available on GitHub, will update once the fix is available.

AxelLin, 2020-06-03:
Hi @Alvin1Zhang 
Thanks for the info.
It's not to push you, but just want to point out if the response is it will
be merged into master *soon*, it will be surprised the fix is not available
after waiting for 3 weeks. If it's not so clear when the fix will be available
but you indeed have the fix already, maybe post it earlier so people can verify
it.

koobest, 2020-06-03:
Hi, 
Same with issue:  https://github.com/espressif/esp-idf/issues/1202#issuecomment-633523139

Please test this fix temporarily.

Alvin1Zhang, 2020-06-03:
@AxelLin Thanks for the details and sorry for the trouble brought to you. Thanks for the suggestion and appreciate the verification of the fix.

Alvin1Zhang, 2020-08-05:
Thanks for reporting and sorry for slow turnaround, the fix has been available at https://github.com/espressif/esp-idf/commit/a7e0cd656c1ba1b0213dcf0e0278d3098f7d40b2, feel free to reopen if the issue still happens. Thanks.

closed 2020-08-05 by Alvin1Zhang