espressif/esp-idf #5276: Error in TCP connection (ERR_EMPTY_RESPONSE)

pucmelda, 2020-05-12:
Hi,
I have project with ESP32-WROOM-32. It reads data from the serial line and provides it on the website. It also acts as a TCP / serial line converter on another port. I use Mongoose server for web server functions. Everything works, but somewhere and sometimes an error occurs during communication - after opening a TCP connection, this connection is terminated on the server side - in response, the TCP flag RST is set (read using Wireshark). Chrome will display an ERR_EMPTY_RESPONSE error. More than 100 products are already in use, this problem only appears somewhere. But in some places almost daily. The problem occurs both with Wifi connection and with wired Ethernet connection (it is implemented by LAN8720 circuit). Listing with simple_monitor will not show anything, even in VERBOSE mode. To eliminate the error, disconnect the network connection is enough - restart the wroom or restart the router or unplug and plug in the Ethernet cable. This problem persists over different versions of esp-idf and mongoose - the first version was esp-idf-v2.1.1 and mongoose 6.14, then esp-idf-v3.3 and mongoose 6.14, and now esp-idf-v4.0 and mongoose 6.17. I don't know if to look for the source of the problem in esp-idf or in mongoose.c. Please advise what this may cause or where to look for the source of the error. Thanks

david-cermak, 2020-05-13:
Hi @pucmelda 

I assume the Mongoose server uses socket interface of esp-idf, correct? Is it possible to share the standard IDF logs (sorry, not sure how the simple-monitor is different)? Could you modify IDF sources or is the IDF supplied as a library to Mongoose?

If yes, could you please choose one IDF version ('d prefer v4.0) and
* enable socket debug logging in lwip [here](https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/include/lwip/opt.h#L3333) set to `LWIP_DBG_ON`
* share the idf monitor logs (debug level is enough)
* share the wireshark log

PS: Sorry for asking stupid questions, I'm not familiar with Mongoose

pucmelda, 2020-05-14:
Hi,
thanks for help. I enabled socket debug logging in lwip and will wait for error to get IDF log and wireshark output.
I use esp-idf v4.0 on esp32_win32_msys2_environment_and_toolchain-20181001. Mongoose uses socket interface of esp-idf, it is just two files - mongoose.c and mongoose.h added to my project. Standard IDF logs are not problem, I only do not use make monitor command to prevent restart of device, when I connected it to notebook.
So, when the error occurs, I will send logs.

pucmelda, 2020-05-16:
Hi,
now I attach IDF log and Wireshark log of error - it is not possible to get www page, chrome show error ERR_EMPTY_RESPONSE. Socket debug logging is enabled. I add to my code restart of eth on jumper - just call esp_eth_stop and after event ETHERNET_EVENT_STOP call esp_eth_start and tcpip_adapter_dhcpc_stop and tcpip_adapter_set_ip_info to set static IP address. This will help immediately and everything work. It is on line 93 in IDF log. IP address of my computer is 192.168.20.112 and IP address of wroom is 192.168.20.30. I attach Wireshark in text format - just packet summary line - I do not know, how to attach pcapng file.

[WireShark 192.168.20.30 web nefunkcni 03.txt](https://github.com/espressif/esp-idf/files/4638302/WireShark.192.168.20.30.web.nefunkcni.03.txt)

[output_2020-05-15_17-23-05.log](https://github.com/espressif/esp-idf/files/4638295/output_2020-05-15_17-23-05.log)

david-cermak, 2020-05-17:
@pucmelda Thanks for sharing the logs. (You can attach the pcap file enclosed in [a zip archive](https://help.github.com/en/github/managing-your-work-on-github/file-attachments-on-issues-and-pull-requests) )

Just to double check if understand correctly: The actual problem occurred *before* the line 93 of the attached log file, correct? (so the rest of the file is just the restarting procedure you've described above?) As for the Wireshark log, the problematic packets are nr. 27, 19, 34, correct? (responding with [RST, ACK] to http's GET)

pucmelda, 2020-05-18:
Hi,
yes, problem occured before the line 93 of the IDF log and problematic packet are nr.27,29 and 34 on Wireshark log.
Device was in error state, when I connected to it for getting log. After I connect and start record IDF log and Wireshark log, I try to get web page via Chrome - I got error ERR_EMPTY_RESPONSE. Then I insert the jumper to stop and again start eth interface - and after it everything works.
Now I can attach next log from other device (from another network) - IP address of device is 192.168.1.250 and it is in error state. I tried to get web page (port 80) and make TCP connection on port 10001 - both failed. I do not reset connection in this case.
[WireShark 192.168.1.250 web+tcp nefunkcni.pcapng.zip](https://github.com/espressif/esp-idf/files/4642962/WireShark.192.168.1.250.web%2Btcp.nefunkcni.pcapng.zip)
[output_2020-05-18_08-31-29.log](https://github.com/espressif/esp-idf/files/4642963/output_2020-05-18_08-31-29.log)

david-cermak, 2020-05-18:
@pucmelda Thanks for the confirmation, in that case the idf-monitor you've posted previously (output_2020-05-15_17-23-05.log) seems to miss the logs from the actual issue (before line 93). Do you still have some original log? I'm interested mainly in logs related to packet nr.20 and further. This should be something very similar to (must be local port=*49321*):
 ```
lwip_recvfrom(58):  addr=192.168.20.112 port=49321 len=-1
```
There're logs like that your output file, but these correspond to opening other connections made after the restart (starting from packet nr.117) and these connections succeeded...

Looking at the new logs, the socket info is still missing. Packet capture shows that probably the speculative connections were disabled (? -- was going to suggest the same, but apparently it didn't help...) since the http packet always preceded initiation of another tcp connection. Anyway, the result looks still the same, connections being actively closed from the ESP side -- the socket logs should reveal more.

pucmelda, 2020-05-18:
Hi, the logs are original - there are no lines lwip_recvfrom in error state - I do not know why.
I can't get logs when an error occurs - the device is in use, and when an error occurs, the customer calls me (sometimes 3 times a day, sometimes once a week) - then I can come there and connect with a laptop and read the log (which I attached here ). So in the error state, the software probably won't get in the code to the place where the lwip_recvfrom message is sent, but it will close the connection using the RST flag before - why, I have no idea.

david-cermak, 2020-05-18:
@pucmelda Would it be possible to keep a laptop (or anything, might be a raspberry pi) connected to the device with logging enabled (Ctrl+T+L makes the idf monitor log to a file)? If not we have to find some other way to redirect stdio and save to flash for example, but that might be more complicated. I'm sorry but having the log (of the actual issue) is essential for understanding the problem.
Agree that the `lwip_recvfrom()` might not have been present, but the `lwip_accept(...)...` must have, as the connection was indeed accepted as seen from the packet capture.

pucmelda, 2020-05-18:
Ok, I will try to make a deal with him about permanent laptop connection

pucmelda, 2020-05-19:
Hi,
so I have log of device, where error occured. I connected the laptop around 7 pm - everything worked. Around 11 pm the owner found out that he had an ERR_EMPTY_RESPONSE error in Chrome - so he restarted the connection with a jumper - this is line 154434 in the log. It was ETHERNET_EVENT_STOP event, then ETHERNET_EVENT_START event, and everything worked until now. So the error occurred sometime before line 154434. I added logs with the etatherm tag to test my code, these are fine, I hope you don't mind finding the error.
[log.etatherm.20200518181230-2.txt.zip](https://github.com/espressif/esp-idf/files/4649150/log.etatherm.20200518181230-2.txt.zip)

david-cermak, 2020-05-19:
Hi @pucmelda 

Thanks for sharing the log (would be great to have a related pcap to match the logs with packets), but yes, I can see the issue on line 152568:
```
lwip_accept(56) returning new sock=63 addr=192.168.20.170 port=51200
...
lwip_select(64, 0x3ffea250, 0x3ffea248, 0x3ffea240, tvsec=1 tvusec=0)
lwip_select: timeout expired
```
(now expecting that the related session was closed as previously -- would've seen [ACK+RST] on port 51200 in wireshark) we could assume that no data was received on the socket 63
(timeouted after a second, but previous packet log showed [syn+ack] and [ack+push] within milliseconds). This indicates that, very likely, the issue is *NOT* in the webserver!

This (and some other hints like out-of-sequence acks) points us to lower layers of lwIP, which is hard to believe, unless there's an issue 
* with interface drivers -- have noticed the logs of `eth1_frame_make error -2` -- it this expected? But chances are low since you've said the same is reproducible on WiFi interface, too.
* some weird memory corruption -- very unlikely too, as the problem happening with multiple versions of IDF

Could you please share your sdkconfig, as well?
Would it be possible to share some sketch of an application, that could be used to reproduce the issue locally? As far as I understand you haven't been able to reproduce the failure locally, correct -- just happening on customer side?

pucmelda, 2020-05-21:
Hi, I don't have a WireShark log because the device is in a basement where there is no network. If necessary, I think we could somehow connect the laptop to the network to get the WireShark logo. But I also think we will see [ACK + RST] from esp. I have no idea where it happens in the ESP-IDF code - I don't see it that deep.
 As for the error history - at first the software only worked as a UART to Wifi converter. Web pages were for setup purposes only. It worked - somewhere there were some problems, but we didn't do an in-depth analysis of the source of the error. Then we added web pages for displaying data and the incidence of the error increased - then we switched from IDF v.2.2.1 to IDF v.3.3 (in November 2019). The error continued to appear somewhere, mostly during the Wifi connection - we found that it only occurs somewhere. Enough applications run stably, only somewhere this error occurs regularly - to eliminate it, disconnect the device and connect or restart either the device or the router is enough, and everything goes again for some time. Somewhere it help to replace the router. It also seems that the frequency of the error is affected by network traffic - at Christmas time, and especially after the beginning of coronavirus quarantine, when children (and their youtube, minecraft ...) significantly increased network traffic, the error began to appear much more often (again only in some networks). We believed that this was just a bit of incompatibility between some routers and ESP, and that a new version of ESP-IDF or the use of newer routers would solve it. But rather it got worse - by switching to ESP-IDF v4.0 and for me specifically by replacing the router from TP-Link WR841N to TP-Link Archer C5, the frequency of this error increased several times. Now it occurs at least once a week, before that once a month. For a colleague (where the logs come from) it is even more pronounced - the error now occurs several times a day. Conversely, in many places the device runs for a long time without a single error (for example in our office - you can look at http://85.163.236.6:50000, the server password is etatherm). I have no idea why.

[364a.zip](https://github.com/espressif/esp-idf/files/4661687/364a.zip)
I am attaching the whole directory of my application, I have deleted the functions related to communication on the UART, the network part is there in its entirety. I really don't know what to do with it.

david-cermak, 2020-05-22:
@pucmelda Thanks for sharing the application, that's very helpful! Will try to reproduce the issue locally, but it may take some time. In the meantime, could you please check if you are able to connect to that server from the ESP itself after this issue appears? This could at least help ruling out the network stack and/or interface drivers.

To achieve that simply run the below function from a thread every 5/10 seconds.
``` c
static void sock_test()
{
    char rx_buffer[128] = {0};
    int ret, sock_errno;
    struct sockaddr_in addr = {     .sin_addr = { .s_addr = 0x100007f},
                                    .sin_family = AF_INET,
                                    .sin_port = htons(80) };
    const char *payload = "GET / HTTP/1.1\r\n\r\n";

    int sock = socket(AF_INET, SOCK_STREAM, IPPROTO_IP);
    if (sock < 0) {
        ESP_LOGE(TAG, "Unable to create socket: errno %d",errno);
        goto failed;
    }
    ESP_LOGI(TAG, "Socket created");
    ret = connect(sock, (struct sockaddr *)&addr, sizeof(addr));
    if (ret != 0) {
        ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);
        goto failed;
    }
    ESP_LOGI(TAG, "Successfully connected");
    ret = send(sock, payload, strlen(payload), 0);
    if (ret < 0) {
        ESP_LOGE(TAG, "Error while sending: errno %d", errno);
        goto failed;
    }

    ret = recv(sock, rx_buffer, sizeof(rx_buffer) - 1, 0);
    uint32_t opt = sizeof(sock_errno);
    getsockopt(sock, SOL_SOCKET, SO_ERROR, &sock_errno, &opt);
    ESP_LOGI(TAG, "[recv sock=%d]  ret=%d errno=%d\n", sock, ret, sock_errno);    
    if (ret < 0) {
        ESP_LOGE(TAG, "Error while receiving: errno %d", errno);
    } else if (ret > 0){
        rx_buffer[ret] = 0; // Null-terminate whatever we received and treat like a string
        ESP_LOGI(TAG, "Received %d bytes:", ret);
        ESP_LOGI(TAG, "%s", rx_buffer);
    } else {
        ESP_LOGE(TAG, "Closed connection during receiving, ret=%d, errno=%d", ret, errno);
    }

failed:
    close(sock); 
}
```
So normally, you would get info logs on "Receive XX bytes", but after the error occurred I'd expect to see the error message "Closed connection during receiving...". No need to keep a laptop connected, actively logging, just capture the logs after the issue appears.
Also please enable other tcp debugging options:
```
#define SOCKETS_DEBUG                   LWIP_DBG_ON
#define TCP_INPUT_DEBUG                 LWIP_DBG_ON
#define TCP_OUTPUT_DEBUG                LWIP_DBG_ON
#define TCPIP_DEBUG                     LWIP_DBG_ON
```

It would be very helpful, if you can get the logs on this repeating loopback connection after the error occurred. Thanks!

pucmelda, 2020-05-22:
Ok, i will do it.

> Od: "david-cermak" <notifications@github.com>
> Komu: "espressif/esp-idf" <esp-idf@noreply.github.com>
> Datum: 22.05.2020 10:16

@pucmelda Thanks for sharing the application, that's very helpful! Will try to reproduce the issue locally, but it may take some time. In the meantime, could you please check if you are able to connect to that server from the ESP itself after this issue appears? This could at least help ruling out the network stack and/or interface drivers.
To achieve that simply run the below function from a thread every 5/10 seconds.
static void sock_test()
{
    char rx_buffer[128] = {0};
    int ret, sock_errno;
    struct sockaddr_in addr = {     .sin_addr = { .s_addr = 0x100007f},
                                    .sin_family = AF_INET,
                                    .sin_port = htons(80) };
    const char *payload = "GET / HTTP/1.1\r\n\r\n";

    int sock = socket(AF_INET, SOCK_STREAM, IPPROTO_IP);
    if (sock < 0) {
        ESP_LOGE(TAG, "Unable to create socket: errno %d",errno);
        goto failed;
    }
    ESP_LOGI(TAG, "Socket created");
    ret = connect(sock, (struct sockaddr *)&addr, sizeof(addr));
    if (ret != 0) {
        ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);
        goto failed;
    }
    ESP_LOGI(TAG, "Successfully connected");
    ret = send(sock, payload, strlen(payload), 0);
    if (ret < 0) {
        ESP_LOGE(TAG, "Error while sending: errno %d", errno);
        goto failed;
    }

    ret = recv(sock, rx_buffer, sizeof(rx_buffer) - 1, 0);
    uint32_t opt = sizeof(sock_errno);
    getsockopt(sock, SOL_SOCKET, SO_ERROR, &sock_errno, &opt);
    ESP_LOGI(TAG, "[recv sock=%d]  ret=%d errno=%d\n", sock, ret, sock_errno);    
    if (ret < 0) {
        ESP_LOGE(TAG, "Error while receiving: errno %d", errno);
    } else if (ret > 0){
        rx_buffer[ret] = 0; // Null-terminate whatever we received and treat like a string
        ESP_LOGI(TAG, "Received %d bytes:", ret);
        ESP_LOGI(TAG, "%s", rx_buffer);
    } else {
        ESP_LOGE(TAG, "Closed connection during receiving, ret=%d, errno=%d", ret, errno);
    }

failed:
    close(sock); 
}
So normally, you would get info logs on "Receive XX bytes", but after the error occurred I'd expect to see the error message "Closed connection during receiving...". No need to keep a laptop connected, actively logging, just capture the logs after the issue appears.
Also please enable other tcp debugging options:
#define SOCKETS_DEBUG                   LWIP_DBG_ON
#define TCP_INPUT_DEBUG                 LWIP_DBG_ON
#define TCP_OUTPUT_DEBUG                LWIP_DBG_ON
#define TCPIP_DEBUG                     LWIP_DBG_ON

It would be very helpful, if you can get the logs on this repeating loopback connection after the error occurred. Thanks!

—
[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/espressif/esp-idf/issues/5276#issuecomment-632563130",
"url": "https://github.com/espressif/esp-idf/issues/5276#issuecomment-632563130",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

pucmelda, 2020-05-23:
Hi, I attach the log with function sock_test() in error state - the message from this function is "Unable to create socket: errno 23". After a while of logging I restart ethernet connection by jumper (in line 1712 on log), and from now it is working again ("Socket created","Successfully connected","[recv sock=58]  ret=127 errno=0" and "Received 127 bytes") - web pages start working too.
[output_2020-05-23_10-58-05.zip](https://github.com/espressif/esp-idf/files/4671493/output_2020-05-23_10-58-05.zip)

david-cermak, 2020-05-25:
Hi @pucmelda 

Thanks again for sharing this log! It seems my initial guess was wrong and this looks as simple as socket exhaustion issue -- sorry haven't noticed from the previous logs.

Could you please try to connect from multiple endpoints (not necessarily ip addresses, just sessions == terminals) trying to open a socket and not initialing the http request, just keeping that socket open, using for example netcat:
```
nc esp_address 80
```
If opening about 8 such connections, you should be able to simulate the issue -- based on the config option:
```
(10) Max number of open sockets
```

Could you please confirm that this is the problem you are seeing?

pucmelda, 2020-05-25:
Hi,
yes, it looks that this is it. I used windows telnet and after opening 6 connections, the issue happend. In Wireshark log I saw that connection was closed (RST+ACK). Interestingly, chrome reported ERR_CONNECTION_RESET instead of ERR_EMPTY_RESPONSE - I do not know why.
As I understand it, this could be the source of the problem - some sockets will not close. But I thought the browser should always close the socket after loading the page - I'll have to find out which sockets remain open. Could it be a feature of the android - when the application is switched or hibernated while loading, the connection will not close? Because we mainly use android phones to browse the pages of device.
Is it possible to somehow find out and list open connections (from which IP address and port) in the code? And force socket closure after some time?

david-cermak, 2020-05-26:
@pucmelda Yes, it is possible to list all the sockets in use. The numbers of socket descriptors depend on these compile time settings:
```
LWIP_SOCKET_OFFSET
CONFIG_LWIP_MAX_SOCKETS
```
so you can iterate over the list of all the sockets and get remote connection info if the socket active (API return nonzero), for example
``` c
for (int i=0; i<=CONFIG_LWIP_MAX_SOCKETS; ++i) {
    struct sockaddr_in6 addr;
    socklen_t addr_size = sizeof(addr);
    int sock = LWIP_SOCKET_OFFSET + i;
    int res = getpeername(sock, (struct sockaddr *)&addr, &addr_size);
    if (res == 0) {
        ESP_LOGI(TAG, "sock: %d -- addr: %x, port: %d", sock, addr.sin6_addr.un.u32_addr[3], addr.sin6_port);        
    }
}
```
(note the default config of esp-lwip is IPv4+IPv6 so we get an IP4 mapped address -- that's why printing the last octets of the IPv6 above)

Yes it should be possible to close the socket from the application code, esp-lwip supports it, but not sure how would the webserver cope with the situation.

I think the first step should be to find out sources of these connections. Of course increasing max number of sockets in menuconfig (hoping that eventually the unused sockets shall be closed) and checking if a new socket could be open to restart the webserver otherwise are means to workaround the issue.

> Interestingly, chrome reported ERR_CONNECTION_RESET instead of ERR_EMPTY_RESPONSE

This could be caused by the speculative connection i've mentioned earlier. Some versions of chromium do that by default unless connecting to a local network. I recall seeing both from your wireshark logs, including connections from firefox, creating connections one after another.

pucmelda, 2020-05-26:
Hi, I tried your code for listing sockets, but it jumps to reset for some assert - I do not know what is wrong. Can you help me?
I (12826) etatherm: sock: 54 -- addr: 400fe380, port: 45504
I (12826) etatherm: sock: 55 -- addr: 400fe380, port: 45248
I (12826) etatherm: sock: 58 -- addr: 400fe380, port: 7921
C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008f46c on core 0
0x4008f46c: xQueueGenericReceive at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/freertos/queue.c:1449 (discriminator 3)
ELF file SHA256: 8d802b05859b801b076629c51f3366622e292da472d999458237d3dfcbd50376
Backtrace: 0x40088390:0x3ffd8120 0x4008875d:0x3ffd8140 0x4008f46c:0x3ffd8160 0x401437f5:0x3ffd81a0 0x40136065:0x3ffd81c0 0x401367b1:0x3ffd81e0 0x401367ce:0x3ffd8200 0x401369ad:0x3ffd8220 0x401
37ffe:0x3ffd8280 0x400d873c:0x3ffd82a0 0x400d280f:0x3ffd8430
0x40088390: invoke_abort at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/esp32/panic.c:157
0x4008875d: abort at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/esp32/panic.c:172
0x4008f46c: xQueueGenericReceive at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/freertos/queue.c:1449 (discriminator 3)
0x401437f5: sys_mutex_lock at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/lwip/port/esp32/freertos/sys_arch.c:78 (discriminator 1)
0x40136065: sock_inc_used at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/lwip/lwip/src/api/sockets.c:351
0x401367b1: tryget_socket_unconn at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/lwip/lwip/src/api/sockets.c:468
 (inlined by) tryget_socket at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/lwip/lwip/src/api/sockets.c:501
0x401367ce: get_socket at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/lwip/lwip/src/api/sockets.c:520
0x401369ad: lwip_getaddrname at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/lwip/lwip/src/api/sockets.c:2793
0x40137ffe: lwip_getpeername at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/lwip/lwip/src/api/sockets.c:2834
0x400d873c: getpeername at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/lwip/lwip/src/include/lwip/sockets.h:630
 (inlined by) app_main at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/364b21/main/etatherm.c:3681
0x400d280f: main_task at C:/Jednocip/Espressif/msys32-20191231/home/pch/esp/esp-idf-v4.0/components/esp32/cpu_start.c:554

Rebooting...
ets Jun  8 2016 00:22:57

david-cermak, 2020-05-26:
@pucmelda Off by one issue I guess:
``` c
for (int i=0; i<CONFIG_LWIP_MAX_SOCKETS; ++i) {
```
Sorry for that, haven't checked the snippet before posting it. Also the address you're getting seems like an IPv6 link local in your case, just update the print part if needed.

pucmelda, 2020-05-26:
Hi,
this is not the problem, it still behaves the same - an error occurs in the call to the getpeername function. This most often occurs on the second call (sock = 55). I run the cycle in the app_main function a few seconds after initialization - I don't know why at least 2 sockets are already active.
I add the log. 
[log.etatherm.20200526174431.txt.zip](https://github.com/espressif/esp-idf/files/4683746/log.etatherm.20200526174431.txt.zip)

david-cermak, 2020-05-27:
@pucmelda Please run this code once after `tcpip_adapter_init()`(before listing the sockets):
``` c
    int sockets[CONFIG_LWIP_MAX_SOCKETS];
    for (int i = 0; i < CONFIG_LWIP_MAX_SOCKETS; i++) {
        int type = (i % 2 == 0) ? SOCK_DGRAM : SOCK_STREAM;
        int family = (i % 3 == 0) ? PF_INET6 : PF_INET;
        sockets[i] = socket(family, type, IPPROTO_IP);
    }
    for (int i = 0; i < CONFIG_LWIP_MAX_SOCKETS; i++) {
        close(sockets[i]);
    }
```
This is just to allocate the lwIP structures (and the locks!) for each socket. I'm sorry have missed the fact that this is needed in `v4.0`.

pucmelda, 2020-05-27:
Hi, thank you, it works now. But I have 2 questions:
1. why two sockets are always open (sock 54,55), even if no one is communicating with the device?
2. the ip address and port values ​​are strange - the communication is between a PC with the IP address 192.168.1.106 and a device with the IP address 192.168.1.252, but the listing is 3ffeb744, which should be the IP address 63.254.183.68. In addition, the port numbers are not the same as the WireShark logo. So how is it?
I put a list of sockets in the code in the main loop (prefix main) once in a while and in the function of processing the request for a web page (prefix handle_web).
The relevant part of the IDF log and WireShark log attached.
[logsock01.txt](https://github.com/espressif/esp-idf/files/4688602/logsock01.txt)
[WireShark.192.168.1.252 01.txt](https://github.com/espressif/esp-idf/files/4688603/WireShark.192.168.1.252.01.txt)

david-cermak, 2020-05-27:
@pucmelda 

> why two sockets are always open 

My guess is that these are the sockets used by your webserver. To find out you can place a breakpoint to `socket()` in lwip/socket.c and see the code creating them (if no debugger you can use `esp_backtrace_print(levels);` to print out current call stack -- see esp_debug_helpers.h)

> the ip address and port values ​​are strange

The snippet I've shared was meant to merely demonstrate the method of listing endpoints and to be tweaked to specific network needs. So the ports are in network format, need to use `hton()` to convert back (52966=0xcee6, 0xe6ce=59086) and the addressed you're getting are IPv6 link locals, so need to print all octets:
``` c
..., addr.sin6_addr.un.u32_addr[0], addr.sin6_addr.un.u32_addr[1], addr.sin6_addr.un.u32_addr[2], addr.sin6_addr.un.u32_addr[3]);
```

pucmelda, 2020-05-29:
Thanks a lot. It looks like the problem really arises when browsing from an Android device - when it fall asleep, turn off or switch the network, the connection remains open. It probably happened so often to a colleague, because he has a large garden with several Wifi subnets, where the phone jumps between them when moving. I have no idea why the Mongoose server probably doesn't have active functionality like Socket Timeout, or it has it turned off. I'll try to look in the documentation again.
Back to ESP sockets - interestingly, when CONFIG_LWIP_MAX_SOCKETS is set to 10, only 8 slots are used - two are simply never used - I don't know why. I increased CONFIG_LWIP_MAX_SOCKETS to 16 and added a code to monitor the communication - when the connection is inactive for 120 seconds, I will disconnect it. We will test now, but I think this was really the source of the error. Thanks again for the help, I'll write how the testing turned out.

pucmelda, 2020-06-04:
Hi,
it looks that error in TCP connection (ERR_EMPTY_RESPONSE) is solved - when I close innactive connection after 120s of innactivity, everythink works good. Interestingly, some inactive connections are from our Android devices, but some are unknown foreign IP addresses (such as 89.248.174.151, 108.184.117.1 ...). Probably some web robots ...

Please one more help - before we figured out this cause, I made software that regularly disconnected from the network and then connected every 20 minutes. For Ethernet (esp_eth_stop (), then esp_eth_start (), tcpip_adapter_dhcpc_stop (), and tcpip_adapter_set_ip_info ()), everything works fine. For wifi, I did it using esp_wifi_disconnect () and esp_wifi_connect () - it seemed fine, but it turned out that wifi does not always connect after a few reconnections. I did IDF logs and in my opinion there is a problem in memory allocation -
E (133280163) wifi: esf_buf: t=2 l=76 max:32, alloc:32 no eb, TXQ_BLOCK=0
W (133280163) wifi: alloc eb len=76 type=2 fail, heap:53176
What do I need to do to properly disconnect and reconnect to the Wifi network?
I am attaching a log where the connection does not work:
[serial_20200604_082839a.txt](https://github.com/espressif/esp-idf/files/4728777/serial_20200604_082839a.txt)
and a log, where reconnection works good:
[serial_20200604_091044a.txt](https://github.com/espressif/esp-idf/files/4728783/serial_20200604_091044a.txt)

Can you help me, please?

pucmelda, 2020-06-04:
Now I have learned that the connection problem only arises somewhere. There are applications where the software runs correctly for almost 8 days (over 560x esp_wifi_disconnect (), esp_wifi_connect ()) without problems. I will try to make a log entry at the moment of disconnection and impossibility of reconnection.

david-cermak, 2020-06-08:
@pucmelda 
Sorry for the late reply. These messages indicate that the WiFi library failed to allocate a buffer (approximately 1.6k), it shows that the free heap size is > 50k at the same time. This means that the heap must be very fragmented in this case.

Yes, calling the `esp_wifi_disconnect()` and `esp_wifi_connect()` is the correct way to disconnect and reconnect WiFi.

The issue you're seeing seems to be related to the TCP/IP stack allocating buffers for outgoing data, when the WiFi is disconnected (this is okay, since these buffers get freed eventually if the application has enough heap space, but may cause trouble otherwise for peak heap consumption when WiFi library connecting). I would suggest deinitializing the tcpip adapter at the same time, to have the TCP/IP layer disconnect and release buffers when WiFi is 
disconnected before connecting again to get around this issue.

pucmelda, 2020-06-08:
Hi,
I left the laptop connected there over the weekend - once the device disconnected and no longer connected - the user had to turn the device off / on (line 353863). The software disconnected / connected the wifi every 20 minutes - if it failed to reconnect, it tried every 30 seconds. There are 299 events in the log (system_event: SYSTEM_EVENT_STA_DISCONNECTED) - mostly due to reason 8 (assoc leave) - which is probably a correct response to the esp_wifi_disconnect () function. Several times there is disconnection for reason 15 (4-way keying handshake timeout) followed by reason 205 (other reason). The next time it try, the reconnection succeeds. This is until line 177555, when the connection fails and the reasons 15 and 205 alternate. Until it is switched off / on. Then it again works with the occasional connection error with reason 15 and 205.
This is a completely different reason than in the previous log, where the SYSTEM_EVENT_STA_DISCONNECTED event had reason 2 (auth expire) followed by reason 205 - so there was probably a buffer non-allocation error.
I will try to make even more records from which to find out exactly what is happening and what to do about it.
If you can find out from the log what is wrong and what to do about it, I will be very grateful.
[log.etatherm.20200605140206.txt.zip](https://github.com/espressif/esp-idf/files/4746831/log.etatherm.20200605140206.txt.zip)

david-cermak, 2020-06-09:
Hi @pucmelda 

What I think is going on here are issues related to the logging itself (yes, the Heisenberg's principle :-), because of the excessive logs in `wpa` module for the debug level and the 4-way handshake timeout and the related failures. To avoid these issue you can:

* programmatically raise logging level for certain modules: `esp_log_level_set("wpa", ESP_LOG_INFO);`, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html#_CPPv417esp_log_level_setPKc15esp_log_level_t

* increase the stdout uart's baudrate in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esptoolpy-monitor-baud

What I would suggest in general, though, is to open a separate issue for the WiFi lib issues. This would not only help getting feedback from WiFi experts but also makes it easier for other users to search for related issues in IDF. 

Moreover if asking a general question, please prefer posting a topic in https://www.esp32.com/ since the GitHub issues are supposed to cover and track software issues in IDF repository. If you feel there's no longer an issue in TCP/IP stack in IDF, please consider closing this issue.

pucmelda, 2020-06-09:
Hi,
if it were this way, I don't understand that it works without problems somewhere and not elsewhere - but I will try to apply your advice and test it further. If necessary, I will open a new issue.

I consider the original issue solved, thank you very much for your help.

closed 2020-06-09 by pucmelda