espressif/esp-idf #5277: Erasing a larger memory area in the spi flash triggers the watchdog

ataweg, 2020-05-12:
when the program executes the following snipplet aTG1WDT_SYS_RESET occurs in esp_ota_begin()

```
                  ESP_LOGI( TAG, "Partition type %d, subtype %d, address 0x%08x, size 0x%08x label %s",
                           state->update_partition->type,
                           state->update_partition->subtype,
                           state->update_partition->address,
                           state->update_partition->size,
                           state->update_partition->label );
// !!! here the watchdog triggers
                  err = esp_ota_begin( state->update_partition, OTA_SIZE_UNKNOWN, &state->update_handle );
                  ESP_LOGI( TAG, "here I am" );

```
the corresponding log:
```
I (17:51:35.318) ota(385): Partition type 0, subtype 16, address 0x00130000, size 0x00110000 label ota_0
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:9492
load:0x40078000,len:15524
ho 0 tail 12 room 4
load:0x40080400,len:4368
entry 0x400806bc
I (183) boot: ESP-IDF v4.2-dev-1415-ga2263571b-dirty 2nd stage bootloader
I (183) boot: compile time 17:47:12
D (185) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (192) boot: chip revision: 1
I (196) boot_comm: chip revision: 1, min. bootloader chip revision: 0
D (203) boot.esp32: magic e9
D (206) boot.esp32: segments 04
D (209) boot.esp32: spi_mode 02
D (213) boot.esp32: spi_speed 00
D (216) boot.esp32: spi_size 02
I (219) boot.esp32: SPI Speed      : 40MHz
I (224) boot.esp32: SPI Mode       : DIO
I (228) boot.esp32: SPI Flash Size : 4MB
W (233) boot.esp32: PRO CPU has been reset by WDT.
W (239) boot.esp32: WDT reset info: PRO CPU PC=0x400803c3
0x400803c3: _DoubleExceptionVector at C:/Espressif/esp-idf/components/freertos/xtensa/xtensa_vectors.S:457

D (245) boot.esp32: WDT reset info: PRO CPU STATUS        0x00000000
D (251) boot.esp32: WDT reset info: PRO CPU PID           0x00000000
D (257) boot.esp32: WDT reset info: PRO CPU PDEBUGINST    0x06003003
D (264) boot.esp32: WDT reset info: PRO CPU PDEBUGSTATUS  0x0000003c
D (270) boot.esp32: WDT reset info: PRO CPU PDEBUGDATA    0x400803c3
0x400803c3: _DoubleExceptionVector at C:/Espressif/esp-idf/components/freertos/xtensa/xtensa_vectors.S:457

D (277) boot.esp32: WDT reset info: PRO CPU PDEBUGPC      0x400803c3
0x400803c3: _DoubleExceptionVector at C:/Espressif/esp-idf/components/freertos/xtensa/xtensa_vectors.S:457

D (283) boot.esp32: WDT reset info: PRO CPU PDEBUGLS0STAT 0x00002020
D (290) boot.esp32: WDT reset info: PRO CPU PDEBUGLS0ADDR 0xef963af4
D (296) boot.esp32: WDT reset info: PRO CPU PDEBUGLS0DATA 0x00000000
W (302) boot.esp32: WDT reset info: APP CPU PC=0x40088ea8
0x40088ea8: panic_handler at C:/Espressif/esp-idf/components/esp_system/port/panic_handler.c:493 (discriminator 3)

D (308) boot.esp32: WDT reset info: APP CPU STATUS        0x00000000
D (315) boot.esp32: WDT reset info: APP CPU PID           0x00000000
D (321) boot.esp32: WDT reset info: APP CPU PDEBUGINST    0x05003000
D (328) boot.esp32: WDT reset info: APP CPU PDEBUGSTATUS  0x00000020
D (334) boot.esp32: WDT reset info: APP CPU PDEBUGDATA    0x00000400
D (341) boot.esp32: WDT reset info: APP CPU PDEBUGPC      0x40088ea8
0x40088ea8: panic_handler at C:/Espressif/esp-idf/components/esp_system/port/panic_handler.c:493 (discriminator 3)

D (347) boot.esp32: WDT reset info: APP CPU PDEBUGLS0STAT 0x00000000
D (353) boot.esp32: WDT reset info: APP CPU PDEBUGLS0ADDR 0x3ffb4c00
D (360) boot.esp32: WDT reset info: APP CPU PDEBUGLS0DATA 0x00000000
D (366) boot: Enabling RTCWDT(9000 ms)
I (370) boot: Enabling RNG early entropy source...
D (376) bootloader_flash: mmu set paddr=00010000 count=1 size=c00 src_addr=10000 src_addr_aligned=10000
D (385) boot: mapped partition table 0x10000 at 0x3f400000
D (391) flash_parts: partition table verified, 7 entries

```

I have no idea how I can fix it.
Viele Gruesse
Axel

mahavirj, 2020-05-14:
@ataweg This should be fixed with https://github.com/espressif/esp-idf/commit/b2417819fdbb5d00103b2a0612bb98d09f9e5533. Please update ESP-IDF codebase or try with this fix.

ataweg, 2020-05-14:
The issue is related to the current head revision of esp-idf. So the fix doesn't help here.

mahavirj, 2020-05-15:
Can you please share your application code and sdkconfig?

ataweg, 2020-05-24:
I have found the source of the evil. Since I could observe some correct OTA downloads in the meantime, it was very likely that the cause for the misbehavior is not in the ESP-IDF. After a long debug session I was able to fix the problem. 
The project here is based on [https://github.com/igrr/esp32-cam-demo](https://github.com/igrr/esp32-cam-demo). There in the orignal code of camera.c the vsync interrupt is disabled before the isr handler is registered, which enables the interrupt. Swaping the order of these two steps helps in my project. The fix solves my problem, but I suspect the deeper cause elsewhere. My observations suggest that.
Viele Gruesse
Axel

mahavirj, 2020-05-26:
> The project here is based on https://github.com/igrr/esp32-cam-demo and https://github.com/tekker/esp32-ov7670-hacking 

Can you please raise PR on above project with fix that you created? (CC: @igrr) If this issue is not related to ESP-IDF then maybe we can close it?

igrr, 2020-05-26:
Sorry, I should have archived that repository long ago! There is an officially supported camera driver now, https://github.com/espressif/esp32-camera, please try using that instead.

mahavirj, 2020-06-03:
@ataweg For now I will close this issue as it does not seem to be ESP-IDF specific, but please feel free to re-open in case your observations differ.

closed 2020-06-03 by mahavirj