espressif/esp-idf #5281: unable to use different oob method during provisioning

ankit-thealchemist, 2020-05-13:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitv1
- Kit version [v1|
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    idf 4.0
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [|Yes (please give details)] vs code
- Power Supply:         [USB]
## Problem Description

//Detailed problem description goes here.

Hello, 
I am implementing the OOB authentication, it has been mention in the docs that to disable the no OOB behavior we must set the value of the static OOB to some thing. so to do this  I follow the example given https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/main/ble_mesh_demo_main.c
and change the structure of the provision as below 
const uint8_t static_oob_val[] = {0x01};
esp_ble_mesh_prov_t provision = {
    .static_val = static_oob_val,
    .static_val_len = sizeof(static_oob_val),
    .uuid = dev_uuid,
    .oob_info=ESP_BLE_MESH_PROV_OOB_STRING,
    .output_size = 4,
    .output_actions = ESP_BLE_MESH_DISPLAY_STRING,
};

to my surprise that still the app nrf mesh as well as the esp ble mesh provisioned the device with no oob support. What I am missing here? is there is any example which shows  how to provision the node with the OOB? 

### Expected Behavior

the expected behavior or the desired result is shown in the video by the nordic semiconductors 
https://www.youtube.com/watch?v=XthbU9NP0Yg the provisioning of  the device is done with  blink of the led. there is no option to select oob method during the provisioning in that video.

Alvin1Zhang, 2020-05-13:
Thanks for reporting, we will look into.

Campou, 2020-06-22:
Hi @ankit-thealchemist 

Sorry for the late reply.

In the example `onoff_server`, you can set the oob method like the following in the structure `esp_ble_mesh_prov_t`,
```
const uint8_t static_oob_val[] = {1};
esp_ble_mesh_prov_t provision = {
.uuid = dev_uuid,
.static_val = static_oob_val,
.static_val_len = sizeof(static_oob_val),
.output_size = 4,
.output_actions = ESP_BLE_MESH_DISPLAY_STRING,
.input_size = 4,
.input_actions = ESP_BLE_MESH_PUSH,
};
```
Then on the nRF Mesh side, you need to choose to use No OOB, Static OOb, Output OOb or Input OOB for authentication.
* For Static OOB, you need to input the same value which is used in the example;
* For Output OOB, you need to get the number or string from the `ESP_BLE_MESH_NODE_PROV_OUTPUT_NUMBER_EVT` or `ESP_BLE_MESH_NODE_PROV_OUTPUT_STRING_EVT`, and input it on the app side;
* And for Input OOB, after the `ESP_BLE_MESH_NODE_PROV_INPUT_EVT` comes, you need to invoke `esp_ble_mesh_node_input_number` or `esp_ble_mesh_node_input_string` to input the value which is displayed on the app.

BTW, I use the nRF Mesh App (version 2.4.1) to test these authentication methods.

Thanks.

ankit-thealchemist, 2020-06-22:
> Hi @ankit-thealchemist
>
> Sorry for the late reply.
>
> In the example `onoff_server`, you can set the oob method like the following in the structure `esp_ble_mesh_prov_t`,
>
> ```
> const uint8_t [...]

yes you are right @Campou I have to choose the OOB method during provisioning. That's exactly my question, why? OR My precise question is how to disable the no OOB method of provisioning 
 in esp ble mesh device?

I perceive that that OOB authentication is for the security reason.
consider a scenario that I make the ble devices and they are waiting to be provisioned. They are guarded with the static OOB method for MITM attack.  Now suppose some one open the nrf ble mesh and provisioned the device with the no oob method. Isn't it a big security flaw. Why would any one uses the other  OOB method when NO oob method is present.

Campou, 2020-07-30:
Hi @ankit-thealchemist 

I think two solutions can be used.
* Using Public Key OOB, so when an attacker uses the nRF Mesh App to provision the device, the public key will be unreachable
* Add some specific checks in the stack, for example, when an unprovisioned device supports Static OOB, and if it receives a `Provisioning Start PDU` with No OOB choosn, the PDU will be ignored by the device.

Thanks.

ankit-thealchemist, 2020-07-31:
@Campou will try the above mention method and will let you know.

closed 2020-07-31 by ankit-thealchemist

Campou, 2020-08-10:
Hi @ankit-thealchemist 

Have provided a menuconfig option to use the second solution for checking if No OOB could be used on the node side, and this change will be synced to Github later.

Thanks.

ankit-thealchemist, 2020-08-10:
Hello @Campou, 

Thats an excellent news! keenly waiting for the update to be live!
thanks for the support.

ankit-thealchemist, 2020-09-23:
> Hi @ankit-thealchemist
> 
> Have provided a menuconfig option to use the second solution for checking if No OOB could be used on the node side, and this change will be synced to Github later.
> 
> Thanks.

do the given functionality in  menuconfig become live on git hub?

Campou, 2020-09-24:
Hi @ankit-thealchemist 

Currently not, please have a try with this patch [0001-ble_mesh-stack-Not-using-No-OOB-if-others-are-suppor.patch.zip](https://github.com/espressif/esp-idf/files/5273001/0001-ble_mesh-stack-Not-using-No-OOB-if-others-are-suppor.patch.zip), which contains the changes for this.

Thanks.

florian-ats, 2021-01-11:
Hi @Campou 

I confirm that this patch + the BLE_MESH_NOT_USE_NO_OOB parameter selection, produce the expected effect; prevent the provision of a device configured with a static key, when there is "No OOB" selected on the provisioner side.

[image]

[image]

Would it be possible to integrate the patch into the ESP-IDF repository?

hipal31, 2022-07-15:
Hello @ankit-thealchemist 

I am facing similar issue right now. I want to disable No OOB because I am using Static OOB. Can you please tell me suggest me solution apart from patch?

ankit-thealchemist, 2022-07-15:
Hello @hipal31 
We can't do nothing... Patch is the only option for now..

ingeniero-david, 2025-04-14:
Hi @ankit-thealchemist 
Any plans to include the patch into the ESP-IDF repository?