espressif/esp-idf #5284: espcoredump.py & xtensa-esp32-elf-gdb not working in docker

vtunr, 2020-05-13:
## Environment

- Environment : WSL2 (ubuntu18.04)
- Docker image : release/v4.0 & latest

## Problem Description

My goal is to use the docker container to process coredump. 
Running `espcoredump.py` lead to 
```
================== CURRENT THREAD REGISTERS ===================
Traceback (most recent call last):
  File "/opt/esp/idf/components/espcoredump/espcoredump.py", line 1190, in <module>
    main()
  File "/opt/esp/idf/components/espcoredump/espcoredump.py", line 1185, in main
    operation_func(args)
  File "/opt/esp/idf/components/espcoredump/espcoredump.py", line 1065, in info_corefile
    p = gdbmi_getinfo(p, handlers, "info registers")
  File "/opt/esp/idf/components/espcoredump/espcoredump.py", line 979, in gdbmi_getinfo
    p.stdin.write(bytearray("-interpreter-exec console \"%s\"\n" % gdb_cmd, encoding='utf-8'))
BrokenPipeError: [Errno 32] Broken pipe
```
Trying to run xtensa-esp32-elf-gdb by itself is not working inside the container : 
```
root@40e761ae8a4a:/project# xtensa-esp32-elf-gdb
xtensa-esp32-elf-gdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory
```
which is probably the cause of the `espcoredump.py` error.

### Steps to reproduce

Run : 
```
> docker run --rm -v $PWD:/project -w /project -t espressif/idf:latest xtensa-esp32-elf-gdb
> docker run --rm -v $PWD:/project -w /project -t espressif/idf:release-v4.0 xtensa-esp32-elf-gdb
> docker run --rm -v $PWD:/project -w /project -t espressif/idf:release-v4.0 python /opt/esp/idf/components/espcoredump/espcoredump.py info_corefile  /project/device.elf -c /project/coredump.bin -t raw
```

Alvin1Zhang, 2020-05-14:
@vtunr Thanks for reporting, we will look into.

PaulMichell, 2020-05-22:
I have just hit the same error from xtensa-esp32-elf-gdb running on a clean Kubunutu 20.04 with stable esp-idf.  It appears to be trying to load python 2.7. This system has 3.8.2 installed by default.  Really don't want to have to install 2.7 to be able to run GDB?

igrr, 2020-05-22:
@vtunr @PaulMichell We will switch to building GDB with Python 3 libraries in the next toolchain release. For now there are two options:

* Build the toolchain (including GDB) from source, either disabling Python scripting support, or configuring it to use Python 3: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup-scratch.html
* Install Python libraries: `apt-get install libpython2.7-dev` (do `apt-get update` first, if running inside IDF Docker)

Sorry for the trouble.

KaeLL, 2020-09-02:
> We will switch to building GDB with Python 3 libraries in the next toolchain release.

ETA?

closed 2020-09-16 by espressif-bot

igrr, 2020-09-16:
> We will switch to building GDB with Python 3 libraries in the next toolchain release.

This turned out more complex than expected, as libraries for different Python 3.x versions are not binary compatible. So if we build GDB, say, with Python 3.7 libraries, it won't work for users who have 3.8 installed.
It is possible to link Python statically, but that increases the toolchain distributable size.

For now, we have merged #5817 to add libpython2.7 to the docker image. This is also in the process of being backported to other release branches.

KaeLL, 2021-01-05:
> It is possible to link Python statically, but that increases the toolchain distributable size.

Bigger toolchain is less worse than a broken one, no?