espressif/esp-idf #5285: Odd toolchain bug(?)

lucastcox, 2020-05-13:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version: v4.2-dev-318-g605da33c3
- Build System:         [CMake]
- Compiler version: (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: VsCode
- Power Supply: on-board 3v3 regulator

## Problem Description

I have a "networkTask" class that manages ethernet, wifi, websockets etc. There is a member variable 'mNetworkSubscriber.' It's of type "subscriberHandle_t", which is a typedef'd QueueHandle_t. I was having some crashes when attempting to block on a queue using this QueueHandle, so I created a local queue in NetworkTask, and blocked on it. Oddly enough, that worked fine. 

While trying to find out what was going on, I ended up writing the following lines of code: 

(Note: mNetworkSubscriber is a queueHandle that is created by xQueueCreate in the NetworkTask constructor. 
networkSubscriber is a queueHandle that is created by xQueueCreate in the context of the networkTask.)

```C
    ESP_LOGE("", "member: %p, non-member %p", mNetworkSubscriber, networkSubscriber);
    mNetworkSubscriber = networkSubscriber;
    ESP_LOGE("", "member: %p, non-member %p", mNetworkSubscriber, networkSubscriber);
```

I got the following printout: 

```bash
E (568) "" member: 0x20004136, non-member 0x3ffc28b0
E (568) "" member: 0x20004136, non-member 0x3ffc28b0
```

### Expected Behavior

member equals non-member after assignment.

### Actual Behavior

member variable retains its previous value after assignment

Quite strange behavior, which tells me that something is getting corrupted, or there is a bug in the toolchain preventing the value of mNetworkSubscriber from being modified. Let me know if you need anything else (project, elf files, etc). In order to provide that, we'll need to sign a mutual NDA, but I've used espressif's mutual NDA before at a different company, so that will not be an issue.

igrr, 2020-05-14:
Hi @lucastcox, could you please try logging the address of mNetworkSubscriber and also the value of `this` pointer? I.e.
```c++
ESP_LOGE("", "member: %p, non-member %p &member %p this %p", mNetworkSubscriber, networkSubscriber, &mNetworkSubscriber, this);
```
and post the result? Thanks.

lucastcox, 2020-05-14:
@igrr Sure. Here you go: 

```c
E (1439) : member: 0x16004136, non-member 0x3ffc7028
E (1449) : member: 0x16004136, non-member 0x3ffc7028
E (1449) : member: 0x16004136, non-member 0x3ffc7028 &member 0x400da600 this 0x400da600
```

lucastcox, 2020-05-14:
And here is the backtrace when I attempt to block on the 'mNetworkSubscriber' queuehandle. 

```shell
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4008dc25  PS      : 0x00060933  A0      : 0x8008e977  A1      : 0x3ffc3830  
0x4008dc25: compare_and_set_native at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/soc/include/soc/compare_set.h:28
 (inlined by) spinlock_acquire at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/soc/include/soc/spinlock.h:91
 (inlined by) vPortCPUAcquireMutex at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:202
 (inlined by) vPortEnterCritical at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/freertos/xtensa/port.c:394

A2      : 0x16004182  A3      : 0x0000cdcd  A4      : 0x00060923  A5      : 0x00060920  
A6      : 0x0000abab  A7      : 0xb33fffff  A8      : 0x0000cdcd  A9      : 0x3ffc3830  
A10     : 0x00000003  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x00000001  
A14     : 0x00000004  A15     : 0x3ffb48fc  SAR     : 0x00000005  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x16004182  LBEG    : 0x40087da4  LEND    : 0x40087dd2  LCOUNT  : 0x00000000  
0x40087da4: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:168

0x40087dd2: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 6a2a6165d1553a0110ee19b5cb4788dbc2fb476bb8af075d70f266a044d2272d

Backtrace: 0x4008dc22:0x3ffc3830 0x4008e974:0x3ffc3860 0x400d9606:0x3ffc38a0 0x400da59d:0x3ffc38c0 0x400da609:0x3ffc3910 0x4008daad:0x3ffc3930
0x4008dc22: compare_and_set_native at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/soc/include/soc/compare_set.h:28
 (inlined by) spinlock_acquire at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/soc/include/soc/spinlock.h:91
 (inlined by) vPortCPUAcquireMutex at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/freertos/xtensa/include/freertos/portmacro.h:202
 (inlined by) vPortEnterCritical at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/freertos/xtensa/port.c:394

0x4008e974: xQueueGenericReceive at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/freertos/queue.c:1461 (discriminator 4)

0x400d9606: Itc::Wait(void*, void*, unsigned int) at /home/lucas/Projects/RemoteIO/build/../main/Source/Itc.cpp:69

0x400da59d: NetworkTask::MainTask() at /home/lucas/Projects/RemoteIO/build/../main/Tasks/NetworkTask.cpp:55

0x400da609: taskBouncer(void*) at /home/lucas/Projects/RemoteIO/build/../main/Tasks/NetworkTask.cpp:13

0x4008daad: vPortTaskWrapper at /home/lucas/Projects/RemoteIO/build/../esp-idf/components/freertos/xtensa/port.c:143
```

igrr, 2020-05-14:
> `this 0x400da600`

*This* looks very odd, because 0x400dxxxx is an address in IROM (program code in Flash). Normally a `this` pointer should be allocated on the heap, so the address would be somewhere in the 0x3ffxxxxx range.

Please check how Itc::Wait is called — make sure that the valid `this` pointer is used.

If you enable the GDB stub in menuconfig, you can then run `bt` (backtrace) command in GDB, and it will show you the backtrace along with arguments passed at each level of the call stack (as long as they are not optimized out).

lucastcox, 2020-05-14:
Thank you so much for the help. As soon as you mentioned this "this" pointer being invalid, it immediately jogged my memory about when I've encountered this a few years ago. Found the issue. In order to manage the freertos task in C++, I pass "this" into xTaskCreate, and then have the C function cast (void *params) to my task class and call the task member function. 

However, (must have been an accidental accepting of an intellisense suggestion in VSCode) I accidentally passed a random function into xTaskCreate instead of _this_. Thanks again, and that rule-of-thumb persists: _It's never the toolchain's fault_

closed 2020-05-14 by lucastcox