espressif/esp-idf #5287: mqtt can't be established over ppp when WIFI enabled : esp-tls: mbedtls_ssl_handshake returned -0x4310

chaabanemalki, 2020-05-13:
## Environment
- Development Kit:   custom board
- Module or chip used:  chip
- IDF version : v3.3.1-211-g6f9a7264c
- Build System:   Make 
- Compiler version : 1.22.0-80-g6c4433a5
- Operating System:  Windows
- (Windows only) environment type: MSYS2 mingw32

## Problem Description

Hello,

I establish ppp connection using a modem connected to my esp32 and then connect to my server using mqtt (mqtt_cfg.transport = MQTT_TRANSPORT_OVER_SSL). I'm able to subscribe and send data with no any issues.
But when I enable the WIFI in AP mode, I don't seem to be able to connect to the server via mqtt ?

I get the following errors

E (27372) esp-tls: mbedtls_ssl_handshake returned **-0x4310**
I (27372) esp-tls: Certificate verified.
V (27377) UPLOAD: CELLIB: PPP OUTPUT: 45 bytes

E (27378) esp-tls: Failed to open new connection
E (27378) TRANS_SSL: Failed to open a new connection
E (27383) MQTT_CLIENT: Error transport connect
D (27392) MQTT_CLIENT: Reconnect after 10000 ms

or

E (1192746) esp-tls: mbedtls_ssl_handshake returned **-0x10**
I (1192747) esp-tls: Certificate verified.
E (1192753) esp-tls: Failed to open new connection
E (1192754) TRANS_SSL: Failed to open a new connection
E (1192758) MQTT_CLIENT: Error transport connect
D (1192767) MQTT_CLIENT: Reconnect after 10000 ms

I'm not able to find -0x4310 and -0x10 meaning.  Where do I find them ?
Also, Does the WIFI and my ppp uses the same tcp stack ? if yes, it makes sense why I'm running on this issue.

My wifi code : 

`    esp_event_loop_init(event_handler, NULL);
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    wifi_config_t wifi_config = {
        .ap = {
            .max_connection = WIFI_AP_MAX_CONN_DEVICES,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK},
    };
    strcpy((char *)wifi_config.ap.ssid, _ssid);
    strcpy((char *)wifi_config.ap.password, _passwd);
    wifi_config.ap.ssid_len = strlen(_ssid);

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_IF_WIFI_AP, WIFI_PROTOCOL_11B));`

Alvin1Zhang, 2020-05-14:
@chaabanemalki Thanks for reporting, we will look into.

david-cermak, 2020-05-18:
Hi @chaabanemalki 

mbedtls has the notion of high level codes and other error strings so should look separately for 0x4300 and 0x10 (or use `mbedtls_strerror()` to decode), which indicate allocation issues.
```
RSA - The private key operation failed : BIGNUM - Memory allocation failed
```
 Does your application have enough free heap available after the AP has been enabled?

Note that both PPP and WiFi use the same TCP/IP stack, but as a separate interface and are supposed to work together just fine.

Alvin1Zhang, 2020-08-18:
@chaabanemalki Thanks for reporting, would you please help share if any updates for the issue? Thanks.

chaabanemalki, 2020-10-05:
Sorry for the late response. This feature no longer needed for us now.

I'll give it a try when I move to esp-ide v4.1

closed 2020-10-05 by chaabanemalki

AllanOricil, 2024-12-29:
Thank you @david-cermak