espressif/esp-idf #5288: export.sh doesn't work on macOS Catalina 10.15.x

tmspzz, 2020-05-13:
## Environment

- Development Kit: n/a
- Kit version (for WroverKit/PicoKit/DevKitC):  n/a
- Module or chip used: n/a
- IDF version: v4.1-beta1-182-g9778b163b
- Build System:  CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:  macOS
- Using an IDE?:  No
- Power Supply: n/a

## Problem Description

export.sh doesn't work on macOS Catalina

### Expected Behavior

export.sh works on macOS Catalina

### Actual Behavior

export.sh assumes bash to be the default shell. This assumption doesn't hold on macOS Catalina where the default shell is zsh.

### Steps to reproduce

- set zsh as default shell
- run export.sh

## Debug Logs
export.sh stops with 
```
IDF_PATH must be set before sourcing this script"
```

## Solution

Modify export.sh with the following (I can submit a PR)

```diff
diff --git a/export.sh b/export.sh
index 0408a76bb..6b692e273 100644
--- a/export.sh
+++ b/export.sh
@@ -15,7 +15,7 @@ function realpath_int() {
 function idf_export_main() {
     # The file doesn't have executable permissions, so this shouldn't really happen.
     # Doing this in case someone tries to chmod +x it and execute...
-    if [[ -n "${BASH_SOURCE}" && ( "${BASH_SOURCE[0]}" == "${0}" ) ]]; then
+    if [[ -n "${SHELL_SOURCE}" && ( "${SHELL_SOURCE[0]}" == "${0}" ) ]]; then
         echo "This script should be sourced, not executed:"
         echo ". ${BASH_SOURCE[0]}"
         return 1
@@ -24,12 +24,12 @@ function idf_export_main() {
     if [[ -z "${IDF_PATH}" ]]
     then
         # If using bash, try to guess IDF_PATH from script location
-        if [[ -n "${BASH_SOURCE}" ]]
+        if [[ -n "${SHELL_SOURCE}" ]]
         then
             if [[ "$OSTYPE" == "darwin"* ]]; then
-                script_dir="$(realpath_int $BASH_SOURCE)"
+                script_dir="$(realpath_int $SHELL_SOURCE)"
             else
-                script_name="$(readlink -f $BASH_SOURCE)"
+                script_name="$(readlink -f $SHELL_SOURCE)"
                 script_dir="$(dirname $script_name)"
             fi
             export IDF_PATH="${script_dir}"
@@ -95,7 +95,16 @@ function idf_export_main() {
     echo ""
 }

+if [ -n "$ZSH_VERSION" ]; then
+   SHELL_SOURCE=${(%):-%x}
+elif [ -n "$BASH_VERSION" ]; then
+   SHELL_SOURCE=BASH_SOURCE
+else
+   echo "This script only works on bash or zsh."
+fi
+
 idf_export_main

+
 unset realpath_int
 unset idf_export_main
```

igrr, 2020-05-14:
Hi @tmspzz, IDF_PATH detection for zhs was implemented in https://github.com/espressif/esp-idf/commit/1c4c584e2845a34eb6d2e52e81fd5d3b9d0a73f3. Will backport this to release/v4.1

tmspzz, 2020-05-14:
@igrr that's nice! Thanks! 

Also, I'm pretty sure you're aware of it, but macOS loudly complains about using `python` instead of `python3`. And of course `/usr/bin/env python` defaults to python2

igrr, 2020-05-14:
No, I'm not aware of that. What error are you getting with regards to `python`?

tmspzz, 2020-05-14:
I think it's the same as https://github.com/espressif/esp-idf/issues/4774

But since the scripts start with `#!/usr/bin/env python`  there is no (good?) way to switch to python3

igrr, 2020-05-14:
How about `brew install python3` (or any other method of your choice), and then make sure /usr/local/bin/python symlink points to /usr/local/bin/python3?

gojimmypi, 2020-05-23:
@tmspzz if indeed the same as #4774, [this solution](https://github.com/espressif/esp-idf/issues/4774#issuecomment-633079990) worked for me. 

I'm not sure how well that might work on a Mac.

igrr, 2020-09-22:
Has been fixed on release/v4.1 with https://github.com/espressif/esp-idf/commit/6ae54e7fd29fd28698dfd5c90d43033185f0b199.

closed 2020-09-22 by igrr

malachib, 2021-02-21:
Any chance this can backport into a v3.3.5? 🙏

igrr, 2021-02-21:
Hi @malachib, 3.3 releases did not use install.sh/export.sh yet, please refer to the getting started guides for that version: https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started/index.html#setup-toolchain.

malachib, 2021-02-21:
Really?  I see the `install.sh` right there in the v3.3.4 tag checkout -- but not an issue, I've already got the toolchain set up as described in that link so good to go.  Thank you for the response