espressif/esp-idf #5289: nimBLE functionalities and upcoming improvements

AlpinProgrammer, 2020-05-14:
Hello !

Always developing BLE with bluedroid, I came across nimBLE which offers important improvements.

However, performing some tests on a merged central / peripheral test, I was not able to make both central and peripheral active concurrently.
Do you confirm my findings that peripheral/central coexistence is not supported ?

Besides, I see that further improvements are continuously made on the bluedroid stack and the related examples, mesh as well.
Do you intend on maintaining, developing and improving nimBLE in the same manner ?

Thanks in advance !
Best regards.

dhrishi, 2020-05-14:
@AlpinProgrammer  To answer your question, NimBLE supports concurrent peripheral and central role. Please share more details (as per the IDF issue template) on the issue you are facing.

Adding such examples, which will be helpful for the users, is already in the plan and we are working on it.

And yes, we have been fixing issues, adding features and maintaining our fork of NimBLE host stack . NimBLE being a submodule in ESP-IDF, you won't see much activity here. See this: https://github.com/espressif/esp-nimble. Another thing is, we have not changed the NimBLE API and hence you won't see much code in ESP-IDF except for the porting layer and a host-controller init/deinit API.

If there are specific improvements/features you are looking for, please let us know. We would be happy to help.

dhrishi, 2020-05-20:
@AlpinProgrammer Can you please close this if your questions are answered? Feel free to raise a new issue for a specific problem and provide all the details.

AlpinProgrammer, 2020-05-20:
Thank you for your fast answer as always.

One last thing is I see in the sdkconfig that it is possible to allow several BLE connections.
However, there is no example about gattc_multiconnect. Is it currently compatible with multiconnect anyways ?

Regards.

dhrishi, 2020-05-20:
@AlpinProgrammer Yes, multiple connections are possible as well. Please set `CONFIG_BT_NIMBLE_MAX_CONNECTIONS` to the number of simultaneous connections you want. The default value is 3 (same as the default for controller connections) and maximum is 9 (as supported by the controller)

We will try to add an example for multi connections as well. Thanks for your inputs!

Alvin1Zhang, 2020-05-22:
@AlpinProgrammer Thanks for reporting and updates, feel free to reopen. Thanks.

closed 2020-05-22 by Alvin1Zhang

AlpinProgrammer, 2020-05-22:
Thank you for your help !