espressif/esp-idf #5291: Faulty tcp_pbc structure management / ENOBUF errors

sharkx, 2020-05-14:
## Environment

- Development Kit:      [none]
- Kit version [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version v4.2-dev-560-g5f771fb7b
- Build System:         [idf.py]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [eclipse]
- Power Supply:         [USB]

## Problem Description

The ESP32 module runs in APSTA mode and acts like a bridge for passing small packets (max 100 bytes, usually around 25 bytes) received from any connected to all others. There is a listener that receives the packets and then queues them; the queue is processed periodically and packets are sent to all other known IP addresses. All is done through TCP connections.

There is a server socket (classic bind-listen-accept sequence) that runs perfectly.

The issue is for the send. the problem is that when the sockets are closed, the associated `tcp_pcb` struct is not freed, but rather goes into "timewait" state, and is cleaned afterwards. 

After a certain number of packets are sent (wans't able to determine precisely the context, but generally after the max number of `tcp_pcb` slots is reached), the call to `tcp_alloc` tries to call `tcp_kill_timewait` and 'kills' the oldest `tcp_pcb` structure.

Which seems fine, but when a large number of packets is sent, the test-kill-alloc sequences for two tasks may overlap and produce incorrect behavior.

Here's a possible timeline that works
```
task 1 : test ---- kill ---- alloc
task 2 : ---------------------------test ---- kill ---- alloc
```
And another that doesn't; the task2 finds the free slot from the first task, takes it and does not free its own, making the first task to fail.
```
task 1 : test ---- kill ----------------------- alloc XXX FAILS !
task 2 : --------------------test ---- alloc
```
I'm not entirely sure that tis is 100% accurate. It's what I've beed able to determine from logs, but right now it looks like the only logical explanation.

What I'm sure of is that i get a call to `netconn_alloc` and one for `netconn_free` for sockets created by `accept()` but not for those created with `socket()` and I start getting ENOBUF errors. Also, it seems that for sockets created by `accept()`, the `netconn_free` call does not come from `closesocket()`.

The code for the transmit sequence is also pretty basic and classic (only relevant parts are shown).
```
void queueTransmitTCPFrame(void* _pElement) {
	struct sockaddr_in dest_addr;
	// ...
	int sock = socket(addr_family, SOCK_STREAM, ip_protocol);
	if (sock < 0) 	{
		// error handling
	}
	else 	{
		// setsockopt(g_server_sock, SOL_SOCKET, TCP_NODELAY, &_flag, sizeof(_flag));
		// setsockopt(g_server_sock, SOL_SOCKET, SO_CONTIMEO, &tv, sizeof(tv));

		err = connect(sock, (struct sockaddr*) &dest_addr, sizeof(dest_addr));
		if (err != 0) {
			// error handling
		}
		else {
			int sent = send(sock, ...);
			if (sent != packetSize) {
				// error handling
			}
		}

		shutdown(sock, 2);
		closesocket(sock);
	}
}
```

### Expected Behavior

The tcp_pcb structure to be freed on `closesocket` or any other solution that does not put the `tcp_pcb` structure in wait state and 'killed' afterwards.

### Actual Behavior

As described above

### How to reproduce

I don't have enough modules to create a test scenario besides the ones that I use for development, but if there are any things to try, I will gladly try and send the results back,

Alvin1Zhang, 2020-05-15:
@sharkx Thanks for the detailed bug report and letting us know about this. We will look into. @liuzfesp

sharkx, 2020-05-15:
Here is some more info. I have printf'ed calls to netconn_alloc, netconn_free, tcp_alloc, tcp_free as well as in my code with the last line per group being the result of a call to `tcp_pcb_num_cal`

Here's the first log; I tried to arrange the sequences in columns on a per-task basis
```
netconn_alloc(0x3ffbb088)
tcp_alloc(0x3ffceb2c)
[001] outgoing socket=49

							tcp_alloc(0x3ffc9688)
							netconn_alloc(0x3ffbb5dc)
							[002] incoming socket=50
							[002] shutdown(50, 2)
							[002] closesocket(50)
							 netconn_free(0x3ffbb5dc)
							[002] incoming socket 50 closed
							[002] time_wait:0, total:3

							tcp_alloc(0x3ffc979c)
							netconn_alloc(0x3ffcea3c)
							[003] incoming socket=50
							[003] shutdown(50, 2)
							[003] closesocket(50)
							 netconn_free(0x3ffcea3c)
							[003] incoming socket 50 closed
							[003] time_wait:1, total:4

							tcp_alloc(0x3ffceca4)
							netconn_alloc(0x3ffbb5c4)
							[004] incoming socket=50
							[004] shutdown(50, 2)
							[004] closesocket(50)
							 netconn_free(0x3ffbb5c4)
							[004] incoming socket 50 closed
							[004] time_wait:2, total:5

							tcp_alloc(0x3ffcf1e0)
							netconn_alloc(0x3ffbb5c4)
							[005] incoming socket=50
							[005] shutdown(50, 2)
							[005] closesocket(50)
							 netconn_free(0x3ffbb5c4)
							[005] incoming socket 50 closed
							[005] time_wait:3, total:6
							 tcp_free(0x3ffcf1e0)

							tcp_alloc(0x3ffcee5c)
							netconn_alloc(0x3ffbb5c4)
							[006] incoming socket=50
							[006] shutdown(50, 2)
							[006] closesocket(50)
							 netconn_free(0x3ffbb5c4)
							[006] incoming socket 50 closed
							 tcp_free(0x3ffcee5c)
							[006] time_wait:3, total:5

							tcp_alloc(0x3ffcee5c)
							netconn_alloc(0x3ffbb5c4)
							[007] incoming socket=50
							[007] shutdown(50, 2)
							[007] closesocket(50)
							 netconn_free(0x3ffbb5c4)
							 tcp_free(0x3ffcee5c)
							[007] incoming socket 50 closed
							[007] time_wait:3, total:5

[001] shutdown(49, 2)
[001] closesocket(49)
 netconn_free(0x3ffbb088)
[001] Outgoing socket 49 closed
[001] time_wait:3, total:5
```
For some reason, some sequences have a `tcp_free` call, some don't; when there is no such call, the time_wait number increases to a point when it reaches the max number allowed.

sharkx, 2020-05-15:
And another one, where there are more overlapping tasks; a bit harder to follow, but the situation is the same ... some sequences don't have a `tcp_free` call and the `time_wait` number of blocks increases.

I'm attaching it as file; use a <tab> size of 4 to get the proper indent.
[log.txt](https://github.com/espressif/esp-idf/files/4632232/log.txt)

freakyxue, 2020-05-15:
hi @sharkx 
In the tcp_alloc(u8_t prio) function,
It can be modified in this way to solve this problem

tcp_alloc(u8_t prio)
{
  struct tcp_pcb *pcb;
#if ESP_LWIP 
  tcp_pcb_num_t tcp_pcb_num;
#endif
  LWIP_ASSERT_CORE_LOCKED();
#if ESP_LWIP
    
  tcp_pcb_num_cal(&tcp_pcb_num);

  while (tcp_pcb_num.total >= MEMP_NUM_TCP_PCB) { //  Where to modify
    if (tcp_pcb_num.time_wait > 0){
      tcp_kill_timewait();
    } else if (tcp_pcb_num.last_ack > 0){
      tcp_kill_state(LAST_ACK);
    } else if (tcp_pcb_num.closing > 0){
      tcp_kill_state(CLOSING);
    } else if (tcp_pcb_num.fin_wait2 > 0){
      tcp_kill_state(FIN_WAIT_2);
    } else if (tcp_pcb_num.fin_wait1 > 0){
      tcp_kill_state(FIN_WAIT_1);
    } else {
      tcp_kill_prio(prio);
    }
  }

sharkx, 2020-05-15:
Hello,

Thanks for the reply, but I'm not sure it solves the problem.

Changing the `if` for a `while` does not do much good... quite the opposite, I think.

When `tcp_pcb_num.total < MEMP_NUM_TCP_PCB` it will not execute, just like the `if`.

When `tcp_pcb_num.total >= MEMP_NUM_TCP_PCB` it will execute in an endless loop, because you don't change the `tcp_pcb_num.total` after killing the oldest block. Even if you would, it will only kill until you get `MEMP_NUM_TCP_PCB - 1` items left, which does not fix the sync issue, because two tasks compete for the same last available block.

Can you please explain in which situations a block ens up in the `time_wait` state ?

Best regards,

freakyxue, 2020-05-18:
[image]

freakyxue, 2020-05-18:
while (tcp_pcb_num.total >= MEMP_NUM_TCP_PCB)
    {
        if (tcp_pcb_num.time_wait > 0){
            tcp_kill_timewait();
        } else if (tcp_pcb_num.last_ack > 0){
            tcp_kill_state(LAST_ACK);
        } else if (tcp_pcb_num.closing > 0){
            tcp_kill_state(CLOSING);
        } else if (tcp_pcb_num.fin_wait2 > 0){
            tcp_kill_state(FIN_WAIT_2);
        } else if (tcp_pcb_num.fin_wait1 > 0){
            tcp_kill_state(FIN_WAIT_1);
        } else {
            tcp_kill_prio(prio);
        }
       tcp_pcb_num_cal(&tcp_pcb_num);  
    }

freakyxue, 2020-05-18:
hi @sharkx 
I'm sorry to add one line less code to the scheme submitted last time.
When TCP is disconnected, the control block will have `time wait` status

sharkx, 2020-05-18:
Hello,

Thanks for the update; I have tried, more or less, the same approach.

So, if TCP is disconnected, there's no problem in 'killing' all connections
in this state, right ?

Regards,
Stefan.

freakyxue, 2020-05-18:
yes,In 'time wait' state, there is no problem in deleting the TCP control block

sharkx, 2020-05-18:
Thanks.

To remove all time_wait blocks, iI have used this (not the most efficient
code, but had less changes from original; one pass through the list should
have done it) :

static uint8_t tcp_kill_timewait_inner(void)
 {
   struct tcp_pcb *pcb, *inactive;
   u32_t inactivity;

   inactivity = 0;
   inactive = NULL;

   /* Go through the list of TIME_WAIT pcbs and get the oldest pcb. */
   for (pcb = tcp_tw_pcbs; pcb != NULL; pcb = pcb->next) {
     if ((u32_t)(tcp_ticks - pcb->tmr) >= inactivity) {
       inactivity = tcp_ticks - pcb->tmr;
       inactive = pcb;
     }
   }
   if (inactive != NULL) {
    LWIP_DEBUGF(TCP_DEBUG, ("tcp_kill_timewait: killing oldest TIME-WAIT
PCB %p (%"S32_F")\n",
                            (void *)inactive, inactivity));
     tcp_abort(inactive);
     return true;
   }
  return false;
}

static void tcp_kill_timewait(void)
{
    while(tcp_kill_timewait_inner()) {
    }
}

freakyxue, 2020-05-19:
@sharkx 
If there are still problems in the test of this scheme, please continue to put forward

sharkx, 2020-05-19:
Hello,

Thanks. So far it seems ok, at least for the connection block management.
But now, since the connections work, I get a lot of ENOTCONN, ECONNRESET or
ECONNABORTED ...

Best regards,

sharkx, 2020-05-21:
Hello,

After more testing, I'm not entirely sure the call to tcp_alloc is
thread-safe (or task-safe); I have added some sync code (a simple
semaphoe created by xSemaphoreCreateBinary) to the function, and it seems
to behave better; I'm still investigating this.

I'm not sure if this is the place for this question, but I tried on the
forums without success, and it's related to my problem.

I think one of the main causes for the socket problems I have is this : I
use the ESPs as APs in a mesh-like network, but not using the ESP APIs; I
have developed my own, before the time of ESP-IDF.

For instance there is an AP node that has 10 stations connected to it; when
some node tries to send a broadcast packet to the network, that AP node
will try to relay the packet to all 10 stations.

Suppose one of the nodes has a problem and the call to connect takes too
long.
Here comes the dilemma...

If send sequentially, packets that come after the blocking one take too
long to send and the network get stalled.

If sent using tasks/threads, sockets get locked and exhausted, as they come
in limited supply.

I think it could really help to be able to trim down the socket connection
time, but I can't seem to find a way to do this.

I would really appreciate if you could point me to the right direction; as
a side note, I'm totally comfortable with altering locally lwip code and
test, but I'm not sure where to look.

Thanks in avance
Best regards.

freakyxue, 2020-05-22:
hi @ sharkx 
while (tcp_pcb_num.total >= MEMP_NUM_TCP_PCB) // Where to modify
{
    if (tcp_pcb_num.time_wait > 0){
        tcp_kill_timewait();
    } else if (tcp_pcb_num.last_ack > 0){
        tcp_kill_state(LAST_ACK);
    } else if (tcp_pcb_num.closing > 0){
        tcp_kill_state(CLOSING);
    } else if (tcp_pcb_num.fin_wait2 > 0){
        tcp_kill_state(FIN_WAIT_2);
    } else if (tcp_pcb_num.fin_wait1 > 0){
        tcp_kill_state(FIN_WAIT_1);
    } else {
        tcp_kill_prio(prio);
    }
   tcp_pcb_num_cal(&tcp_pcb_num);  // Where to modify
}
Have you ever tested this solution? Because someone had the same problem as you before and solved it with this solution.

sharkx, 2020-05-22:
Hello,

Not exactly like this, because it does not solve the problem of concurrent
access when two or more threads compete for the last TCP_PCB.

But, since you can kill ALL time_wait blocks, my version
of tcp_kill_timewait deletes *all of them*, not just the oldest. And it
seems to do the job.

This together with a counting semaphore that does not allow threads to try
and send data if another N socket-connect-send-close sequences are in
progress.

Here's my test version for  tcp_kill_timewait

static uint8_t tcp_kill_timewait_inner(void)
 {
   struct tcp_pcb *pcb, *inactive;
   u32_t inactivity;

   inactivity = 0;
   inactive = NULL;

   /* Go through the list of TIME_WAIT pcbs and get the oldest pcb. */
   for (pcb = tcp_tw_pcbs; pcb != NULL; pcb = pcb->next) {
     if ((u32_t)(tcp_ticks - pcb->tmr) >= inactivity) {
       inactivity = tcp_ticks - pcb->tmr;
       inactive = pcb;
     }
   }
   if (inactive != NULL) {
    LWIP_DEBUGF(TCP_DEBUG, ("tcp_kill_timewait: killing oldest TIME-WAIT
PCB %p (%"S32_F")\n",
                            (void *)inactive, inactivity));
     tcp_abort(inactive);
     return true;
   }
  return false;
}

static void tcp_kill_timewait(void)
{
    while(tcp_kill_timewait_inner()) {
    }
}

Best regards,
Stefan.

freakyxue, 2020-05-22:
hi  @ sharkx 
Can you do a test with your code?
Open this macro:#define TCP_DEBUG            LWIP_DBG_ON /In  lwip/port/esp32/includ/lwipopts.h
Test with this code:
while (tcp_pcb_num.total >= MEMP_NUM_TCP_PCB) // Where to modify
{
if (tcp_pcb_num.time_wait > 0){
tcp_kill_timewait();
} else if (tcp_pcb_num.last_ack > 0){
tcp_kill_state(LAST_ACK);
} else if (tcp_pcb_num.closing > 0){
tcp_kill_state(CLOSING);
} else if (tcp_pcb_num.fin_wait2 > 0){
tcp_kill_state(FIN_WAIT_2);
} else if (tcp_pcb_num.fin_wait1 > 0){
tcp_kill_state(FIN_WAIT_1);
} else {
tcp_kill_prio(prio);
}
tcp_pcb_num_cal(&tcp_pcb_num); // Where to modify
}

I want to see if this log is printed:tcp_alloc: no available tcp pcb.
thanks!

Alvin1Zhang, 2020-08-18:
@sharkx Thanks for reporting, would you please help share if any update for this issue? Thanks.

closed 2020-11-02 by Alvin1Zhang