espressif/esp-idf #5292: Device won't join WPA2 ENTERPRISE AP if setting threshold.authmode = WIFI_AUTH_WPA2_ENTERPRISE

AxelLin, 2020-05-15:
## Environment

- Module or chip used: ESP32-WROOM-32
- IDF version: v4.2-dev-1541-g5e436be77f68
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: NO
- Power Supply: USB

## Problem Description

Seems someting wrong with threshold.authmode for WIFI_AUTH_WPA2_ENTERPRISE.
Test using examples/wifi/wpa2_enterprise.

If I add below change:

```
diff --git a/examples/wifi/wpa2_enterprise/main/wpa2_enterprise_main.c b/examples/wifi/wpa2_enterprise/main/wpa2_enterprise_main.c
index a3d453b6eb28..f18b8f1ac2e4 100644
--- a/examples/wifi/wpa2_enterprise/main/wpa2_enterprise_main.c
+++ b/examples/wifi/wpa2_enterprise/main/wpa2_enterprise_main.c
@@ -118,6 +118,7 @@ static void initialise_wifi(void)
     wifi_config_t wifi_config = {
         .sta = {
             .ssid = EXAMPLE_WIFI_SSID,
+            .threshold.authmode = WIFI_AUTH_WPA2_ENTERPRISE,
         },
     };
     ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
```

Then it seems does not join the target AP at all.
The log won't display "wpa2_task".

### Expected Behavior
Setting threshold.authmode = WIFI_AUTH_WPA2_ENTERPRISE should be able to join AP with
WPA2 enterprise.

### Actual Behavior
Seems wpa2_task is not started at all if setting threshold.authmode = WIFI_AUTH_WPA2_ENTERPRISE.

Alvin1Zhang, 2020-05-15:
@AxelLin Thanks for reporting and letting us know about this. We will look into.

AxelLin, 2020-05-21:
Hi @sagb2015 
BTW, since you are working on the authmode threshold for WIFI_AUTH_WPA2_ENTERPRISE,
had better also check if setting threshold.authmode=WPA3_PSK and authmode=WPA2_WPA3_PSK works or not if the target AP is WPA3_PSK or WPA2_WPA3_PSK.

Alvin1Zhang, 2020-06-08:
We had a fix which was merged recently, will share updates when it is available on GitHub. Thanks.

closed 2020-06-13 by AxelLin