espressif/esp-idf #5295: a2dp_source connect failed

chegewara, 2020-05-15:
Im not sure it is actually a bug or just limitation of stack.
I am trying to merge HFP_HF with A2DP_SOURCE. HFP_FP example works great so far. 
Now here is what i am trying to do:
- connect with smartphone, connection is initiated by smartphone to hfp_fp, it works
- after connecting, try to call a2dp_source_connect() and here is log i get:
```
D (44519) BT_APPL: BTA got event 0x1209

E (44519) BT_APPL: reset flags
I (44519) BT_A2D: A2D_FindService uuid: 110b
W (44539) A2DP 2: event => 0
I (44539) : params => 1, 0
D (45089) BT_APPL: BTA got event 0x10e

D (45089) BT_APPL: bta_dm_sm_execute event:0xe
I (45139) BT_A2D: a2d_sdp_cback status: 0
D (45139) BT_APPL: BTA got event 0x1215

W (45139) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2

W (45139) A2DP 2: event => 0
I (45149) : params => 0, 0
D (45149) BT_APPL: ## ON A2DP IDLE ## peer_sep = 0, service id = 3
D (52119) BT_APPL: BTA got event 0x10e

D (52129) BT_APPL: bta_dm_sm_execute event:0xe
```
i see it is connecting and right after then it disconnects. My problem is i dont know if this is even possible and i dont know events from this logs.

Thanks for any tips or suggestions

chegewara, 2020-05-15:
```
I (3058) BT_AVRC: AVRC_AddRecord uuid: 110c
I (3068) BT_AVCT: AVCT_CreateConn: 1, control:3
D (3068) BT_AVCT: avct_ccb_alloc 0
D (3068) BT_AVRC: AVRC_Open role: 1, control:3 status:0, handle:0
I (3068) BT_AVRC: AVRC_AddRecord uuid: 110e
```

Wth-Esp, 2020-05-19:
Hi @chegewara 

Please refer to [a2dp_hfp_example](https://github.com/espressif/esp-adf/tree/master/examples/get-started/pipeline_a2dp_sink_and_hfp) in ADF. 

Thanks
Weitianhua

chegewara, 2020-05-19:
Thanks. 
I could run hfp with a2dp sink, the thing is i dont want to run a2dp sink, just HFP + AVRC, but im guessing its not possible.

What i needed is to control music on smartphone + accept/reject calls without streaming data to esp32.

Wth-Esp, 2020-05-19:
Hi @chegewara 

As the concepts in BT, what you need is to use ESP32 as `Hands-Free sink` and `AVRC controller`.
The smartphone acts as `Hands-Free Audio Gateway ` and `AVRC Target`. 

The smartphone may use `A2DP source` and `AVRC Target` together, so I think it's not impossible for an `A2DP source` to connect with another `A2DP source`. That is the question.

Thanks
Weitianhua

chegewara, 2020-05-19:
I am new to bluetooth classic, so i thought its better to ask. 

Thanks

closed 2020-05-19 by chegewara

Wth-Esp, 2020-05-19:
Feel free to ask :)