espressif/esp-idf #5297: BLE-MESH messages between SIG models and Vendor models

FranciscoGilCC, 2020-05-16:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):
    // v4.2-dev-358-ga0644bf8a
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [Yes (Vscode with Espressif IDF extension)]
- Power Supply:         [USB]

## Problem Description

I can send messages with generic Opcodes from vendor models to SIG models, but I can't receive generic messages on vendor models even though I've also registered a generic Opcode for my server model.

`esp_ble_mesh_model_op_t test_onoff_srv_op[] = {
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_TEST_ONOFF_GET,       1),
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_TEST_ONOFF_SET,       1),
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_TEST_ONOFF_SET_UNACK, 1),
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET_UNACK,      1),
    ESP_BLE_MESH_MODEL_OP_END,
};`

### Expected Behavior

I expected messages with `ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET_UNACK` Opcode to trigger the callback function for custom server models.

### Actual Behavior

Message is completely ignored and callback function is never called.

### Steps to reproduce

1. Setup a BLE MESH node with a SIG GENERIC ONOFF client model and another node with a Vendor server model with the above Opcodes.
2. Trigger sending a message with `ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET_UNACK` Opcode
3. Verify nothing is received on node with vendor server model.

Campou, 2020-08-10:
Hi @Mantaba 

Sorry for the late reply.

Because for a vendor model, the opcode of its message must be 3 octets. If using the SIG-defined opcodes for vendor models, and when the message is received, the mesh stack will find the corresponding model based on the received opcode among SIG models, no proper model will be found, thus the message will be ignored.

Thanks.

Alvin1Zhang, 2020-08-18:
@Mantaba Thanks for reporting and sorry for the slow turnaround, would you please help share if any updates of this issue?Thanks.

closed 2025-03-14 by Alvin1Zhang