espressif/esp-idf #5298: CMake Error at ../../../../tools/cmake/component.cmake:468 (add_library)

w3llschmidt, 2020-05-16:
- Development Kit: ESP32-Wrover-Kit
- IDF version  v4.2-dev-1541-g5e436be77
- Build System: idf.py
- Compiler version:xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:  Linux
- Using an IDE?: No

## Problem Description

`ESP32/esp-idf/examples/wifi/getting_started/station`

$idf.py build

```
-- Configuring done
CMake Error at ../../../../tools/cmake/component.cmake:468 (add_library):
  Cannot find source file:

    /home/robotron/Dokumente/ESP32/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
Call Stack (most recent call first):
  ../../../../components/lwip/CMakeLists.txt:138 (idf_component_register)

```

w3llschmidt, 2020-05-16:
Also checked this:

```
robotron@robotron:~/Dokumente/ESP32/esp-idf/examples/wifi/getting_started/station$ sudo find /home/robotron/Dokumente -name ip4_*.c
/home/robotron/Dokumente/ESP32/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_addr.c
/home/robotron/Dokumente/ESP32/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_frag.c
/home/robotron/Dokumente/ESP8266/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4_addr.c
/home/robotron/Dokumente/ESP8266/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4_frag.c

```

Alvin1Zhang, 2020-05-18:
@w3llschmidt Thanks for reporting, and glad that you found a solution for this issue. Feel free to reopen if you still have this issue. Thanks.

closed 2020-05-18 by Alvin1Zhang

semihkaya0941, 2021-11-10:
I'm getting same error too

After saying idf.py build in terminal : 
`CMake Error at /../../../../tools/cmake/component.cmake:470 (add_library):
  Cannot find source file:

    /home /../../../../components/dht/main.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  components/dht/CMakeLists.txt:1 (idf_component_register)

CMake Error at  /../../../..//esp-idf/tools/cmake/component.cmake:470 (add_library):
  No SOURCES given to target: __idf_dht
Call Stack (most recent call first):
  components/dht/CMakeLists.txt:1 (idf_component_register)
`

My CMakeLists.txt : 

`cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS /home/humankeymyprojects/workspace/esp_projects/ttest_dht11/components)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(ttest_dht11)
`

betagtiti, 2022-04-02:
how do