espressif/esp-idf #5299: 硬件上电重启2次

shenyong-iot01, 2020-05-16:
您好，
     我这边做的CWRGB灯类产品，在进行快速开关电重启时会出现如下log，造成灯闪烁，闪烁PWM-IO为GPIO32，请帮忙分析下原因，需要资料请提出！
1.
[16:56:05.620] rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:56:05.620] configsip: 0, SPIWP:0xee
[16:56:05.620] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:56:05.620] mode:DIO, clock div:1
[16:56:05.620] load:0x3fff0018,len:4
[16:56:05.620] load:0xffffffff,子ΗCets Jun  8 2016 00:22:57
[16:56:06.551] 
[16:56:06.551] rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[16:56:06.551] configsip: 0, SPIWP:0xee
[16:56:06.551] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:56:06.551] mode:DIO, clock div:1
[16:56:06.551] load:0x3fff0018,len:4
[16:56:06.551] load:0x3fff001c,len:4900
[16:56:06.551] load:0x40078000,len:12424
[16:56:06.551] load:0x40080400,len:6916
[16:56:06.551] entry 0x40080730

2：
[17:10:02.548] rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
[17:10:02.548] configsip: 0, SPIWP:0xee
[17:10:02.548] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0xxets Jun  8 2016 00:22:57
[17:10:02.999] 
[17:10:02.999] rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
[17:10:02.999] configsip: 0, SPIWP:0xee
[17:10:02.999] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

shenyong-iot01, 2020-05-16:
ESP32 3.3.1SDK

costaud, 2020-05-18:
请问：
一共是用多少个GPIO控制LED？ 
是否只有IO32存在该现象 ？ 
PWM控制是使用的哪个硬件模块产生的 ？

shenyong-iot01, 2020-05-18:
@costaud
PWM配置如下：硬件连接当前只使用#define LIGHT_PWM_CHC_GPIO	  		(33)
#define LIGHT_PWM_CHW_GPIO	   		(32) 其他未连接
在出现两个两次rst   log时这两路都会闪烁。
只开启LIGHT_PWM_CHC_GPIO	 （33）情况下重启define LIGHT_PWM_CHW_GPIO	   		(32) 闪烁
只开启LIGHT_PWM_CHW_GPIO	 （32）情况下重启define LIGHT_PWM_CHC_GPIO
 (33) 闪烁

#define LIGHT_PWM_TIMER		   		LEDC_TIMER_0
#define LIGHT_PWM_TIMER_1		   	LEDC_TIMER_1

#define LIGHT_PWM_MODE		  		LEDC_HIGH_SPEED_MODE
#define LIGHT_PWM_CH_NUM			5

#define LIGHT_PWM_CHC_GPIO	  		(33)
#define LIGHT_PWM_CHC_CHANNEL	   	LEDC_CHANNEL_0
#define LIGHT_PWM_CHW_GPIO	   		(32)
#define LIGHT_PWM_CHW_CHANNEL		LEDC_CHANNEL_1
#define LIGHT_PWM_CHR_GPIO	   		(27)
#define LIGHT_PWM_CHR_CHANNEL	   	LEDC_CHANNEL_2
#define LIGHT_PWM_CHG_GPIO	   		(26)
#define LIGHT_PWM_CHG_CHANNEL	   	LEDC_CHANNEL_3
#define LIGHT_PWM_CHB_GPIO	   		(25)
#define LIGHT_PWM_CHB_CHANNEL	   	LEDC_CHANNEL_4

#define LIGHT_PWM_MAX_DUTY			1024

  static void light_drive_init(void)
{
	/*
     * Prepare and set configuration of timers
     * that will be used by LED Controller
     */
    ledc_timer_config_t ledc_timer = {
        .duty_resolution = LEDC_TIMER_10_BIT,   // resolution of PWM duty
        .freq_hz = 10000,                      // frequency of PWM signal
        .speed_mode = LIGHT_PWM_MODE,          // timer mode
        .timer_num = LIGHT_PWM_TIMER           // timer index
    };

	// Set configuration of timer0 for high speed channels
    ledc_timer_config(&ledc_timer);

	//用于相位偏移
	for(int i=0; i<100000; i++);		
	ledc_timer.timer_num = LIGHT_PWM_TIMER_1;
	ledc_timer_config(&ledc_timer);

	/*
     * Prepare individual configuration
     * for each channel of LED Controller
     * by selecting:
     * - controller's channel number
     * - output duty cycle, set initially to 0
     * - GPIO number where LED is connected to
     * - speed mode, either high or low
     * - timer servicing selected channel
     *   Note: if different channels use one timer,
     *         then frequency and bit_num of these channels
     *         will be the same
     */
    ledc_channel_config_t ledc_channel[LIGHT_PWM_CH_NUM] = {
        {
            .channel    = LIGHT_PWM_CHW_CHANNEL,
            .duty       = LIGHT_PWM_MAX_DUTY,
            .gpio_num   = LIGHT_PWM_CHW_GPIO,
            .speed_mode = LIGHT_PWM_MODE,
            .timer_sel  = LIGHT_PWM_TIMER_1
        },
        {
            .channel    = LIGHT_PWM_CHC_CHANNEL,
            .duty       = LIGHT_PWM_MAX_DUTY,
            .gpio_num   = LIGHT_PWM_CHC_GPIO,
            .speed_mode = LIGHT_PWM_MODE,
            .timer_sel  = LIGHT_PWM_TIMER
        },
        {
            .channel    = LIGHT_PWM_CHR_CHANNEL,
            .duty       = 0,
            .gpio_num   = LIGHT_PWM_CHR_GPIO,
            .speed_mode = LIGHT_PWM_MODE,
            .timer_sel  = LIGHT_PWM_TIMER
        },
        {
            .channel    = LIGHT_PWM_CHG_CHANNEL,
            .duty       = 0,
            .gpio_num   = LIGHT_PWM_CHG_GPIO,
            .speed_mode = LIGHT_PWM_MODE,
            .timer_sel  = LIGHT_PWM_TIMER
        },
        {
            .channel    = LIGHT_PWM_CHB_CHANNEL,
            .duty       = 0,
            .gpio_num   = LIGHT_PWM_CHB_GPIO,
            .speed_mode = LIGHT_PWM_MODE,
            .timer_sel  = LIGHT_PWM_TIMER
        },
    };

	uint8_t ch;

	// Set LED Controller with previously prepared configuration
    for (ch = 0; ch < LIGHT_PWM_CH_NUM; ch++) {
        ledc_channel_config(&ledc_channel[ch]);
    }
	
	return;
}

costaud, 2020-05-18:
驱动电路是什么样的 ？ （IO 输出高电平，灯亮还是灭？）

shenyong-iot01, 2020-05-18:
@costaud 高电平灯亮    低电平灯灭

shenyong-iot01, 2020-05-18:
@costaud
您好，这个问题从下面两个方面帮忙确认下，谢谢！ 
1. 首先请帮忙确认为什么有时出现如下log，大概3  ~ 4次就会出现一次。
2. 出现时为什么第一次SW_CPU_RESET和第二次RTCWDT_RTC_RESET  IO电平出现了变化，导致灯闪烁。

[16:56:05.620] rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:56:05.620] configsip: 0, SPIWP:0xee
[16:56:05.620] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:56:05.620] mode:DIO, clock div:1
[16:56:05.620] load:0x3fff0018,len:4
[16:56:05.620] load:0xffffffff,�子ΗCets Jun 8 2016 00:22:57
[16:56:06.551]
[16:56:06.551] rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
[16:56:06.551] configsip: 0, SPIWP:0xee
[16:56:06.551] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:56:06.551] mode:DIO, clock div:1
[16:56:06.551] load:0x3fff0018,len:4
[16:56:06.551] load:0x3fff001c,len:4900
[16:56:06.551] load:0x40078000,len:12424
[16:56:06.551] load:0x40080400,len:6916
[16:56:06.551] entry 0x40080730

costaud, 2020-05-18:
@shenyong-iot01  
关于RTCWDT 请参考https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_cn.pdf  3.8节。 

有一点不太明白，在快速开关机测试时候，是外部断电吗，还是软件重启？ 为什么第一次是 SW_CPU_RESET ？
请把正常上电的log也发一下。

关于电平变化，需要参考以上信息再进行判断。可能也与所用的IO32/33是RTC IO有关。

shenyong-iot01, 2020-05-19:
@costaud 
1. 快速开关机测试 是外部断电（断电源开关或插拔电源插头）
2. SW_CPU_RESET 为什么出现这个就不清楚了
3. 另外我司还有高电平灯灭  低电平灯亮的产品未发现闪烁现象（硬件说更改麻烦及增加成本，这个产品未更改驱动方式验证）。目前来看是SW_CPU_RESET时会出现高电平信号导致灯闪烁。
4. 正常上电log如下（注：bootloader为了快速启动按照贵司提供的文档做了优化，
[如何缩短 ESP32 应用启动时间.pdf](https://github.com/espressif/esp-idf/files/4647749/ESP32.pdf)）

[16:58:08.620] ets Jun  8 2016 00:22:57
[16:58:08.620] 
[16:58:08.620] rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:58:08.620] configsip: 0, SPIWP:0xee
[16:58:08.620] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:58:08.620] mode:DIO, clock div:1
[16:58:08.620] load:0x3fff0018,len:4
[16:58:08.620] load:0x3fff001c,len:4900
[16:58:08.620] load:0x40078000,len:12424
[16:58:08.620] load:0x40080400,len:6916
[16:58:08.620] entry 0x40080730
[16:58:08.620] [CONFIG] Read group data error
[16:58:08.907] Show group sets:
[16:58:08.907] 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
[16:58:08.907] count:0
[16:58:08.907] I (40) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[16:58:08.907] get_duty_mid: 220
[16:58:08.907] restart_times: 1
[16:58:08.907] -=[ =============================================== ]=-
[16:58:08.907] -=[ Firmware author: SYKJ                           ]=-
[16:58:08.907] -=[ SYKJ SDK version: M01.B001_t31                  ]=-
[16:58:08.907] -=[ ESP32 SDK version: v3.3.1-dirty                 ]=-
[16:58:08.907] -=[ Firmware version: E2.05.M00.B001_t11            ]=-
[16:58:08.907] -=[ Build time: GMT+8 2020-05-16 14:20              ]=-
[16:58:08.907] -=[ =============================================== ]=-

closed 2020-08-18 by costaud