espressif/esp-idf #5300: Ble mesh: Require clarification regarding how to reply the message when element response control set to ESP_BLE_MESH_SERVER_RSP_BY_APP

ankit-thealchemist, 2020-05-16:
## Environment

- Development Kit:      esp32 dev kit v1
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    //esp-idf 4.0
- Operating System:     [Windows]
.
- Using an IDE?: [Yes (please give details)]: vscode with esp-idf extension
- Power Supply:         [USB]

## Problem Description
I was implementing the server demo from the example of ble mesh [https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server](url).
In the example I change the configuration server relay  feature enable by setting value to ESP_BLE_MESH_RELAY_ENABLED also the gatt proxy value to ESP_BLE_MESH_GATT_PROXY_ENABLED.

It features 3 elements.  now in the  elements 1 , response control is set to ESP_BLE_MESH_SERVER_AUTO_RSP while other are set to the ESP_BLE_MESH_SERVER_RSP_BY_APP( both get and set). 
Now I then uploaded the code to the two nodes say node A and node B.  I provisioned the nodes, using the nrf mesh app v 2.1.0  and bind the app key to them.

Now suppose I connect to the node A using nrf mesh  and try to control the elements of the node B. To my surprise, the elements which are set to the control response to ESP_BLE_MESH_SERVER_AUTO_RSP, works perfectly, while the elements which sets to  ESP_BLE_MESH_SERVER_RSP_BY_APP, doesn't able to send the status message to the nrf mesh app. 
I donot change anyting in the example. some how both the function esp_ble_mesh_server_model_send_msg() and esp_ble_mesh_model_publish() doesnt sent the message back to nrf mesh.
when I check the logs, same thing comfirms. 
[image]
0x0001 is the unicast address of the nrf mesh app and 0x0002 is the unicast address of the element which has the response control set to ESP_BLE_MESH_SERVER_AUTO_RSP.

[image]
 0x0003 is the unicast address of the model whose response control set to ESP_BLE_MESH_SERVER_RSP_BY_APP. there is no packet which relay back to the nrf mesh app (add 0x0001)

Is there is anything more has to be done to send the message back to the originator through mesh network, when the control response of the model set to ESP_BLE_MESH_SERVER_RSP_BY_APP.

### Expected Behavior
should be able to get the status message in the nrf mesh.

### Actual Behavior
when the control response is set to the ESP_BLE_MESH_SERVER_RSP_BY_APP, message are not received by the nrf mesh.

### Steps to reproduce

1. step1 take more than 2 esp 32 devices and upload the above examples to all of them. provisioned them.

2. ... Now connect to anyone of the device and try to read status of  the 1st element of the other node. you can easily do that, that model has ESP_BLE_MESH_SERVER_AUTO_RSP.  while other element struggle(getting error) because they are set to ESP_BLE_MESH_SERVER_RSP_BY_APP.

### Code to reproduce this issue

```cpp

static esp_ble_mesh_cfg_srv_t config_server = {
    .relay = ESP_BLE_MESH_RELAY_ENABLED,
    .beacon = ESP_BLE_MESH_BEACON_ENABLED,
#if defined(CONFIG_BLE_MESH_FRIEND)
    .friend_state = ESP_BLE_MESH_FRIEND_ENABLED,
#else
    .friend_state = ESP_BLE_MESH_FRIEND_NOT_SUPPORTED,
#endif
#if defined(CONFIG_BLE_MESH_GATT_PROXY_SERVER)
    .gatt_proxy = ESP_BLE_MESH_GATT_PROXY_ENABLED,
#else
    .gatt_proxy = ESP_BLE_MESH_GATT_PROXY_NOT_SUPPORTED,
#endif
    .default_ttl = 7,
    /* 3 transmissions with 20ms interval */
    .net_transmit = ESP_BLE_MESH_TRANSMIT(2, 20),
    .relay_retransmit = ESP_BLE_MESH_TRANSMIT(2, 20),
};

ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_pub_0, 2 + 3, ROLE_NODE);
static esp_ble_mesh_gen_onoff_srv_t onoff_server_0 = {
    .rsp_ctrl.get_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
    .rsp_ctrl.set_auto_rsp = ESP_BLE_MESH_SERVER_AUTO_RSP,
};

ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_pub_1, 2 + 3, ROLE_NODE);
static esp_ble_mesh_gen_onoff_srv_t onoff_server_1 = {
    .rsp_ctrl.get_auto_rsp = ESP_BLE_MESH_SERVER_RSP_BY_APP,
    .rsp_ctrl.set_auto_rsp = ESP_BLE_MESH_SERVER_RSP_BY_APP,
};

ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_pub_2, 2 + 3, ROLE_NODE);
static esp_ble_mesh_gen_onoff_srv_t onoff_server_2 = {
    .rsp_ctrl.get_auto_rsp = ESP_BLE_MESH_SERVER_RSP_BY_APP,
    .rsp_ctrl.set_auto_rsp = ESP_BLE_MESH_SERVER_RSP_BY_APP,
};

Alvin1Zhang, 2020-05-18:
@ankit-thealchemist Thanks for the detailed report and letting us know about this. We will look into. Thanks.

Campou, 2020-07-01:
Hi @ankit-thealchemist 

Sorry for the late reply.

This happens because when the status is replied by the application, the TTL is 0, which causes the status failed to be "relayed" to the App. Please have a try with this patch [0001-ble_mesh-fix-send_ttl-not-updated.patch.zip](https://github.com/espressif/esp-idf/files/4857620/0001-ble_mesh-fix-send_ttl-not-updated.patch.zip).

Thanks.

Alvin1Zhang, 2020-07-10:
@ankit-thealchemist Thanks for reporting, would you please help share if any updates for this issue? Thanks.

ankit-thealchemist, 2020-07-10:
> @ankit-thealchemist Thanks for reporting, would you please help share if any updates for this issue? Thanks.

Ya I am testing it. will let you know by monday.

closed 2020-07-10 by espressif-bot

ankit-thealchemist, 2020-07-11:
> Hi @ankit-thealchemist
>
> Sorry for the late reply.
>
> This happens because when the status is replied by the application, the TTL is 0, which causes the status failed to be "relayed" to the App. Please [...]

Hi @Campou @Alvin1Zhang  I applied the patch and found that it is working properly. Thanks for the resolution. Just a observation that ESP_BLE_MESH_SERVER_AUTO_RSP is slightly faster than the ESP_BLE_MESH_SERVER_RSP_BY_APP.

Alvin1Zhang, 2020-07-13:
Thanks for providing the feedback.