espressif/esp-idf #5302: esp_http_client_get_chunk_length() for chunked responses

doragasu, 2020-05-16:
In the past I have successfully used the esp-http-client, and it was really simple exchanging headers and then receiving the data length specified in the `content-length` reply header. But now I am interfacing an API that responds with chunked transfers. Chunked responses lack the `content-length` header, and currently the API does not allow obtaining the current chunk length. `esp_http_client_fetch_headers()` returns a length of 0 bytes, and `esp_http_client_get_content_length()` returns -1 for chunked transfers.

Since each chunk length is written just before the chunk starts, it should be easy writing a function, something like `esp_http_client_get_chunk_length()` to get the current chunk length of chunked responses.

Alvin1Zhang, 2020-05-18:
@doragasu Thanks for raising this feature request.

shubhamkulkarni97, 2020-05-28:
@doragasu ,

Can you share the use case of this API and details about the chunked server you are using? Is it a local server or an online server?

doragasu, 2020-05-30:
This behavior occurs when interfacing GameJolt Game API: https://gamejolt.com/game-api/doc

You can test for example against the following API call: https://api.gamejolt.com/api/game/v1_2/time/?game_id=496055&format=keypair&signature=4756d315673a00abd4f60966bb12c8b135f5be80

When doing the call above, debug log shows this:
```
D (10303) http_url_set: init, HTTP URL: https://api.gamejolt.com/api/game/v1_2/time/?game_id=496055&format=keypair&signature=4756d315673a00abd4f60966bb12c8b135f5be80
W (10318) http_init: no valid certificate found, len 8192000, start: 0
D (10327) http_method_set: set method 0
D (10329) http_method_set: HTTP method: 0
D (10332) http_open: opening
D (10335) HTTP_CLIENT: Begin connect to: https://api.gamejolt.com:443
D (10341) esp-tls: host:api.gamejolt.com: strlen 16
D (11625) esp-tls: handshake in progress...
D (15832) HTTP_CLIENT: Write header[3]: GET /api/game/v1_2/time/?game_id=496055&format=keypair&signature=4756d315673a00abd4f60966bb12c8b135f5be80 HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: api.gamejolt.com
Content-Length: 0

D (15852) http_open: HTTP open OK, 0 bytes
D (16003) HTTP_CLIENT: on_message_begin
D (16005) HTTP_CLIENT: HEADER=Server:nginx
D (16008) HTTP_CLIENT: HEADER=Date:Sat, 30 May 2020 07:00:49 GMT
D (16011) HTTP_CLIENT: HEADER=Content-Type:text/plain;charset=UTF-8
D (16015) HTTP_CLIENT: HEADER=Transfer-Encoding:chunked
D (16018) HTTP_CLIENT: HEADER=Vary:Accept-Encoding
D (16022) HTTP_CLIENT: HEADER=Expires:Sat, 30 May 2020 07:00:48 GMT
D (16028) HTTP_CLIENT: HEADER=Cache-Control:no-cache
D (16033) HTTP_CLIENT: HEADER=Access-Control-Allow-Origin:*
D (16038) HTTP_CLIENT: HEADER=Access-Control-Allow-Credentials:false
D (16044) HTTP_CLIENT: http_on_headers_complete, status=200, offset=300, nread=300
D (16052) HTTP_CLIENT: http_on_body 138
D (16055) HTTP_CLIENT: http_on_chunk_complete
D (16059) HTTP_CLIENT: http_on_chunk_complete
D (16063) HTTP_CLIENT: http_on_message_complete, parser=40107e1c
D (16069) HTTP_CLIENT: content_length = -1
D (16073) http_finish: HTTP finish: 200: 2147483647 bytes
D (16087) HTTP_CLIENT: is_data_remain=0, is_chunked=1
D (16099) HTTP_CLIENT: is_data_remain=0, is_chunked=1
I (16102) http_recv: server closed the connection
```
The HTTP client works, but there is no way to get the chunk length in advance. That is what I am requesting, to implement a function to get chunk length in advance.

closed 2020-06-12 by igrr