espressif/esp-idf #5303: ESP32 xQueueReceive UART AWS IoT

ncohen17, 2020-05-17:
I get error code -13 (NETWORK_DISCONNECTED_ERROR) when I call xQueueReceive in a loop. When I remove xQueueReceive it works and connects to AWS.

while (NETWORK_ATTEMPTING_RECONNECT == rc || NETWORK_RECONNECTED == rc || SUCCESS == rc)
    {

        rc = aws_iot_mqtt_yield(&mqttClient, 100);

        if (NETWORK_ATTEMPTING_RECONNECT == rc)
        {
            rc = aws_iot_mqtt_yield(&mqttClient, 1000);
            continue;
        }

        ESP_LOGI(AWSIOTTAG, "Stack remaining for task '%s' is %d bytes", pcTaskGetTaskName(NULL), uxTaskGetStackHighWaterMark(NULL));

        if (xQueueReceive(uartDataQueue, &uartData, 0))
        {
            //pubData = (char *)&uartData;
            //ESP_LOGI(AWSIOTTAG, "UART Received : %s", pubData);
            //xQueueReset(uartDataQueue);
        }
        else
        {
            ESP_LOGI(AWSIOTTAG, "No Data");
        }
        vTaskDelay(1000 / portTICK_PERIOD_MS);

        sprintf(cPayload, "%s", "hello from CHIP new");
        paramsQOS1.payloadLen = strlen(cPayload);
        rc = aws_iot_mqtt_publish(&mqttClient, TOPIC, TOPIC_LEN, &paramsQOS1);
        ESP_LOGE(AWSIOTTAG, "Error : %d ", rc);
    }

Please advise

Alvin1Zhang, 2020-05-18:
@ncohen17 Thanks for the report and letting us know about this. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

mahavirj, 2020-06-26:
Closing due to lack of response, please feel free to re-open with additional information per github issue template.

closed 2020-06-26 by mahavirj