espressif/esp-idf #5304: Tools installation fails for ESP-IDF v4.0 in Linux Mint 19.3

PascalBod, 2020-05-17:
## Environment

- Development Kit: N/A
- Kit version: N/A
- Module or chip used: N/A
- IDF version: v4.0
- Build System: N/A
- Compiler version: N/A
- Operating System: Linux (Linux Mint 19.3, based on Ubuntu 18.04)
- Using an IDE?: No
- Power Supply: N/A

## Problem Description

I started setting up ESP-IDF development environment according to [this page](https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html), i.e. for the stable/v4.0 version. When I run `install.sh`, I get errors below, and the installation stops:

```
...
Collecting scandir; python_version < "3.5" (from pathlib2<3,>=2.3.3; python_version < "3.4" and sys_platform != "win32"->virtualenv)
  Downloading https://files.pythonhosted.org/packages/df/f5/9c052db7bd54d0cbf1bc0bb6554362bba1012d03e5888950a4f5c5dadc4e/scandir-1.10.0.tar.gz
Building wheels for collected packages: filelock, distlib, scandir
  Running setup.py bdist_wheel for filelock ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-RP6Hye/filelock/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpOIEZF2pip-wheel- --python-tag cp27:
  /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for filelock
  Running setup.py clean for filelock
  Running setup.py bdist_wheel for distlib ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-RP6Hye/distlib/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp4mlwx4pip-wheel- --python-tag cp27:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for distlib
  Running setup.py clean for distlib
  Running setup.py bdist_wheel for scandir ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-RP6Hye/scandir/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpmaPu8dpip-wheel- --python-tag cp27:
  /usr/lib/python2.7/distutils/extension.py:133: UserWarning: Unknown Extension options: 'optional'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for scandir
  Running setup.py clean for scandir
Failed to build filelock distlib scandir
Installing collected packages: filelock, six, distlib, appdirs, contextlib2, zipp, configparser, scandir, pathlib2, importlib-metadata, singledispatch, typing, importlib-resources, virtualenv
  Running setup.py install for filelock ... done
  Running setup.py install for distlib ... done
  Running setup.py install for scandir ... done
Successfully installed appdirs-1.4.4 configparser-4.0.2 contextlib2-0.6.0.post1 distlib-0.3.0 filelock-3.0.12 importlib-metadata-1.6.0 importlib-resources-1.5.0 pathlib2-2.3.5 scandir-1.10.0 singledispatch-3.4.0.3 six-1.14.0 typing-3.7.4.1 virtualenv-20.0.20 zipp-1.2.0
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython2-posix}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                  [--clear] [--system-site-packages] [--symlinks | --copies] [--no-download | --download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
                  [--symlink-app-data] [--prompt prompt] [-h]
                  dest
virtualenv: error: unrecognized arguments: --no-site-packages
Traceback (most recent call last):
  File "/home/pascalb/esp/esp-idf/tools/idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "/home/pascalb/esp/esp-idf/tools/idf_tools.py", line 1368, in main
    action_func(args)
  File "/home/pascalb/esp/esp-idf/tools/idf_tools.py", line 1184, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'virtualenv', '--no-site-packages', '/home/pascalb/.espressif/python_env/idf4.0_py2.7_env']' returned non-zero exit status 2
```

### Expected Behavior

Successful tools installation.

### Actual Behavior

The installation fails.

### Steps to reproduce

1. start tools installation for the stable version of ESP-IDF  in Linux Mint 19.3

### Code to reproduce this issue

N/A

## Debug Logs

N/A

## Other items if possible

N/A

igrr, 2020-05-17:
@PascalBod the error in v4.0 release has been fixed in release/v4.0 branch, please check https://github.com/espressif/esp-idf/issues/5204#issuecomment-620537960

PascalBod, 2020-05-17:
Thanks a lot, @igrr, I just installed release/v4.0 branch, and, yes, it works!

closed 2020-05-17 by PascalBod