espressif/esp-idf #5305: Invalid command bdist_wheel error while installing tools for latest ESP-IDF in Linux Mint 19.3

PascalBod, 2020-05-17:
## Environment

- Development Kit: N/A
- Kit version: N/A
- Module or chip used: N/A
- IDF version: v4.2-dev-1541-g5e436be77
- Build System: N/A
- Compiler version: N/A
- Operating System: Linux (Linux Mint 19.3, based on Ubuntu 18.04)
- Using an IDE?: N/A
- Power Supply: N/A

## Problem Description

I started setting up ESP-IDF development environment according to [this page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html), i.e. for the latest version. When I run `install.sh`, I get the error message below. The installation proceeds, and succeeds. I can then flash a DevKitC and monitor the running application.

**Note:** before starting the installation, I set python3 as the default python.

```
...
Building wheels for collected packages: distlib
  Running setup.py bdist_wheel for distlib ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-o_noic51/distlib/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpxqgwvyhfpip-wheel- --python-tag cp36:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for distlib
  Running setup.py clean for distlib
Failed to build distlib
Installing collected packages: zipp, importlib-metadata, distlib, six, importlib-resources, appdirs, filelock, virtualenv
  Running setup.py install for distlib ... done
Successfully installed appdirs-1.4.4 distlib-0.3.0 filelock-3.0.12 importlib-metadata-1.6.0 importlib-resources-1.5.0 six-1.14.0 virtualenv-20.0.20 zipp-3.1.0
...
```

### Expected Behavior

No error message.

### Actual Behavior

An error message is displayed.

### Steps to reproduce

1. set python3 as default python
2. start tools installation for the latest version of ESP-IDF

### Code to reproduce this issue

N/A

## Debug Logs

N/A

## Other items if possible

N/A

Alvin1Zhang, 2020-05-18:
Thanks for the report and letting us know about this. We will look into.

igrr, 2020-05-18:
Hi @PascalBod, could you please attach the full output of install.sh? From the part you have posted it isn't clear why only a few packages are being installed, not the rest of the requirements.txt.

PascalBod, 2020-05-19:
Hi @igrr,
Sorry for my late answer. Please find attached the [full output of the installation](https://github.com/espressif/esp-idf/files/4651917/20200519-installLatest.txt).
Note: version is now `v4.2-dev-1575-ga3520970f`.

igrr, 2020-05-19:
I see, thanks, it looks like this error happens when we are trying to install `virtualenv` package.

Can you please try installing `wheel` package? I think on Ubuntu you can do this with `sudo apt-get install python3-wheel`, if your python3 was installed using apt-get, or `python3 -m pip install --user wheel` otherwise.

PascalBod, 2020-05-20:
I installed `python3-wheel` and `python3-virtualenv` packages, and the installation doesn't display the error message anymore. Thanks!

BTW, regarding the [webpage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html) providing python3 prerequisites, I now remember that I reported a few days ago that I had to install `python3-pip`, `python3-setuptools` and `python3-wheel` packages. The webpage was modified accordingly, excepted for `python3-wheel` :slightly_smiling_face: Elisa told me that this was due to a developer's feedback. Could be that some package dependency exist in Ubuntu that do not exist in Linux Mint (?). And then I forgot about `python3-wheel` :slightly_frowning_face:

closed 2020-05-20 by PascalBod