espressif/esp-idf #5308: Linking to standard library fails when compiling in official Docker container

inkychris, 2020-05-18:
## Environment
- `espressf/idf` Docker image: tags `v3.3.2` (also had other linking issues), `release-v4.2`, and `latest`.
- Build System: Make with the following environment variables set for the container:
```
AR: xtensa-esp32-elf-ar
AS: xtensa-esp32-elf-as
CC: xtensa-esp32-elf-gcc
CXX: xtensa-esp32-elf-g++
LD: xtensa-esp32-elf-ld
RANLIB: xtensa-esp32-elf-ranlib
STRIP: xtensa-esp32-elf-strip
```
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0 (v4.2 image)

## Problem Description
A linker error is encountered when compiling an application, failing to find symbols which appear to be for pthread. The unresolved symbols are flagged within the standard library rather than client code.

### Expected Behavior
Application that doesn't directly use pthread is linked successfully to the standard libraries.

### Actual Behavior
Linker failure:
```
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-stdio.o):(.literal+0x20): warning: _close_r is not implemented and will always fail
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-fseeko.o):(.literal+0x40): warning: _fstat_r is not implemented and will always fail
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-stdio.o):(.literal+0xc): warning: _lseek_r is not implemented and will always fail
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-fopen.o):(.literal+0x18): warning: _open_r is not implemented and will always fail
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-stdio.o):(.literal+0x4): warning: _read_r is not implemented and will always fail
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-stdio.o):(.literal+0x10): warning: _write_r is not implemented and will always fail
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-fclose.o):(.literal+0x10): warning: pthread_setcancelstate is not implemented and will always fail
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-errno.o):(.literal+0x0): warning: __getreent is not implemented and will always fail
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o):(.literal._ZL28read_encoded_value_with_basehjPKhPj+0x4): undefined reference to `abort'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o): in function `read_encoded_value_with_base(unsigned char, unsigned int, unsigned char const*, unsigned int*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:257: undefined reference to `abort'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o): in function `size_of_encoded_value':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:88: undefined reference to `abort'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_personality.o): in function `base_of_encoded_value(unsigned char, _Unwind_Context*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:118: undefined reference to `abort'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_terminate.o): in function `__cxxabiv1::__terminate(void (*)())':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48: undefined reference to `abort'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_terminate.o):/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:51: more undefined references to `abort' follow
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_alloc.o):(.literal._ZN9__gnu_cxx7__mutex4lockEv[__gnu_cxx::__mutex::lock()]+0x8): undefined reference to `pthread_mutex_lock'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_alloc.o):(.literal._ZN9__gnu_cxx13__scoped_lockD5Ev[__gnu_cxx::__scoped_lock::~__scoped_lock()]+0x8): undefined reference to `pthread_mutex_unlock'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_alloc.o):(.literal.startup._GLOBAL__sub_I__ZN9__gnu_cxx9__freeresEv+0x0): undefined reference to `pthread_mutex_init'
/opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /opt/esp/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/libstdc++.a(eh_alloc.o):(.literal.startup._GLOBAL__sub_I__ZN9__gnu_cxx9__freeresEv+0x4): undefined reference to `pthread_mutex_destroy'
[... cut ...] tdc++.a(eh_term_handler.o):(.data._ZN10__cxxabiv119__terminate_handlerE+0x0): undefined reference to `abort'
collect2: error: ld returned 1 exit status
```
### Steps to reproduce
1. Create minimal app which includes the standard library:
```
// main.cpp
#include <string>

int main() {
    std::string test = "Hello, world!";
    return 0;
}
```
2. Attempt to compile it inside a container:
```
$ docker run --rm -it -v ${PWD}:/src -w /src espressif/idf:release-v4.2 xtensa-esp32-elf-g++ main.cpp
```

igrr, 2020-05-18:
@inkychris please try adding `--specs=nosys.specs` to the compiler command line — this will include default (no-op) implementations of system calls.

igrr, 2020-05-19:
Sorry, apparently we still don't have the stubs for `abort` and for the pthread functions in libnosys. Will fix this in the future toolchain release.
For now you could work around this by defining no-op implementations of the missing functions.

inkychris, 2020-05-19:
Ah okay, was just about to give this a try. Will add the stubs in manually and see how that goes!

igrr, 2020-05-19:
@inkychris Perhaps you could give a hint about what you are trying to achieve? You can get the program to link this way, which might be useful for some cases (like, configuring autoconf/automake based projects), but you won't get an application which can actually run on the chip this way.

inkychris, 2020-05-19:
I have a static library which is built and distributed for multiple platforms and there is also a private unit-test app that tests the library. This is all done in plain makefiles so for the most part, libraries can generally be compiled by swapping out the respective `CC`, `CXX`, etc. variables which is done using different containerised toolchains. The unit-tests are then run either natively or emulated. The library is compiling fine but the test app (which I'd then like to run on the emulator) is failing in the way described above.

TobiasBoeschMiele, 2020-06-03:
@inkychris, is your test app using [Catch2](https://github.com/catchorg/Catch2)?

TobiasBoeschMiele, 2020-06-03:
Since I receive similar error messages I'll ask for help in this issue.
@igrr can you try to integrate the [Catch2 tutorial](https://github.com/catchorg/Catch2/blob/master/docs/tutorial.md#writing-tests) into [esp-idf bluetooth peripheral example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph) to see if it works when done right? (Maybe I make some cmake mistakes here)
I want to test the application with the Catch2 test framework on the target and get a lot of linker errors like shown above.

There is a [cmake integration guide](https://github.com/catchorg/Catch2/blob/b8ce814ee6c5145ad1521f80bfd2c339012d1a23/docs/cmake-integration.md) which i used to integrate the Catch2 in the mentioned esp-idf example.

igrr, 2020-06-03:
@TobiasBoeschMiele Catch2 is header-only, so it should be possible to use it in IDF. You will need to enable exceptions and likely RTTI support in menuconfig.

Note that I am assuming that your top level project is an ESP-IDF build system project. If your top level project is a plain CMake project, then please make sure you go through the related section of the docs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-esp-idf-in-custom-cmake-projects.

inkychris, 2020-06-03:
I'm using gtest but I think the docs @igrr just linked may be useful. I'm guessing I need to compile and link to any missing idf components that the standard library needs like pthread.

TobiasBoeschMiele, 2020-06-03:
> @TobiasBoeschMiele Catch2 is header-only, so it should be possible to use it in IDF. You will need to enable exceptions and likely RTTI support in menuconfig.
> 
> Note that I am assuming that your top level project is an ESP-IDF build system project. If your top level project is a plain CMake project, then please make sure you go through the related section of the docs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-esp-idf-in-custom-cmake-projects and check the related example: https://github.com/espressif/esp-idf/tree/7d75213/examples/build_system/cmake/idf_as_lib

Top level is an esp-idf project. I'm trying to integrate catch2. I'll try to
>enable exceptions and likely RTTI support in menuconfig

I'm tinkering with gtest, too. Thanks for sharing your expirience @inkychris.

TobiasBoeschMiele, 2020-06-03:
So I tried to use the options mentioned @igrr 

>enable exceptions and likely RTTI support in menuconfig

without success.

An example error entry i get looks like this:
```
c:/espressif_esp_idf_tools/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/espressif_esp_idf_tools/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0\libgcc.a(unwind-dw2-fde.o): in function `size_of_encoded_value':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libgcc/unwind-pe.h:88: undefined reference to `abort'
```

This one is very similar to the ones in the description of @inkychris

TobiasBoeschMiele, 2020-06-04:
I think I'll give up on this and run the tests on PC with gcc. It is high level software anyway so hardware stubbing using a wrapper can be done.

antoinecarme, 2020-11-06:
Just had a similar issue. Would be nice to have this fixed in a coming release of the tool-chain.

https://github.com/antoinecarme/ml2cpp/blob/master/tests/small_devices_experiments/xtensa_missing_symbols.i

StridingDragon, 2022-11-16:
Has this ever been addressed? I am still seeing this issue.

closed 2023-02-24 by espressif-bot

igrr, 2023-02-24:
This should be possible now with
```
docker run --rm -it -v ${PWD}/src:/src -w /src espressif/idf:latest xtensa-esp32-elf-g++ -ffunction-sections -Wl,-gc-sections main.cpp
```

(we'll make it work also without -Wl,-gc-sections in a future release)