espressif/esp-idf #5311: Any example of esp_hid_device for Classic ？

elmagnificogi, 2020-05-18:
Any example of esp_hid_device for Classic ？
Now only have the ble version,but the host has Classic and BLE

Wth-Esp, 2020-05-19:
Hi @elmagnificogi 

Please wait for a few days, the HID component has been recently merged into the Master branch and will soon sync on Github. Please keep the IDF update.

Thanks
Weitianhua

keelung-yang, 2020-05-29:
@Wth-Esp I'm waiting for this too. I can only wait for one week from now on, in one of our projects. So I want to know whether it will be synchronized into GitHub repo in next week?

Wth-Esp, 2020-05-29:
Hi all,
It's already on Github (branch: master)

https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_hid_host	
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_hid_device

Thanks
Weitianhua

Alvin1Zhang, 2020-05-29:
@elmagnificogi @keelung-yang Thanks for reporting and sharing updates. Feel free to reopen if any updates. Thanks.

Wth-Esp, 2020-05-29:
HI all,

The HID demo for classic BT is under a plan now. Maybe you can follow the BLE demo first and change the GAP API and HID API on the existing demo.

Thanks
Weitianhua

Wth-Esp, 2020-05-29:
Here is the demo.
https://github.com/espressif/esp-idf/pull/3425

elmagnificogi, 2020-05-29:
> Here is the demo.
> #3425
3425 the hid has some issue that i could fix it.so i need the offical example

ESP-IDF BLE HID Device Demo != Bluetooth classic HID

Wth-Esp, 2020-09-16:
Hi, all

We are now ready to review the [PR](https://github.com/espressif/esp-idf/pull/3425) which added HID Device for Classic BT.

So, please wait for some time.

Thanks
Wei Tianhua

Wth-Esp, 2021-01-18:
Hi all,

We have almost done the HID Device component based on Classic BT and we decide to have more tests on it including the BQB test case suit. I promise here the feature shall release with IDF v4.4.

Thanks for your patience, we much appreciate it.

elmagnificogi, 2021-01-18:
> Hi all,
> 
> We have almost done the HID Device component based on Classic BT and we decide to have more tests on it including the BQB test case suit. I promise here the feature shall release with IDF v4.4.
> 
> Thanks for your patience, we much appreciate it.

could we test it now? i wait for almost 1 year

closed 2021-09-10 by espressif-bot