espressif/esp-idf #5312: reusable console outside of uart

aep, 2020-05-18:
**Is your feature request related to a problem? Please describe.**

the console component allows building nicely reusable command components,
but they cannot be re-used if the i/o is not uart.

**Describe the solution you'd like**

provide a way to feed the console main-loop from custom i/o

**Additional context**

i'm the author of devguard.io, a free and open source remote shell on embedded systems. A port to esp32 is in the making, but i'd rather use the existing console components than bring my own shell

igrr, 2020-05-18:
@aep I think it should be possible to achieve this, using funopen (3) to create custom stdin/stdout streams. We can add an example with TLS as a transport, would that help?

aep, 2020-05-18:
wow that sounds alot easier than I thought.
Of course I would welcome an example, but It actually sounds easy enough.

I do wonder how to handle the FILE to the console component though. It opens stdin internally directly.

igrr, 2020-05-18:
`stdin` and `stdout` are thread-local pointers, so you can simply reassign them in the task which deals with the console. We can add some function to substitute stdin/stdout usage inside linenoise with user-provided streams, but simply assigning custom FILE pointers should work for now.

igrr, 2020-05-18:
Unrelated use case, but logic would be the same: https://esp32.com/viewtopic.php?f=13&t=13997#p55141

aep, 2020-05-19:
unfortunately i could not get it working. linenoise calls fread in getCurserPosition with 1 byte, but the function i passed to fropen gets called with a buffer of 128 repeatedly without ever returning back to fread

aep, 2020-05-19:
finally got it working by simply removing getCurserPosition. the funopen idea works fine, except i had to setvbuf(stdin, NULL, _IONBF,0) otherwise it would just buffer up forever.

thanks so much for your help!

closed 2020-05-19 by aep