espressif/esp-idf #5313: On windows using git bash or cmd, ninja monitor just terminate without any error message

callalilychen, 2020-05-18:
## Environment

- Development Kit: not relevant
- Kit version (for WroverKit/PicoKit/DevKitC): not relevant
- Module or chip used:  not relevant
- IDF version (run ``git describe --tags`` to find it): v4.0
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): not relevant
- Operating System:    Windows
- (Windows only) environment type:  ESP Command Prompt|Plain Command Prompt|git bash.
- Using an IDE?:  No
- Power Supply:  not relevant

## Problem Description

`ninja monitor` in build folder just terminate without any error message. No terminal for serial interface is started.
`idf.py monitor` in root folder works

### Expected Behavior
`ninja monitor` in build folder should also works like `idf.py monitor` in root folder 

### Actual Behavior
`ninja monitor` in build folder just terminate without any error message. No terminal for serial interface is started.

### Steps to reproduce
```
mkdir build
cd build
cmake ..
ninja flash
ninja monitor
```

### Code to reproduce this issue

It is reproducible to use a esp idf example

Alvin1Zhang, 2020-05-19:
@callalilychen Thanks for the report and letting us know about this. We will look into.

ESP-Marius, 2024-08-15:
As far as I can see `ninja monitor` works fine on master now. 

If anyone still has issues with this feel free to reopen the issue

closed 2024-08-15 by ESP-Marius