espressif/esp-idf #5315: install.sh cannot access to tooling URL

tatsutaigu, 2020-05-19:
## Environment

- IDF version: v4.2-dev-1575-ga3520970f 
- Operating System:   Windows10  (Ubuntu on WSL)
- (Windows only) environment type: Ubuntu on WSL
- environment: proxy are set in each files and proxy are worked.
git proxy is set in the .bashrc.
~/.bashrc
~/.wgetrc
~/.curlrc

## Problem Description
it is similar [https://github.com/espressif/esp-idf/issues/4364](https://github.com/espressif/esp-idf/issues/4364) .
install.sh cannot access to tooling URL.

```
$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2020r1-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz to /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp
Done200%
WARNING: file size mismatch for /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp, expected 85490835, got 11
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz to /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz to /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp
Done200%
WARNING: file size mismatch for /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp, expected 85490835, got 11
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz to /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz to /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp
Done200%
WARNING: file size mismatch for /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp, expected 85490835, got 11
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz to /home/tatsutaigu/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/home/tatsutaigu/esp/esp-idf/tools/idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "/home/tatsutaigu/esp/esp-idf/tools/idf_tools.py", line 1488, in main
    action_func(args)
  File "/home/tatsutaigu/esp/esp-idf/tools/idf_tools.py", line 1181, in action_install
    tool_obj.download(tool_version)
  File "/home/tatsutaigu/esp/esp-idf/tools/idf_tools.py", line 598, in download
    raise DownloadError()
__main__.DownloadError
```

below is solution.
each tooling are downloaded before running `install.sh`.
I guess `install.sh` may not access to tooling URL to through proxy.

```
$ cd ~/.espressif/dist
$ wget https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz
Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/home/tatsutaigu/.wget-hsts'. HSTS will be disabled.
--2020-05-19 09:49:34--  https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz
Resolving PROXY_URL.coml (PROXY_URL.coml)... xxx.xxx.xxx.xxx
Connecting to PROXY_URL.coml (PROXY_URL.coml)|xxx.xxx.xxx.xxx|:PORT... connected.
Proxy request sent, awaiting response... 200 OK
Length: 85490835 (82M) [application/octet-stream]
Saving to: ‘xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz’

                        xtensa-esp  35%[======================>                                          ]  28.93M   432KB/s    eta 2m 47s  
						
$ wget https://dl.espressif.com/dl/xtensa-esp32s2-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz
$ wget https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-linux-amd64-2.28.51-esp-20191205.tar.gz
$ wget https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32s2ulp-linux-amd64-2.28.51-esp-20191205.tar.gz
$ wget https://github.com/espressif/openocd-esp32/releases/download/v0.10.0-esp32-20200420/openocd-esp32-linux64-0.10.0-esp32-20200420.tar.gz

$ cd ~/esp/esp-idf
$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r1-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Installing openocd-esp32@v0.10.0-esp32-20200420
file openocd-esp32-linux64-0.10.0-esp32-20200420.tar.gz is already downloaded
Extracting /home/muto/.espressif/dist/openocd-esp32-linux64-0.10.0-esp32-20200420.tar.gz to /home/muto/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200420

....

Successfully installed Flask-0.12.5 Flask-Compress-1.5.0 Flask-SocketIO-2.9.6 Jinja2-2.11.2 MarkupSafe-1.1.1 Pygments-2.5.2 Werkzeug-0.16.1
bitstring-3.1.7 brotli-1.0.7 cffi-1.14.0 click-7.1.2 cryptography-2.9.2 enum34-1.1.10 future-0.18.2 gdbgui-0.13.2.0 gevent-1.5.0 greenlet-0.4.15 ipaddress-1.0.23 itsdangerous-1.1.0 pycparser-2.20 pyelftools-0.26 pygdbmi-0.9.0.2 pyparsing-2.3.1 pyserial-3.4 python-engineio-3.12.1
python-socketio-4.5.1 reedsolo-1.5.3 six-1.14.0
All done! You can now run:

  . ./export.sh

```

Alvin1Zhang, 2020-05-19:
@tatsutaigu Thanks for reporting, we will look into. Thanks.

tatsutaigu, 2020-05-19:
additional information
Each tooling files which are downloaded by install.sh is Bad request.
```
$ cat ~/.espressif/dist/binutils-esp32s2ulp-linux-amd64-2.28.51-esp-20191205.tar.gz.tmp
Bad request
```

igrr, 2020-05-19:
@tatsutaigu Regarding this part:

> environment: proxy are set in each files and proxy are worked.

Can you please clarify, how is the proxy set? What kind of proxy are you using (HTTP, SOCKS5, etc)? Also, what Python version are you using?

tatsutaigu, 2020-05-19:
@igrr 
Proxy is HTTP,
python is as below.
```
$ python -V
Python 2.7.15+ 
```

If you need more information, please let me know.

tatsutaigu, 2020-05-20:
@igrr 
additional information:
python 3.6 is already installed.
However, default python version is 2.7.
```
$ python3.6 -V
Python 3.6.8

$ python -V
Python 2.7.15+
```

igrr, 2020-05-20:
As mentioned in https://github.com/espressif/esp-idf/issues/4364#issuecomment-579978666, switching to Python 3 as the default python interpreter might be enough to get the proxy working, could you please try that? You can do this by creating a `python` symlink to `python3`.

tatsutaigu, 2020-05-20:
Thank you for your advice.

I can install ESP-IDF via proxy.
I wrote the solution with Ubuntu on WSL as below.

```
cd /usr/bin
sudo unlink python
sudo unlink python-config
sudo ln -s python3.6 python
sudo ln -s python3.6-config python-config

cd ~
sudo apt install python3-pip

cd ~/esp/esp-idf
./install.sh
```

closed 2020-05-20 by tatsutaigu