espressif/esp-idf #5316: let ESP-IDF support read HTML/CSS/JS files

walidamriou, 2020-05-19:
When we want to display HTML page in a web server based ESP32 we put HTML page with the CSS and js inside the main file or .h file, why not make ESP-IDF read HTML/CSS/js files from a folder or after calling it in the main from the source like we call .h files?

Alvin1Zhang, 2020-05-19:
@walidamriou Thanks for raising this feature request, we will evaluate. Thanks.

igrr, 2020-05-19:
Hi @walidamriou,

> When we want to display HTML page in a web server based ESP32 we put HTML page with the CSS and js inside the main file or .h file

This isn't the recommended approach. Have you met some recommendation to do this in IDF docs or examples?

Please refer to https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving and https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/restful_server which demonstrate loading of HTML files into SPIFFS and FAT partitions, respectively.

sealj553, 2020-05-19:
If you don't need/want a filesystem, you can add `set(COMPONENT_EMBED_TXTFILES "html_file_name.html")` in your `CMakeLists.txt`, then access your file from your .c file like so:
`extern const uint8_t html_file_array[] asm("_binary_html_file_name_html_start");`

Docs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#embedding-binary-data

igrr, 2020-05-21:
@walidamriou If the answers above resolve your question, could you please close the issue? Thanks.

closed 2020-05-21 by walidamriou